zocker-160/keyboard-center #4: ModuleNotFoundError: No module named 'lib.configparser'

aquova, 2021-09-07:
**Distribution:**
Arch Linux

**Installation method:**
Via https://aur.archlinux.org/packages/keyboard-center/

**Python Version**
Python 3.9.6

**Issue**
Installation of the package runs correctly, but when I attempt to start it I get the error -

```
Traceback (most recent call last):
  File "/usr/lib/keyboard-center/main.py", line 13, in <module>
    from lib.configparser import Configparser
ModuleNotFoundError: No module named 'lib.configparser'
```

I'm not familiar with `lib.configparser`, but it wasn't a valid import on any system I tried.

zocker-160, 2021-09-07:
thank you for the report, looking into it

aquova, 2021-09-08:
Ah, I believe I have found the issue. There needs to be a `__init__.py` in the `lib` directory. Once I added a blank one, the GUI started up.

Here is my output of `tree /usr/lib/keyboard-center`

```
/usr/lib/keyboard-center/
├── assets
│   ├── ibus-keyboard_dark.svg
│   ├── ibus-keyboard.svg
│   ├── input-keyboard-virtual.png
│   ├── input-keyboard-virtual.svg
│   ├── MEMORY_1.png
│   ├── MEMORY_2.png
│   ├── MEMORY_3.png
│   ├── MEMORY_4.png
│   └── Memorykeys.svg
├── config
│   └── testconfig.yaml.example
├── devices
│   ├── allkeys.py
│   ├── keyboard.py
│   ├── keymaps.py
│   └── __pycache__
│       ├── allkeys.cpython-39.pyc
│       ├── keyboard.cpython-39.pyc
│       └── keymaps.cpython-39.pyc
├── gui
│   ├── aboutWindow.ui
│   ├── CEntryButton.py
│   ├── customwidgets.py
│   ├── keypressWidget.ui
│   ├── mainwindow.ui
│   ├── __pycache__
│   │   ├── CEntryButton.cpython-39.pyc
│   │   ├── customwidgets.cpython-39.pyc
│   │   ├── ressources_rc.cpython-39.pyc
│   │   ├── Ui_aboutWindow.cpython-39.pyc
│   │   ├── Ui_keypressWidget.cpython-39.pyc
│   │   ├── Ui_mainwindow.cpython-39.pyc
│   │   └── Ui_serviceWindow.cpython-39.pyc
│   ├── ressources.qrc
│   ├── ressources_rc.py
│   ├── serviceWindow.ui
│   ├── Ui_aboutWindow.py
│   ├── Ui_keypressWidget.py
│   ├── Ui_mainwindow.py
│   └── Ui_serviceWindow.py
├── lib
│   ├── configparser.py
│   ├── inotify_simple.py
│   ├── __pycache__
│   │   ├── configparser.cpython-39.pyc
│   │   ├── inotify_simple.cpython-39.pyc
│   │   ├── pynotifier.cpython-39.pyc
│   │   └── servicehelper.cpython-39.pyc
│   ├── pynotifier.py
│   └── servicehelper.py
├── main.py
├── __pycache__
│   ├── main.cpython-39.pyc
│   └── service.cpython-39.pyc
└── service.py
```

zocker-160, 2021-09-08:
Ah yes I made a mistake with the `.gitignore` which also ignored all `__init__.py` files...

zocker-160, 2021-09-08:
I think that should fix this issue, if not, please reopen.

commit 7b84b8b7 (2021-09-08): fix gitignore to not include `__init__.py`

fix #4
closed 2021-09-08 by zocker-160