zocker-160/keyboard-center #5: Not working on G910

microdou, 2021-09-07:
Hardware:
```
Gaming Keyboard G910
0x46d:0xc32b
```

Error Message from `keyboard-center`:
```
FATAL ERROR
NO supported keyboard found! :(
(err:None)
```

zocker-160, 2021-09-07:
thank you for this report, currently only `046d:c335 Logitech, Inc. G910 Orion Spectrum Mechanical Keyboard` is supported, because that is the one I own.

Would you be willing to do some testing so I can add support for the Orion Spark?

Are you on Arch / Manjaro or Ubuntu?

microdou, 2021-09-07:
Sure. It is vanilla Arch.

zocker-160, 2021-09-08:
I just created a small script for you, which should hopefully help us with this: 

https://github.com/zocker-160/keyboard-center/blob/master/src/newDeviceDebugger.py

- make sure the background service is not running `systemctl --user stop keyboard-center.service`
- run with `python3 newDeviceDebugger.py`
- it should detect your keyboard and then you just press the G1-G9 keys one after each other and it should show binary data similar to this:

```
got data from keyboard:
b'\x11\xff\x08\x00\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:
b'\x11\xff\x08\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
```

you stop with CTRL + C

microdou, 2021-09-08:
I couldn't get "got data from keyboard" with current script...

```
$ sudo python newDeviceDebugger.py

DEBUG: Searching for keyboard...
DEBUG: Keyboard found! :)
DEBUG: requesting USB endpoint...
DEBUG: check and detach kernel driver if active
INFO: starting service...
DEBUG: Using selector: EpollSelector
DEBUG: disabling g810-led gkey mapping
INFO: g810-led could not be found

^[OP^[OQ^[OR^[OS^[[15~^[[17~^[[18~^[[19~^[[20~
```

zocker-160, 2021-09-08:
hmm interesting, you seem to get data in though...

ok two things: 
- firstly you don't need to run it with `sudo`, it does not need any root privileges.
- secondly, could you try running `evtest`(in the Arch community repo) and check if the Gkeys are detected as events?

Because if they are detected as events, then there is a Kernel driver running which catches the USB data before my application does.

microdou, 2021-09-08:
Yeah, G keys are definitely working and captured by kernel. But somehow the script is not getting what it should get.

`evtest` detects all the keys.
```
$sudo evtest

No device specified, trying to scan all of /dev/input/event*
Available devices:
...
/dev/input/event24:     Logitech Gaming Keyboard G910
/dev/input/event25:     Logitech Gaming Keyboard G910 Keyboard
/dev/input/event26:     Logitech Gaming Keyboard G910 Consumer Control
Select the device event number [0-26]: 24

Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x46d product 0xc32b version 0x111
Input device name: "Logitech Gaming Keyboard G910"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 1 (KEY_ESC)
    Event code 2 (KEY_1)
    Event code 3 (KEY_2)
    Event code 4 (KEY_3)
    Event code 5 (KEY_4)
    Event code 6 (KEY_5)
    Event code 7 (KEY_6)
    Event code 8 (KEY_7)
    Event code 9 (KEY_8)
    Event code 10 (KEY_9)
    Event code 11 (KEY_0)
    Event code 12 (KEY_MINUS)
    Event code 13 (KEY_EQUAL)
    Event code 14 (KEY_BACKSPACE)
    Event code 15 (KEY_TAB)
    Event code 16 (KEY_Q)
    Event code 17 (KEY_W)
    Event code 18 (KEY_E)
    Event code 19 (KEY_R)
    Event code 20 (KEY_T)
    Event code 21 (KEY_Y)
    Event code 22 (KEY_U)
    Event code 23 (KEY_I)
    Event code 24 (KEY_O)
    Event code 25 (KEY_P)
    Event code 26 (KEY_LEFTBRACE)
    Event code 27 (KEY_RIGHTBRACE)
    Event code 28 (KEY_ENTER)
    Event code 29 (KEY_LEFTCTRL)
    Event code 30 (KEY_A)
    Event code 31 (KEY_S)
    Event code 32 (KEY_D)
    Event code 33 (KEY_F)
    Event code 34 (KEY_G)
    Event code 35 (KEY_H)
    Event code 36 (KEY_J)
    Event code 37 (KEY_K)
    Event code 38 (KEY_L)
    Event code 39 (KEY_SEMICOLON)
    Event code 40 (KEY_APOSTROPHE)
    Event code 41 (KEY_GRAVE)
    Event code 42 (KEY_LEFTSHIFT)
    Event code 43 (KEY_BACKSLASH)
    Event code 44 (KEY_Z)
    Event code 45 (KEY_X)
    Event code 46 (KEY_C)
    Event code 47 (KEY_V)
    Event code 48 (KEY_B)
    Event code 49 (KEY_N)
    Event code 50 (KEY_M)
[... 112 lines cut ...]
    Event code 240 (KEY_UNKNOWN)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
  Event type 17 (EV_LED)
    Event code 0 (LED_NUML) state 0
    Event code 1 (LED_CAPSL) state 0
    Event code 2 (LED_SCROLLL) state 0
Key repeat handling:
  Repeat type 20 (EV_REP)
    Repeat code 0 (REP_DELAY)
      Value    250
    Repeat code 1 (REP_PERIOD)
      Value     33
Properties:
Testing ... (interrupt to exit)
```

`sudo` is required on my system. Maybe you have your privilege is elevated on your account?

I placed some mod to `grub` for my Logitec mouse, maybe that causes problem? Will remove it and try again.
```
/boot/grub/grub.cfg

GRUB_CMDLINE_LINUX_DEFAULT="loglevel=3 quiet nvidia-drm.modeset=1 acpi_backlight=vendor usbhid.mousepoll=1 systemd.unified_cgroup_hierarchy=false"
```

microdou, 2021-09-08:
I removed `usbhid.mousepoll=1` from grub. Still not working.

Well, I think I can try running the keyboard in vmware ...

microdou, 2021-09-08:
Still the same thing when keyboard is connected into Arch under vmware.

zocker-160, 2021-09-08:
> These are the only keys captured by your script.

yeah ok so those media keys are not getting catched by a driver it seems like, which is the reason why they get captured by the script, I wonder why that is not the case for the Gkeys though....

zocker-160, 2021-09-08:
> sudo is required on my system. Maybe you have your privilege is elevated on your account?

for what exactly do you need sudo? Attaching to a USB device shouldn't need root rights afaik, but I could be wrong

> evtest detects all the keys.

could you try pressing the Gkeys after running `evtest` ? It should show them in case they get catched by a Kernel driver

microdou, 2021-09-09:
`sudo` is required on my side, otherwise ...
```
$ python newDeviceDebugger.py 

DEBUG: Searching for keyboard...
DEBUG: Keyboard found! :)
DEBUG: requesting USB endpoint...
DEBUG: check and detach kernel driver if active
Traceback (most recent call last):
  File "/home/user/newDeviceDebugger.py", line 127, in <module>
    main()
  File "/home/user/newDeviceDebugger.py", line 105, in main
    if keyboard.is_kernel_driver_active(usbInterface[0]):
  File "/usr/lib/python3.9/site-packages/usb/core.py", line 1107, in is_kernel_driver_active
    self._ctx.managed_open()
  File "/usr/lib/python3.9/site-packages/usb/core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/lib/python3.9/site-packages/usb/core.py", line 131, in managed_open
    self.handle = self.backend.open_device(self.dev)
  File "/usr/lib/python3.9/site-packages/usb/backend/libusb1.py", line 804, in open_device
    return _DeviceHandle(dev)
  File "/usr/lib/python3.9/site-packages/usb/backend/libusb1.py", line 652, in __init__
    _check(_lib.libusb_open(self.devid, byref(self.handle)))
  File "/usr/lib/python3.9/site-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 13] Access denied (insufficient permissions)
```

```
$ evtest

No device specified, trying to scan all of /dev/input/event*
Not running as root, no devices may be available.
Available devices:
Select the device event number [0-0]: 
```

microdou, 2021-09-09:
`$ sudo evtest`, select keyboard, and then press G1-9
```
Event: time 1631151999.985661, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7003a
Event: time 1631151999.985661, type 1 (EV_KEY), code 59 (KEY_F1), value 1
Event: time 1631151999.985661, -------------- SYN_REPORT ------------
^[OPEvent: time 1631152000.122635, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7003a
Event: time 1631152000.122635, type 1 (EV_KEY), code 59 (KEY_F1), value 0
Event: time 1631152000.122635, -------------- SYN_REPORT ------------

Event: time 1631152000.606759, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7003b
Event: time 1631152000.606759, type 1 (EV_KEY), code 60 (KEY_F2), value 1
Event: time 1631152000.606759, -------------- SYN_REPORT ------------
^[OQEvent: time 1631152000.755692, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7003b
Event: time 1631152000.755692, type 1 (EV_KEY), code 60 (KEY_F2), value 0
Event: time 1631152000.755692, -------------- SYN_REPORT ------------

Event: time 1631152002.099692, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7003c
Event: time 1631152002.099692, type 1 (EV_KEY), code 61 (KEY_F3), value 1
Event: time 1631152002.099692, -------------- SYN_REPORT ------------
^[OREvent: time 1631152002.218696, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7003c
Event: time 1631152002.218696, type 1 (EV_KEY), code 61 (KEY_F3), value 0
Event: time 1631152002.218696, -------------- SYN_REPORT ------------

Event: time 1631152002.594693, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7003d
Event: time 1631152002.594693, type 1 (EV_KEY), code 62 (KEY_F4), value 1
Event: time 1631152002.594693, -------------- SYN_REPORT ------------
^[OSEvent: time 1631152002.710695, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7003d
Event: time 1631152002.710695, type 1 (EV_KEY), code 62 (KEY_F4), value 0
Event: time 1631152002.710695, -------------- SYN_REPORT ------------

Event: time 1631152003.017729, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7003e
Event: time 1631152003.017729, type 1 (EV_KEY), code 63 (KEY_F5), value 1
Event: time 1631152003.017729, -------------- SYN_REPORT ------------
^[[15~Event: time 1631152003.141675, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7003e
Event: time 1631152003.141675, type 1 (EV_KEY), code 63 (KEY_F5), value 0
Event: time 1631152003.141675, -------------- SYN_REPORT ------------

Event: time 1631152004.344727, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7003f
Event: time 1631152004.344727, type 1 (EV_KEY), code 64 (KEY_F6), value 1
Event: time 1631152004.344727, -------------- SYN_REPORT ------------
^[[17~Event: time 1631152004.473742, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7003f
Event: time 1631152004.473742, type 1 (EV_KEY), code 64 (KEY_F6), value 0
Event: time 1631152004.473742, -------------- SYN_REPORT ------------

Event: time 1631152004.664790, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70040
Event: time 1631152004.664790, type 1 (EV_KEY), code 65 (KEY_F7), value 1
Event: time 1631152004.664790, -------------- SYN_REPORT ------------
^[[18~Event: time 1631152004.788713, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70040
Event: time 1631152004.788713, type 1 (EV_KEY), code 65 (KEY_F7), value 0
Event: time 1631152004.788713, -------------- SYN_REPORT ------------

Event: time 1631152004.930676, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70041
Event: time 1631152004.930676, type 1 (EV_KEY), code 66 (KEY_F8), value 1
Event: time 1631152004.930676, -------------- SYN_REPORT ------------
^[[19~Event: time 1631152005.059731, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70041
Event: time 1631152005.059731, type 1 (EV_KEY), code 66 (KEY_F8), value 0
Event: time 1631152005.059731, -------------- SYN_REPORT ------------

Event: time 1631152005.184704, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70042
Event: time 1631152005.184704, type 1 (EV_KEY), code 67 (KEY_F9), value 1
Event: time 1631152005.184704, -------------- SYN_REPORT ------------
^[[20~Event: time 1631152005.309794, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70042
Event: time 1631152005.309794, type 1 (EV_KEY), code 67 (KEY_F9), value 0
Event: time 1631152005.309794, -------------- SYN_REPORT ------------

```

I notice the output is equivalent to that from keys F1-9

microdou, 2021-09-09:
Looks like error is captured at this line in the script:

`except core.USBError`

zocker-160, 2021-09-09:
interesting, apparently the check if kernel driver is running, needs root rights on some systems `keyboard.is_kernel_driver_active(usbInterface[0])`, which I did not know, need to investigate why that is.

> $ sudo evtest, select keyboard, and then press G1-9

thank you for this output, it seems like your G1-9 keys are mapped to F1-F9, which actually was the case on my system as well, but that was, because I had `g810-led` installed.
Do you have any software installed, which does control the LEDs and could also be responsible for the F1-9 mapping?

microdou, 2021-09-09:
Well, I just installed `g810-led`, and your script works!
It appears that the outputs are exactly the same as Orion Spectrum. I think the only difference between the two versions of G910 is product ID

microdou, 2021-09-09:
I guess the solution will be:
1. Add a product variant to `src/devices/keyboard.py`
2. Add `g810-led` (or some of its dependencies) as dependency

zocker-160, 2021-09-09:
oh so it was `g810-led` which had a side effect I was using without knowing.....

yeah I will add your keyboard with the new ID to the devices, that should be simple, I will push an update in a few minutes.

But I do wonder though, if you still might encounter issues, since when installed from AUR, the background service does run in user mode and not as root, but it seems like on some systems, root is required, I need to find a solution for that.

As for point 2, I can do that, but I actually wanted to not be dependent on `g810-led`, I need to check what it is doing since it does remove the default mapping.

microdou, 2021-09-09:
Well, after installation of `g810-led-git`, `root` is not required to run your script anymore ...

I guess `g810-led` did some setting. I notice it added a udev rule `/etc/udev/rules.d/g810-led.rules`:
```
ACTION=="add", SUBSYSTEMS=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c336", MODE="666" RUN+="/usr/bin/g213-led -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEMS=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c330", MODE="666" RUN+="/usr/bin/g410-led -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEMS=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c33a", MODE="666" RUN+="/usr/bin/g413-led -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEMS=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c342", MODE="666" RUN+="/usr/bin/g512-led -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEMS=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c33c", MODE="666" RUN+="/usr/bin/g513-led -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEMS=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c333", MODE="666" RUN+="/usr/bin/g610-led -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEMS=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c338", MODE="666" RUN+="/usr/bin/g610-led -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEMS=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c331", MODE="666" RUN+="/usr/bin/g810-led -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEMS=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c337", MODE="666" RUN+="/usr/bin/g810-led -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEMS=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c33f", MODE="666" RUN+="/usr/bin/g815-led -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEMS=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c32b", MODE="666" RUN+="/usr/bin/g910-led -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEMS=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c335", MODE="666" RUN+="/usr/bin/g910-led -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEMS=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c339", MODE="666" RUN+="/usr/bin/gpro-led -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c336", MODE="666" RUN+="/usr/bin/g213-led -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c330", MODE="666" RUN+="/usr/bin/g410-led -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c33a", MODE="666" RUN+="/usr/bin/g413-led -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c342", MODE="666" RUN+="/usr/bin/g512-led -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c33c", MODE="666" RUN+="/usr/bin/g513-led -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c333", MODE="666" RUN+="/usr/bin/g610-led -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c338", MODE="666" RUN+="/usr/bin/g610-led -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c331", MODE="666" RUN+="/usr/bin/g810-led -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c337", MODE="666" RUN+="/usr/bin/g810-led -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c32b", MODE="666" RUN+="/usr/bin/g910-led -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c335", MODE="666" RUN+="/usr/bin/g910-led -p /etc/g810-led/profile"
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c339", MODE="666" RUN+="/usr/bin/gpro-led -p /etc/g810-led/profile"
```

microdou, 2021-09-09:
OK, new version correctly recognizes the keyboard. However, another issue occurs.

Looks like the app isn't able to show key names.

[image]

microdou, 2021-09-09:
It's likely just an issue of font color

zocker-160, 2021-09-09:
yes I think the background-color is here the problem, I will remove it for now until I have a proper color scheme for dark mode.

will push in a few minutes ;)

zocker-160, 2021-09-09:
ok jet another version -_-
0.1.8 should fix the issue in dark mode

microdou, 2021-09-09:
Works great! Now one more issue:

G keys are still mapped to F1-9. It looks like custom setting is not in effect, although app claims it's saved.

microdou, 2021-09-09:
Actually, app claims setting is unsaved upon every exit
[image]
, so I suppose setting is never saved.

zocker-160, 2021-09-09:
is the background service running?

`systemctl --user status keyboard-center.service`

microdou, 2021-09-09:
```
$ systemctl --user status keyboard-center.service                                                                                                                                               

× keyboard-center.service - background service for the Keyboard-Center application
     Loaded: loaded (/usr/lib/systemd/user/keyboard-center.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Thu 2021-09-09 12:34:50 EDT; 5min ago
    Process: 6762 ExecStart=python3 /usr/lib/keyboard-center/service.py /home/user (code=exited, status=1/FAILURE)
   Main PID: 6762 (code=exited, status=1/FAILURE)

Sep 09 12:34:50 os python3[6762]:     virtualKeyboard = uinput.Device(
Sep 09 12:34:50 os python3[6762]:   File "/usr/lib/python3.9/site-packages/uinput/__init__.py", line 178, in __init__
Sep 09 12:34:50 os python3[6762]:     self.__uinput_fd = fd or fdopen()
Sep 09 12:34:50 os python3[6762]:   File "/usr/lib/python3.9/site-packages/uinput/__init__.py", line 84, in fdopen
Sep 09 12:34:50 os python3[6762]:     return _libsuinput.suinput_open()
Sep 09 12:34:50 os python3[6762]:   File "/usr/lib/python3.9/site-packages/uinput/__init__.py", line 70, in _open_error_handler
Sep 09 12:34:50 os python3[6762]:     raise OSError(code, msg)
Sep 09 12:34:50 os python3[6762]: OSError: [Errno 19] Failed to open the uinput device: No such device
Sep 09 12:34:50 os systemd[2209]: keyboard-center.service: Main process exited, code=exited, status=1/FAILURE
Sep 09 12:34:50 os systemd[2209]: keyboard-center.service: Failed with result 'exit-code'.
```

zocker-160, 2021-09-09:
do you have the `uinput` kernel module loaded?

`modinfo uinput`

microdou, 2021-09-09:
seems so

```
$ modinfo uinput

filename:       /lib/modules/5.10.63-1-lts/kernel/drivers/input/misc/uinput.ko.xz
license:        GPL
description:    User level driver support for input subsystem
author:         Aristeu Sergio Rozanski Filho
alias:          devname:uinput
alias:          char-major-10-223
srcversion:     438F2017826496DD7B6AF01
depends:        
retpoline:      Y
intree:         Y
name:           uinput
vermagic:       5.10.63-1-lts SMP mod_unload
```

microdou, 2021-09-09:
```
$ journalctl --user -u keyboard-center.service

-- Boot ccfa95494bdb4fa9b2bdcb4267a17b1d --
Sep 09 12:47:47 os systemd[2563]: Started background service for the Keyboard-Center application.
Sep 09 12:47:47 os python3[2571]: DEBUG: Config file location: /home/user/.config/keyboard-center/settings.yml
Sep 09 12:47:47 os python3[2571]: DEBUG: loading config file /home/user/.config/keyboard-center/settings.yml
Sep 09 12:47:47 os python3[2571]: DEBUG: config loaded: {'mappings': {'MEMORY_1': {'MACRO_1': {'name': 'example-A', 'string': 'A', 'type': 'key', 'valu>
Sep 09 12:47:47 os python3[2571]: INFO: searching for supported keyboard...
Sep 09 12:47:47 os python3[2571]: INFO: keyboard found: Logitech G910 Orion Spark
Sep 09 12:47:47 os python3[2571]: INFO: saving deviceID into config
Sep 09 12:47:47 os python3[2571]: DEBUG: saving into config file /home/user/.config/keyboard-center/settings.yml
Sep 09 12:47:47 os python3[2571]: DEBUG: requesting USB endpoint...
Sep 09 12:47:47 os python3[2571]: DEBUG: time taken to find keyboard in ms: 0.05561518669128418
Sep 09 12:47:47 os python3[2571]: DEBUG: check and detach kernel driver if active
Sep 09 12:47:47 os python3[2571]: DEBUG: creating uinput device...
Sep 09 12:47:47 os python3[2571]: Traceback (most recent call last):
Sep 09 12:47:47 os python3[2571]:   File "/usr/lib/keyboard-center/service.py", line 286, in <module>
Sep 09 12:47:47 os python3[2571]:     main()
Sep 09 12:47:47 os python3[2571]:   File "/usr/lib/keyboard-center/service.py", line 242, in main
Sep 09 12:47:47 os python3[2571]:     virtualKeyboard = uinput.Device(
Sep 09 12:47:47 os python3[2571]:   File "/usr/lib/python3.9/site-packages/uinput/__init__.py", line 178, in __init__
Sep 09 12:47:47 os python3[2571]:     self.__uinput_fd = fd or fdopen()
Sep 09 12:47:47 os python3[2571]:   File "/usr/lib/python3.9/site-packages/uinput/__init__.py", line 84, in fdopen
Sep 09 12:47:47 os python3[2571]:     return _libsuinput.suinput_open()
Sep 09 12:47:47 os python3[2571]:   File "/usr/lib/python3.9/site-packages/uinput/__init__.py", line 70, in _open_error_handler
Sep 09 12:47:47 os python3[2571]:     raise OSError(code, msg)
Sep 09 12:47:47 os python3[2571]: OSError: [Errno 19] Failed to open the uinput device: No such device
Sep 09 12:47:48 os systemd[2563]: keyboard-center.service: Main process exited, code=exited, status=1/FAILURE
Sep 09 12:47:48 os systemd[2563]: keyboard-center.service: Failed with result 'exit-code'.
```

zocker-160, 2021-09-09:
it fails to create the uinput device, maybe that would need root on your system?
That could probably be the issue.

zocker-160, 2021-09-09:
you can test that by cloning this repo and just run the `service.py` inside the src folder manually

microdou, 2021-09-09:
Oh, you were right, `uinput` was not loaded.

`$ sudo modprobe uinput` resolves the issue!

zocker-160, 2021-09-09:
I hope it finally works now -_-, (and yes I know there is still a lot of work to do...)

If it works, then I cannot thank you enough for your patience, it was very very helpful!

microdou, 2021-09-09:
Thank you! You are actually the one who saved my keyboard after migrating to Linux. I will do further tests on functionality.

Hopefully there is an app for me to make use of G903 mouse & charging pad in future :)

microdou, 2021-09-09:
Just found out Solaar works for G903 !!!

closed 2021-09-09 by microdou

zocker-160, 2021-09-09:
yes for wireless mice Solaar is the software to get ;)

zocker-160, 2021-09-10:
just for the record (in case anyone else reads this issue):

I was able to reproduce the permission problem and the cause for that are indeed the udev rules!
Version 0.1.9 will feature proper udev rules ;)