zocker-160/keyboard-center #6: Project blocks RGB support

aquova, 2021-09-11:
I would like to say that I have been very pleased with this project, as it's been a great way to assign macros for this board.

However, I did notice that with this project installed, RGB support seems to be completely blocked. I tried both the [OpenRGB](https://openrgb.org/) and [keyleds](https://github.com/keyleds/keyleds) projects, which worked fine without this package, but when enabled neither are able to provide RGB support, OpenRGB is unable to even detect the keyboard.

zocker-160, 2021-09-11:
understood, I need to check, why this is the case, because I use an USB interface, which does not send and RGB data or information.

zocker-160, 2021-09-11:
I did some testing regarding this issue in combination with the RGB software you mentioned.

Sadly there is a fundamental issue going on, where in order to read from the USB device, you need to detach the Kernel driver and request access to the USB interface.

By doing so no other application can access those data at the same time until I release it.

When I use OpenRGB, it does claim the USB interface for itself and my application is not able to read any data anymore. Sadly this also is the case vice versa where, when I read from the interface, OpenRGB does not get and send any data anymore.

This is a fundamental issue which AFAIK is not solvable.

The only solution I could think of is maybe doing a colab with one of those other projects to offer both RGB and macro functions in one application.

zocker-160, 2021-09-12:
Update: 

I think this is solvable by using `libhidraw` instead of `libusb` for reading the incoming data. 

I have been testing with `libhidapi` in the past, which does not register any of the GKeys, but reading the raw data should work. Also `libhidraw` is be nonblocking and should solve the issue with other RGB software.

zocker-160, 2021-09-12:
YES I have a working prototype, I will push an update ASAP.

zocker-160, 2021-09-13:
Keyboard Center 0.1.10 is now compatible with OpenRGB!

You can now use both at the same time, @aquova would love to get feedback if that works for you or not.

commit e488c846 (2021-09-13): use libhidraw if possible and libusb only as backup

fixes #6:
- incompatibility issue with OpenRGB and other RGB software
closed 2021-09-13 by zocker-160