zocker-160/keyboard-center #7: Please add support for G815.

nickbuss, 2021-09-12:
I can test for you when you are ready.

zocker-160, 2021-09-12:
Currently I am working on fixing the other open issue, once that is done.

I will add instructions to the readme on how to help with adding a new keyboard. So if all goes well, that will be tomorrow.

Until then I can only thank in advance for your offer to help ;)

zocker-160, 2021-09-13:
I have added [instructions to the readme](https://github.com/zocker-160/keyboard-center#contribute-new-keyboard) on how to provide information for adding a new keyboard.

The USB ID for the G815 (046d : xc33f) (I hope that one is correct) is already set in the script, so you should be good to go. ;)

zocker-160, 2021-09-15:
closing in favor of #8

closed 2021-09-15 by zocker-160