zocker-160/keyboard-center #8: G815 Data

nickbuss, 2021-09-14:
Bus 001 Device 003: ID 046d:c33f Logitech, Inc. G815 RGB MECHANICAL GAMING KEYBOARD
[G815info.txt](https://github.com/zocker-160/keyboard-center/files/7161160/G815info.txt)
[G815capture.txt](https://github.com/zocker-160/keyboard-center/files/7161186/G815capture.txt)

Keypress sequence was M1 M2 M3 MR Game-Mode G1 G2 G3 G4 G5 REW PLAY/PAUSE FF MUTE VOL- VOL+

nickbuss, 2021-09-14:
After running the script my multimedia keys didn't work.  A reboot fix that though.

zocker-160, 2021-09-14:
thank you this data is looking good

> After running the script my multimedia keys didn't work. A reboot fix that though.

yes many keyboard do have the multimedia keys on the same usb interface than the Gkeys, keyboard-center will map those multimedia keys as well, so they should work.

zocker-160, 2021-09-14:
what is this Game-Mode button all about?

zocker-160, 2021-09-14:
I think I have most of the data figured out, but there is some, that I could not.

can you confirm that this comes up, when you press M1:
```
b'\x11\xff\x0b\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
b'\x11\xff\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
```
and for which button does
`b'\x11\xff\x11\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'`
come up?

nickbuss, 2021-09-15:
It appears that the keyboard generates different sequences depending upon if pressing an M key results in a mode change.
Starting from M1 active, pressing M1 M1 generates
```
got data from keyboard:b'\x11\xff\x0b\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x11\xff\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

got data from keyboard:b'\x11\xff\x0b\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x11\xff\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
```

but pressing M1 M2 generates
```
got data from keyboard:b'\x11\xff\x0b\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x11\xff\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

got data from keyboard:b'\x02\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x01\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x03\x00\x00\x00\x00'
got data from keyboard:b'\x11\xff\x0b\x00\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x11\xff\x11\x00\x00\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x11\xff\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
```
That sequence you queried is part of the sequence generated when entering M1.

Game mode is a function that temporarily disables certain preprogrammed keys (by default super and context) while active, but on closer inspection it appears that it doesn't generate a sequence in your script when pressed.

nickbuss, 2021-09-15:
Starting from M3 active, pressing M1 M1 M2 M2 M3 M3 M2 M1 M3 gives:
```
got data from keyboard:b'\x02\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x01\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x03\x00\x00\x00\x00'
got data from keyboard:b'\x11\xff\x0b\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x11\xff\x11\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x11\xff\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

got data from keyboard:b'\x11\xff\x0b\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x11\xff\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

got data from keyboard:b'\x02\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x01\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x03\x00\x00\x00\x00'
got data from keyboard:b'\x11\xff\x0b\x00\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x11\xff\x11\x00\x00\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x11\xff\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

got data from keyboard:b'\x11\xff\x0b\x00\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x11\xff\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

got data from keyboard:b'\x02\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x01\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x03\x00\x00\x00\x00'
got data from keyboard:b'\x11\xff\x0b\x00\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x11\xff\x11\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x11\xff\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

got data from keyboard:b'\x11\xff\x0b\x00\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x11\xff\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

got data from keyboard:b'\x02\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x01\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x03\x00\x00\x00\x00'
got data from keyboard:b'\x11\xff\x0b\x00\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x11\xff\x11\x00\x00\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x11\xff\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

got data from keyboard:b'\x02\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x01\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x03\x00\x00\x00\x00'
got data from keyboard:b'\x11\xff\x0b\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x11\xff\x11\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x11\xff\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

got data from keyboard:b'\x02\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x01\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x03\x00\x00\x00\x00'
got data from keyboard:b'\x11\xff\x0b\x00\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard:b'\x11\xff\x11\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from `keyboard:b'\x11\xff\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

```
I didn't see anything unique in there depending upon where you're jumping from, but that should give you all of the transitioning and non-transitioning sequences.

zocker-160, 2021-09-15:
perfect, thank you for the good data

zocker-160, 2021-09-15:
this is interesting, but I think I should be good by just ignoring the additional data, unless the keyboard expects me to send something back (which I doubt)

zocker-160, 2021-09-15:
@nickbuss 
version 0.1.11 released, which adds support for G815, it would be nice to get feedback, if it actually works.

nickbuss, 2021-09-15:
I'm getting some erroneous keystrokes in addition to the ones programed in as macros.
In M1, G1 is generating an additional f1 keypress but all other G-keys are good.
In M2 with G1-5 set to give a-e respectively, pressing all G-keys in sequence gives
1a2b3c4d5e
In M3 testing as per M2 gives
6a7b8c9d0e

zocker-160, 2021-09-15:
ah so the G2-5 of the M1 are working as expected, but the other Mkeys do still have this weird default mapping?

Could you provide the log of the background service?
`journalctl --user -u keyboard-center.service`

I fear this might be way more complicated to solve.

nickbuss, 2021-09-15:
Here your are.
[log.txt](https://github.com/zocker-160/keyboard-center/files/7170547/log.txt)

zocker-160, 2021-09-15:
this part is interesting
```
Sep 15 21:07:03 baldric python3[8087]: DEBUG: HIDraw disable endpoint found: /dev/hidraw1
Sep 15 21:07:03 baldric python3[8087]: DEBUG: HIDraw endpoint found: /dev/hidraw6
```

maybe the disable endpoint and the read endpoint need to be the same

can you try setting
`disableGKeysInterface = 1`
in `/usr/lib/keyboard-center/devices/keyboard.py` line 175

and then restart the background process `systemctl --user restart keyboard-center.service` and try again if that does change anything

nickbuss, 2021-09-16:
With that change they key sequence G1 G2 G3 under M1 generates two of backspaces at the start of the G3 sequence, but not under any other conditions that I could find ie. G3 only generates backspaces when preceded by those two keys.  Other than that no differences.
Updated log file. 
[log.txt](https://github.com/zocker-160/keyboard-center/files/7177055/log.txt)

zocker-160, 2021-09-16:
Thanks again for the log, thanks to @nirenjan I was made aware, that some keyboads need the disableGkey sequence as a HID feature report instead of the write function.

This change got pushed to version 0.1.12, maybe the G815 needs the same fix.

I don't know if the G815 needs `disableGKeysInterface = 1` or `disableGKeysInterface = 0`, so if you don't mind testing both again and see if that makes any difference. But I do fear though, that this keyboard might need some other special treatment in order to make it work properly.

nickbuss, 2021-09-16:
With this change none of the programmed macros are coming out, but I'm still getting all of the originally described erroneous output. eg M1 G1 only generates an f1, M2 G1 generates 1.  Behavior was the same for both settings of disableGKeysInterface.

Latest log file.
[log.txt](https://github.com/zocker-160/keyboard-center/files/7178032/log.txt)

zocker-160, 2021-09-16:
the fix does not work sadly, the function `hdev.send_feature_report` does fail in the log.

I am currently a bit out of ideas sadly :(.

I will be borrowing a G815 in the coming days though, so I hope I can find a solution to this, since I rally want this keyboard to work.

nirenjan, 2021-09-16:
One thing you can try - if you can capture the USB descriptors, there might be some useful information in there. That's what I did for my G710.

Run `sudo usbhid-dump -m 046d:c33f` and capture the descriptors. For example, this is the output for my G710:

```
$ sudo usbhid-dump -m 046d:c24d
003:043:001:DESCRIPTOR         1631801548.150974
 05 01 09 06 A1 01 05 07 19 00 29 E7 15 00 26 E7
 00 75 08 95 14 85 01 81 00 C0 05 0C 09 01 A1 01
 85 02 15 00 25 01 75 01 95 07 09 B5 09 B6 09 B7
 09 CD 09 E9 09 EA 09 E2 81 02 95 01 81 01 C0 06
 00 FF 09 00 A1 01 15 00 26 FF 00 75 08 09 03 85
 03 95 03 81 02 09 04 85 04 95 07 81 02 09 05 85
 05 15 00 26 FF 00 75 08 95 07 B1 02 09 06 85 06
 95 01 B1 02 09 07 85 07 95 04 B1 02 09 08 85 08
 95 03 B1 02 09 09 85 09 95 0C B1 02 09 80 85 80
 95 07 B1 02 C0

003:043:000:DESCRIPTOR         1631801548.153202
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 95 01 75 08 81 01 95 03 75 01
 05 08 19 01 29 03 91 02 95 05 75 01 91 01 95 06
 75 08 15 00 26 FF 00 05 07 19 00 2A FF 00 81 00
 C0

```

Copy the hex string for each descriptor and paste it into the [USB descriptor and request parser](https://eleccelerator.com/usbdescreqparser/), and parse as HID Report Descriptor. Copy and paste the output here, and I might be able to help out. Please also include the output of usbhid-dump, so I can match it to the different interfaces.

nickbuss, 2021-09-17:
Ok.
```
001:003:001:DESCRIPTOR         1631836807.674306
 05 01 09 06 A1 01 05 07 19 00 29 E7 15 00 26 E7
 00 75 08 95 14 85 01 81 00 C0 05 01 09 02 A1 01
 85 02 09 01 A1 00 05 09 19 01 29 10 15 00 25 01
 95 10 75 01 81 02 05 01 16 01 80 26 FF 7F 75 10
 95 02 09 30 09 31 81 06 15 81 25 7F 75 08 95 01
 09 38 81 06 05 0C 0A 38 02 95 01 81 06 C0 C0 05
 0C 09 01 A1 01 85 03 15 00 25 01 75 01 95 07 09
 B5 09 B6 09 B7 09 CD 09 E9 09 EA 09 E2 81 02 95
 01 81 01 C0 06 43 FF 0A 02 06 A1 01 85 11 75 08
 95 13 15 00 26 FF 00 09 02 81 00 09 02 91 00 C0
 06 43 FF 0A 04 06 A1 01 85 12 75 08 95 3F 15 00
 26 FF 00 09 04 81 00 09 04 91 00 C0

001:003:000:DESCRIPTOR         1631836807.678239
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 95 01 75 08 81 01 95 03 75 01
 05 08 19 01 29 03 91 02 95 05 75 01 91 01 95 06
 75 08 15 00 26 FF 00 05 07 19 00 2A FF 00 81 00
 C0
```
First block report:
```
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x06,        // Usage (Keyboard)
0xA1, 0x01,        // Collection (Application)
0x05, 0x07,        //   Usage Page (Kbrd/Keypad)
0x19, 0x00,        //   Usage Minimum (0x00)
0x29, 0xE7,        //   Usage Maximum (0xE7)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xE7, 0x00,  //   Logical Maximum (231)
0x75, 0x08,        //   Report Size (8)
0x95, 0x14,        //   Report Count (20)
0x85, 0x01,        //   Report ID (1)
0x81, 0x00,        //   Input (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x02,        // Usage (Mouse)
0xA1, 0x01,        // Collection (Application)
0x85, 0x02,        //   Report ID (2)
0x09, 0x01,        //   Usage (Pointer)
0xA1, 0x00,        //   Collection (Physical)
0x05, 0x09,        //     Usage Page (Button)
0x19, 0x01,        //     Usage Minimum (0x01)
0x29, 0x10,        //     Usage Maximum (0x10)
0x15, 0x00,        //     Logical Minimum (0)
0x25, 0x01,        //     Logical Maximum (1)
0x95, 0x10,        //     Report Count (16)
0x75, 0x01,        //     Report Size (1)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x01,        //     Usage Page (Generic Desktop Ctrls)
0x16, 0x01, 0x80,  //     Logical Minimum (-32767)
0x26, 0xFF, 0x7F,  //     Logical Maximum (32767)
0x75, 0x10,        //     Report Size (16)
0x95, 0x02,        //     Report Count (2)
0x09, 0x30,        //     Usage (X)
0x09, 0x31,        //     Usage (Y)
0x81, 0x06,        //     Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0x15, 0x81,        //     Logical Minimum (-127)
0x25, 0x7F,        //     Logical Maximum (127)
0x75, 0x08,        //     Report Size (8)
0x95, 0x01,        //     Report Count (1)
0x09, 0x38,        //     Usage (Wheel)
0x81, 0x06,        //     Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x0C,        //     Usage Page (Consumer)
0x0A, 0x38, 0x02,  //     Usage (AC Pan)
0x95, 0x01,        //     Report Count (1)
0x81, 0x06,        //     Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
[... 63 lines cut ...]
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x01,        //   Report Count (1)
0x75, 0x08,        //   Report Size (8)
0x81, 0x01,        //   Input (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x03,        //   Report Count (3)
0x75, 0x01,        //   Report Size (1)
0x05, 0x08,        //   Usage Page (LEDs)
0x19, 0x01,        //   Usage Minimum (Num Lock)
0x29, 0x03,        //   Usage Maximum (Scroll Lock)
0x91, 0x02,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x95, 0x05,        //   Report Count (5)
0x75, 0x01,        //   Report Size (1)
0x91, 0x01,        //   Output (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x95, 0x06,        //   Report Count (6)
0x75, 0x08,        //   Report Size (8)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x05, 0x07,        //   Usage Page (Kbrd/Keypad)
0x19, 0x00,        //   Usage Minimum (0x00)
0x2A, 0xFF, 0x00,  //   Usage Maximum (0xFF)
0x81, 0x00,        //   Input (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection

// 65 bytes
```

nirenjan, 2021-09-17:
So it looks like the interface 1 and write method are the right values for your keyboard. Beyond that though, the actual byte sequence is anybody's guess. I had to use a USB packet sniffer with the keyboard connected to a Windows VM to figure the byte sequences. It's been too long though, so I don't remember the exact software that I used.

That said, with a recent enough version of Linux, you should be able to capture the USB packets using debugfs or something similar, you still need to connect the keyboard to a Windows VM.

zocker-160, 2021-09-17:
> I had to use a USB packet sniffer with the keyboard connected to a Windows VM to figure the byte sequences.

Yes the same was needed for my G910, but I will try to get my hands on a G815 in the upcoming days in order to do that.

zocker-160, 2021-09-25:
0.1.14 got pushed and is now available

nickbuss, 2021-09-26:
Works good.  As noted the Mkeys leds aren't working.  The only other issue from my end is that the .deb left me with a non-responsive keyboard after doing an in-place upgrade, so I had to do a reboot to bring everything back up.

zocker-160, 2021-09-26:
thank you for the feedback, the keyboard freezing on upgrade is something I need to take a look at, it might be related to the old version though, so it should not happen anymore.

I will keep this issue open until the Mkeys are implemented

commit 1f5d110a (2021-09-27): switch MKey LEDs on G815 when switching profile

fixes #8
closed 2021-09-27 by zocker-160