zocker-160/keyboard-center #10: 'Unable to open device' upon log in

aquova, 2021-09-17:
Most of the times when I log into my computer upon rebooting, keyboard-center does not immediately function. Upon running `systemctl --user status keyboard-center`, the process has crashed with the following errors, possibly due to the process attempting to initialize before the keyboard is ready:

```
Sep 16 22:35:28 mars python3[1187]: ERROR: Task exception was never retrieved
Sep 16 22:35:28 mars python3[1187]: future: <Task finished name='Task-1' coro=<usbListener() done, defined at /usr/lib/keyboard-center/service.py:141> exception=HIDException('unable to open device')>
Sep 16 22:35:28 mars python3[1187]: Traceback (most recent call last):
Sep 16 22:35:28 mars python3[1187]:   File "/usr/lib/keyboard-center/service.py", line 153, in usbListener
Sep 16 22:35:28 mars python3[1187]:     await disableGkeyMapping(keyboardDev, HIDpath_disable)
Sep 16 22:35:28 mars python3[1187]:   File "/usr/lib/keyboard-center/service.py", line 53, in disableGkeyMapping
Sep 16 22:35:28 mars python3[1187]:     with HIDDevice(path=HIDpath) as hdev:
Sep 16 22:35:28 mars python3[1187]:   File "/usr/lib/keyboard-center/lib/hid.py", line 136, in __init__
Sep 16 22:35:28 mars python3[1187]:     raise HIDException('unable to open device')
Sep 16 22:35:28 mars python3[1187]: lib.hid.HIDException: unable to open device
```

Running `systemctl --user restart keyboard-center` is a workaround which allows the program to then function as intended.

zocker-160, 2021-09-17:
thanks for this report, I will add a startup delay and a retry count

zocker-160, 2021-09-17:
with version 0.1.13 the service will retry when opening the HID device fails. 

Current default is 5 times with a delay of 5 seconds each. The number of retries can be changed in the settings files.

zocker-160, 2021-09-20:
closing, please reopen if issue still exists

closed 2021-09-20 by zocker-160