zocker-160/keyboard-center #11: Service exits with error on G910

aurtzy, 2021-09-26:
Running this software on a relatively new install of EndeavourOS with a G910 Orion Spectrum. ```lsusb``` shows ```ID 046d:c335 Logitech, Inc. G910 Orion Spectrum Mechanical Keyboard```

After installing for the first time, everything worked fine until the next boot, in which my gkeys and mkeys stopped being recognized. This seems to have something to do with keyboard-center.service failing every time the software launches. Running ```systemctl --user status keyboard-center.service``` produces the following output:

```
Sep 26 15:35:42 AL-PC python3[3522]:     virtualKeyboard = uinput.Device(
Sep 26 15:35:42 AL-PC python3[3522]:   File "/usr/lib/python3.9/site-packages/uinput/__init__.py", line 178, in __init__
Sep 26 15:35:42 AL-PC python3[3522]:     self.__uinput_fd = fd or fdopen()
Sep 26 15:35:42 AL-PC python3[3522]:   File "/usr/lib/python3.9/site-packages/uinput/__init__.py", line 84, in fdopen
Sep 26 15:35:42 AL-PC python3[3522]:     return _libsuinput.suinput_open()
Sep 26 15:35:42 AL-PC python3[3522]:   File "/usr/lib/python3.9/site-packages/uinput/__init__.py", line 70, in _open_error_handler
Sep 26 15:35:42 AL-PC python3[3522]:     raise OSError(code, msg)
Sep 26 14:41:57 AL-PC python3[586]: OSError: [Errno 19] Failed to open the uinput device: No such device
Sep 26 15:35:42 AL-PC systemd[577]: keyboard-center.service: Main process exited, code=exited, status=1/FAILURE
Sep 26 15:35:42 AL-PC systemd[577]: keyboard-center.service: Failed with result 'exit-code'.
```

Some things I have tried to no avail include:
- Reinstalling. The service started working again after the first install, but stopped working again after a restart. The next ~5 installs could not reproduce this, and I have not been able to get a successful service start since the first reinstall.
- ```modprobe uinput``` changes the service status output on one line, from ```OSError: [Errno 19] Failed to open the uinput device: No such device``` to ```PermissionError: [Errno 13] Failed to open the uinput device: Permission denied```. The service fails either way.

zocker-160, 2021-09-26:
thanks for the report, could you please try the following:

- `systemctl --user stop keyboard-center.service`
- `systemctl --user disable keyboard-center.service´
- reboot system
- after reboot wait a bit to make sure everything is started up properly
- that the service again `systemctl --user start keyboard-center.service`

I need to make sure, that this is actually a permission error and not an issue that only appears on boot / autostart when the uinput device is not ready.

aurtzy, 2021-09-26:
Running ```systemctl --user status keyboard-center.service``` after doing this seems to yield the first error:  

```
Sep 26 17:21:42 AL-PC python3[1394]:     virtualKeyboard = uinput.Device(
Sep 26 17:21:42 AL-PC python3[1394]:   File "/usr/lib/python3.9/site-packages/uinput/__init__.py", line 178, in __init__
Sep 26 17:21:42 AL-PC python3[1394]:     self.__uinput_fd = fd or fdopen()
Sep 26 17:21:42 AL-PC python3[1394]:   File "/usr/lib/python3.9/site-packages/uinput/__init__.py", line 84, in fdopen
Sep 26 17:21:42 AL-PC python3[1394]:     return _libsuinput.suinput_open()
Sep 26 17:21:42 AL-PC python3[1394]:   File "/usr/lib/python3.9/site-packages/uinput/__init__.py", line 70, in _open_error_handler
Sep 26 17:21:42 AL-PC python3[1394]:     raise OSError(code, msg)
Sep 26 17:21:42 AL-PC python3[1394]: OSError: [Errno 19] Failed to open the uinput device: No such device
Sep 26 17:21:42 AL-PC systemd[581]: keyboard-center.service: Main process exited, code=exited, status=1/FAILURE
Sep 26 17:21:42 AL-PC systemd[581]: keyboard-center.service: Failed with result 'exit-code'.
```
I was reading a somewhat similar issue on a [stack overflow post](https://stackoverflow.com/questions/48640935/oserror-errno-19-failed-to-open-the-uinput-device-no-such-device) earlier and according to it, ```lsmod | grep uinput``` should return a uinput entry if the driver is loaded, but this only happens if I run ```modprobe uinput``` in the same session and rebooting will cause it to return nothing again. Is this significant to what might be happening?

zocker-160, 2021-09-26:
> Is this significant to what might be happening?

yes to me this looks exactly like the issue some other user had: https://github.com/zocker-160/keyboard-center/issues/5#issuecomment-916272122

this issue is caused by the uinput kernel module not being loaded and the solution was to run `sudo modprobe uinput`.

It is weird though, that it does not seem to persist over reboots, did you run it as root?

aurtzy, 2021-09-27:
Yes, I have been running ```sudo modprobe uinput``` in these situations. I can't recall anything I've done that would cause uinput to not persist (especially since I installed this distro last week) but if I do I'll definitely comment on it. My issue seems to be exactly the same as the one you mentioned, just that my end result is a "permission denied" error instead of a service start when running ```sudo modprobe input``` beforehand.

zocker-160, 2021-09-27:
I think this problem might be related to a missing udev rule, which is apparently not needed on all systems.

could you please try adding the following line to `/usr/lib/udev/rules.d/60-keyboard-center.rules`:

`KERNEL=="uinput", MODE="0660", GROUP="users", OPTIONS+="static_node=uinput", TAG+="uaccess"`

and after that reboot with the same steps as above, disable service, stop it and then start manually and see, if you still get permission denied.

aurtzy, 2021-09-27:
It works! The permission denied error does not show up anymore, but it seems I still have to manually run ```sudo modprobe uinput``` after a reboot in order for the service to not give me a "Failed to open the uinput device" error.

zocker-160, 2021-09-27:
ok good, I will add this line to the udev rule by default then

Regarding the kernel module, we would need to find out, why it unloads itself despite being manually enabled. Maybe there are some GRUB options needed on EndeavourOS

zocker-160, 2021-09-27:
@aurtzy with the new udev rule in place, can you try to run `modprobe uinput` _without_ root and see if that works?

If yes, then I can also run this from the background service on startup as a workaround in case the kernel module is not loaded.

aurtzy, 2021-09-27:
Can do.

Just rebooted. Running ```modprobe uinput``` without root throws a permission error.

```cat /boot/config-$(uname -r) | grep CONFIG_INPUT_UINPUT``` returns ```cat: /boot/config-5.14.7-arch1-1: No such file or directory```

zocker-160, 2021-09-27:
> Running modprobe uinput without root throws a permission error.

:(

Do you have any entry inside `/boot/config-*`? maybe it is called differently to what `uname -r` returns

aurtzy, 2021-09-27:
I found a workaround from [this](https://unix.stackexchange.com/questions/71064/systemd-automate-modprobe-command-at-boot-time#71069) stackoverflow post! Adding a uinput.conf to /etc/modules-load.d with ```uinput``` in the file resolves the issue when I reboot.  

I'd be glad to keep helping with debugging the issue if this is not satisfactory enough, though, and as a side note I'm open to assisting with development wherever I'm capable if you ever need to try stuff on another user's machine.

If we're still continuing this - I just entered the /boot directory, and I'm not actually seeing any config-* folder.

zocker-160, 2021-09-27:
> If we're still continuing this - I just entered the /boot directory, and I'm not actually seeing any config-* folder.

Understood, nvm then, it seems like that EndeavourOS just works different in that regard.

> I found a workaround from this stackoverflow post! Adding a uinput.conf to /etc/modules-load.d with uinput in the file resolves the issue when I reboot.

Oh perfect, I will be adding this to the packaging then.

> I'm open to assisting with development wherever I'm capable if you ever need to try stuff on another user's machine.

Running the latest version and opening issues is already a huge help for me, thank you for doing that.

zocker-160, 2021-09-27:
I just released a new version with the new rules added to the packaging, feedback if that works for you would be appreciated.

zocker-160, 2021-09-27:
nice great to hear, I will close this issue then, thanks for the help :+1:

closed 2021-09-27 by zocker-160