zocker-160/keyboard-center #13: Service fails to start with ModuleNotFoundError for openrgb

aurtzy, 2021-10-01:
Just updated to 0.1.18-testing. Service seems to be failing due to openrgb not being found:
```
Oct 01 18:00:26 AL-PC systemd[577]: Started background service for the Keyboard-Center application.
Oct 01 18:00:26 AL-PC python3[586]: Traceback (most recent call last):
Oct 01 18:00:26 AL-PC python3[586]:   File "/usr/lib/keyboard-center/service.py", line 23, in <module>
Oct 01 18:00:26 AL-PC python3[586]:     from lib.openrgb.orgb import OpenRGBClient
Oct 01 18:00:26 AL-PC python3[586]:   File "/usr/lib/keyboard-center/lib/openrgb/__init__.py", line 3, in <module>
Oct 01 18:00:26 AL-PC python3[586]:     from openrgb import utils
Oct 01 18:00:26 AL-PC python3[586]: ModuleNotFoundError: No module named 'openrgb'
Oct 01 18:00:26 AL-PC systemd[577]: keyboard-center.service: Main process exited, code=exited, status=1/FAILURE
Oct 01 18:00:26 AL-PC systemd[577]: keyboard-center.service: Failed with result 'exit-code'.
```  

This was checked after I had restarted the service, and rebooting the system does not fix the issue. OpenRGB is not installed.

zocker-160, 2021-10-01:
ooh I messed the import up, sorry about that!

will fix asap

zocker-160, 2021-10-01:
should be fixed in 0.1.19

commit 5d95c93f (2021-10-01): fix openrgb imports

stupid me messed this up completely
fixes #13
closed 2021-10-01 by zocker-160