zocker-160/keyboard-center #14: Ability to hold down G-keys

aurtzy, 2021-11-01:
Currently, pressing a G-key will always trigger its respective macro, which means that even if there is one key assigned, it's treated as a quick press rather than staying held down until the user releases the key, making holding down G-keys not possible. Is this feature something that might be planned for the future?

zocker-160, 2021-11-01:
I have not planned this feature, but it is possible to implement, so I will put it on the TODO.

zocker-160, 2023-03-17:
I know this issue has been open for a very long time and you probably forgot about it, but I finally found the time to work on this and implement it. So there you go. :laughing:

closed 2023-03-17 by zocker-160

aurtzy, 2023-03-19:
Sweet! I completely understand and appreciate all the work you've already done on this software.

Looks to function just fine on my end. Cheers.

zocker-160, 2023-03-19:
Nice glad to hear!

commit f70fc6c8 (2023-03-17): implement ability to hold down G-Keys

closes #14