zocker-160/keyboard-center #15: Shortcuts without Meta, Alt, Ctrl or Shift

xerhaxs, 2021-11-01:
Maybe I just can't find the option, but in my opinion the option to create shortcuts that don't start with Meta, Alt, Ctrl or Shift is missing. For example, you can't put F3+N as a key combination on a G-key. 
Would be a useful extension in my opinion.

xerhaxs, 2021-11-01:
Sorry, I think I expressed myself a little misleadingly.
My point is not that the N is an uppercase or lowercase letter, but that you can't create a working shortcut from the combination of F3 and N, or F3 and n.
Using `xinput test-xi2 [device]` I found out that e.g. the shortcut Meta+N
works in a way that Meta is pressed and held, then N is pressed, held and released after a very short moment and then Meta is released. With F3 and N or any combination that does not start with Meta, Alt, Shift or Ctrl, F3 is pressed and released and then N is pressed and released, so that F3 and N are never pressed at the same time. This creates the problem that you can't put many key combinations on the G keys at all.

zocker-160, 2021-11-01:
Ah yes I think I understand how yeah that is a limitation of the current approach.

It is fixable though, I will put it on my TODO, thanks for bringing this up.

zocker-160, 2021-12-17:
@xerhaxs 0.1.20 released which implements this feature. It took a bit longer sadly, since it was more work than I originally thought.

Would be nice if you could give some feedback on if it works for you and if that is what you were asking for ;).

xerhaxs, 2021-12-19:
It works, but somehow it doesn't. Thanks a lot for adding this function and in principle it works well. When I test it via xinput, the correct sequence of pressed keys is also output. Actually, I wanted to use this function now for a shortcut in Minecraft, however, it is not recognized there - as before. I don't know why, but I think that this is due to Minecraft and not the Keyboard Center.

zocker-160, 2021-12-19:
ah yeah games will have a hard time recognizing those shortcuts, mainly because they happen too fast. From my experience (it changes from game to game) a delay of up to 50ms is needed.

I think the best way to solve this is by adding an option into the UI, which slows the execution of the shortcut down to make it usable in games.

zocker-160, 2021-12-19:
I opened a new issue #17 , since it is not related to custom modifier buttons.

closed 2021-12-19 by zocker-160