zocker-160/keyboard-center #16: Crashes on Start under Arch

poiu477, 2021-11-29:
When launching from terminal I get these errors:
Icon theme "gnome" not found.
service enabled?
service running?
starting service...
service needs reload?
Traceback (most recent call last):
File "/usr/lib/keyboard-center/main.py", line 150, in readConfiguration
_did = self.configparser.getSettings()["usbDeviceID"]
File "/usr/lib/keyboard-center/lib/configparser.py", line 50, in getSettings
return self.settings["settings"]
TypeError: 'NoneType' object is not subscriptable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "/usr/lib/keyboard-center/main.py", line 374, in <module>
main()
File "/usr/lib/keyboard-center/main.py", line 367, in main
window = MainWindow(app, devmode)
File "/usr/lib/keyboard-center/main.py", line 118, in __init__
self.readConfiguration()
File "/usr/lib/keyboard-center/main.py", line 154, in readConfiguration
f"No supported keyboard found! :(\n(err:{_did})",
UnboundLocalError: local variable '_did' referenced before assignment

poiu477, 2021-11-29:
I want to add that I get a notification that says my G910 is found but no window opens

zocker-160, 2021-11-29:
this error points to the application not being able to find and write the configuration file.

could you check if you have any file in `~/.config/keyboard-center` and if yes provide the contents and permissions set on the containing file?

poiu477, 2021-11-29:
It’s an empty file called settings.yml with permissions rw-r—r— 0 and owned
by user

zocker-160, 2021-11-29:
ok yep the config file being empty is the issue here, could you please provide the log of the background service?

`journalctl --user -u keyboard-center.service`

poiu477, 2021-11-29:
Sure, here it is.
-- Journal begins at Fri 2021-11-26 12:32:56 EST, ends at Mon 2021-11-29
06:35:25 EST. --
Nov 27 04:49:11 paulusesarch systemd[3538]: Started background service for
the Keyboard-Center application.
Nov 27 04:49:11 paulusesarch python3[3547]: DEBUG: Config file location:
/home/paulh/.config/keyboard-center/>
Nov 27 04:49:11 paulusesarch python3[3547]: DEBUG: loading config file
/home/paulh/.config/keyboard-center/se>
Nov 27 04:49:11 paulusesarch python3[3547]: DEBUG: config loaded: None
Nov 27 04:49:11 paulusesarch python3[3547]: INFO: searching for supported
keyboard...
Nov 27 04:49:11 paulusesarch python3[3547]: INFO: keyboard found: Logitech
G910 Orion Spectrum
Nov 27 04:49:11 paulusesarch python3[3547]: INFO: saving deviceID into
config
Nov 27 04:49:11 paulusesarch python3[3547]: Traceback (most recent call
last):
Nov 27 04:49:11 paulusesarch python3[3547]:   File
"/usr/lib/keyboard-center/service.py", line 435, in <modul>
Nov 27 04:49:11 paulusesarch python3[3547]:     main()
Nov 27 04:49:11 paulusesarch python3[3547]:   File
"/usr/lib/keyboard-center/service.py", line 317, in main
Nov 27 04:49:11 paulusesarch python3[3547]:
    config.settings["settings"]["usbDeviceID"] = i
Nov 27 04:49:11 paulusesarch python3[3547]: TypeError: 'NoneType' object is
not subscriptable
Nov 27 04:49:11 paulusesarch systemd[3538]: keyboard-center.service: Main
process exited, code=exited, status>
Nov 27 04:49:11 paulusesarch systemd[3538]: keyboard-center.service: Failed
with result 'exit-code'.
-- Boot 8e2c75b3b7e44f1fbcc8bc9cdf4bb039 --
Nov 27 05:59:19 paulusesarch systemd[3552]: Started background service for
the Keyboard-Center application.
Nov 27 05:59:19 paulusesarch python3[3561]: DEBUG: Config file location:
/home/paulh/.config/keyboard-center/>
Nov 27 05:59:19 paulusesarch python3[3561]: DEBUG: loading config file
/home/paulh/.config/keyboard-center/se>
Nov 27 05:59:19 paulusesarch python3[3561]: DEBUG: config loaded: None
Nov 27 05:59:19 paulusesarch python3[3561]: INFO: searching for supported
keyboard...
Nov 27 05:59:19 paulusesarch python3[3561]: INFO: keyboard found: Logitech
G910 Orion Spectrum
Nov 27 05:59:19 paulusesarch python3[3561]: INFO: saving deviceID into
config
Nov 27 05:59:19 paulusesarch python3[3561]: Traceback (most recent call
last):
Nov 27 05:59:19 paulusesarch python3[3561]:   File
"/usr/lib/keyboard-center/service.py", line 435, in <modul>
Nov 27 05:59:19 paulusesarch python3[3561]:     main()
Nov 27 05:59:19 paulusesarch python3[3561]:   File
"/usr/lib/keyboard-center/service.py", line 317, in main
Nov 27 05:59:19 paulusesarch python3[3561]:
    config.settings["settings"]["usbDeviceID"] = i
Nov 27 05:59:19 paulusesarch python3[3561]: TypeError: 'NoneType' object is
not subscriptable
Nov 27 05:59:20 paulusesarch systemd[3552]: keyboard-center.service: Main
process exited, code=exited, status>
Nov 27 05:59:20 paulusesarch systemd[3552]: keyboard-center.service: Failed
with result 'exit-code'.
-- Boot 1a237580fb7b4dbf8d371bc1dc740dc8 --
Nov 27 17:23:01 paulusesarch systemd[3847]: Started background service for
the Keyboard-Center application.
Nov 27 17:23:01 paulusesarch python3[3856]: DEBUG: Config file location:
/home/paulh/.config/keyboard-center/>
Nov 27 17:23:01 paulusesarch python3[3856]: DEBUG: loading config file
/home/paulh/.config/keyboard-center/se>
Nov 27 17:23:01 paulusesarch python3[3856]: DEBUG: config loaded: None
Nov 27 17:23:01 paulusesarch python3[3856]: INFO: searching for supported
keyboard...
[... 1197 lines cut ...]
Nov 29 01:58:04 paulusesarch systemd[3969]: keyboard-center.service: Main
process exited, code=exited, status>
Nov 29 01:58:04 paulusesarch systemd[3969]: keyboard-center.service: Failed
with result 'exit-code'.
Nov 29 01:58:23 paulusesarch systemd[3969]: Started background service for
the Keyboard-Center application.
Nov 29 01:58:23 paulusesarch python3[78441]: DEBUG: Config file location:
/home/paulh/.config/keyboard-center>
Nov 29 01:58:23 paulusesarch python3[78441]: DEBUG: loading config file
/home/paulh/.config/keyboard-center/s>
Nov 29 01:58:23 paulusesarch python3[78441]: DEBUG: config loaded: None
Nov 29 01:58:23 paulusesarch python3[78441]: INFO: searching for supported
keyboard...
Nov 29 01:58:23 paulusesarch python3[78441]: INFO: keyboard found: Logitech
G910 Orion Spectrum
Nov 29 01:58:23 paulusesarch python3[78441]: INFO: saving deviceID into
config
Nov 29 01:58:23 paulusesarch python3[78441]: Traceback (most recent call
last):
Nov 29 01:58:23 paulusesarch python3[78441]:   File
"/usr/lib/keyboard-center/service.py", line 435, in <modu>
Nov 29 01:58:23 paulusesarch python3[78441]:     main()
Nov 29 01:58:23 paulusesarch python3[78441]:   File
"/usr/lib/keyboard-center/service.py", line 317, in main
Nov 29 01:58:23 paulusesarch python3[78441]:

zocker-160, 2021-11-29:
thanks for this, yeah so it is clear that the config file location does not get picked up correctly for some reason.

It tries to load from `.config/keyboard-center/s` for some reason.

could you lastly also provide your Python version `python --version`. I will try to reproduce this issue.

poiu477, 2021-11-29:
Sure thing. Appreciate it

python --version
Python 3.9.9

zocker-160, 2021-11-29:
ok I think the issue is that the copying of the template config failed

could you please try to manually copy it over and see if that solves the issue?

check that `/usr/lib/keyboard-center/config/testconfig-example.yaml` is not empty (if yes than that is also an issue)
and then use the following commands:

```bash
systemctl --user stop keyboard-center.service
rm ~/.config/keyboard-center/settings.yml
cp /usr/lib/keyboard-center/config/testconfig-example.yaml ~/.config/keyboard-center/settings.yml
```
then make sure the copied settings file is owned by user and has permissions set to 644

start the background service again

poiu477, 2021-11-29:
It works!

zocker-160, 2021-11-29:
nice :ok_hand: 

I will put a check and fix in place for this scenario, where the copy of the initial configuration fails.

So pls keep this issue open.

poiu477, 2021-11-29:
will do thanks man, appreciate it!

commit 15a7e00c (2021-11-29): add config file integrity check

fixes #16
closed 2021-11-29 by zocker-160