zocker-160/keyboard-center #17: Ability to slow execution of shortcuts down

zocker-160, 2021-12-19:
Some games do not recognize shortcuts, when they get executed really fast.

An option in the UI to slow it down ("game mode") would be a good solution I think.

zocker-160, 2021-12-27:
I just pushed 0.1.21 adding a 50ms delay when "GameMode" is enabled.

@xerhaxs would you mind testing and check if it makes any difference?

xerhaxs, 2022-01-01:
Sorry I'm just now giving feedback, but I've been on vacation. Well the gamemode works partially - sometimes it works, sometimes not. I could imagine that this is because the 50ms is just too much or too little. I think it would be useful to have the option to choose the delay itself, so that the G-Keys work for each program if you adjust it a little.
Other than that, happy new year.

zocker-160, 2022-01-01:
thanks for the feedback, I think the option to be able to set the delay in the UI makes sense, I will think about how to implement that the best way.

>  Other than that, happy new year.

Thanks and I of course wish you the same :ok_hand: :fireworks: !

zocker-160, 2022-01-05:
ok so I pushed 0.1.23, which gives you the ability to set the delay for each button individually @xerhaxs .

Requesting feedback if that works for you as expected.

xerhaxs, 2022-01-06:
Yes, it works now. It took me quite a while to find the right delay for Minecraft, but now it works without problems. Thanks.
(Minecraft needs a delay of 620ms - for whatever reason. On Windows it works without 620ms delay).

zocker-160, 2022-01-06:
Oh wow 620ms is really high, how odd.

> On Windows it works without 620ms delay

Well not quite true, the Logitech driver on Windows does also add delays, it even adds game specific delays without you noticing it, since it does it all for you automatically.

Great to hear it works now for you - closing this issue then.

closed 2022-01-06 by zocker-160