zocker-160/keyboard-center #18: Crashes on Volume Roller Usage with G910

poiu477, 2021-12-27:
When I adjust video volume with the roller the whole service crashes and I have to start it again.

poiu477, 2021-12-27:
so after further investigation it seems to happen intermittently when multiple inputs are sent at once

poiu477, 2021-12-27:
I can consistently replicate it by repeatedly cycling the memory keys

poiu477, 2021-12-27:
but it has happened in normal use as well

zocker-160, 2021-12-27:
could you please provide the entire log of the background service?

poiu477, 2021-12-28:
-- Journal begins at Mon 2021-12-27 03:30:39 EST, ends at Mon 2021-12-27 22:03:03 EST. --
Dec 27 05:45:43 paulusesarch python3[3312314]: DEBUG: HIDerror: probably exiting? (Success)
Dec 27 05:45:43 paulusesarch python3[3312314]: DEBUG: HIDerror: probably exiting? (Success)
Dec 27 05:45:43 paulusesarch python3[3312314]: DEBUG: HIDerror: probably exiting? (Success)
Dec 27 05:45:43 paulusesarch python3[3312314]: DEBUG: HIDerror: probably exiting? (Success)
Dec 27 05:45:43 paulusesarch python3[3312314]: DEBUG: HIDerror: probably exiting? (Success)
Dec 27 05:45:43 paulusesarch python3[3312314]: DEBUG: HIDerror: probably exiting? (Success)
Dec 27 05:45:43 paulusesarch python3[3312314]: DEBUG: HIDerror: probably exiting? (Success)
Dec 27 05:45:43 paulusesarch python3[3312314]: DEBUG: HIDerror: probably exiting? (Success)
Dec 27 05:45:43 paulusesarch python3[3312314]: DEBUG: HIDerror: probably exiting? (Success)
Dec 27 05:45:43 paulusesarch python3[3312314]: DEBUG: HIDerror: probably exiting? (Success)
Dec 27 05:45:43 paulusesarch python3[3312314]: DEBUG: HIDerror: probably exiting? (Success)
Dec 27 05:45:43 paulusesarch python3[3312314]: DEBUG: HIDerror: probably exiting? (Success)
Dec 27 05:45:43 paulusesarch python3[3312314]: ERROR: Task exception was never retrieved
Dec 27 05:45:43 paulusesarch python3[3312314]: future: <Task finished name='Task-1' coro=<usbListener() done,>
Dec 27 05:45:43 paulusesarch python3[3312314]: Traceback (most recent call last):
Dec 27 05:45:43 paulusesarch python3[3312314]:   File "/usr/lib/keyboard-center/service.py", line 213, in usb>
Dec 27 05:45:43 paulusesarch python3[3312314]:     fromKeyboard = hdev.read(
Dec 27 05:45:43 paulusesarch python3[3312314]:   File "/usr/lib/keyboard-center/lib/hid.py", line 171, in read
Dec 27 05:45:43 paulusesarch python3[3312314]:     size = self.__hidcall(
Dec 27 05:45:43 paulusesarch python3[3312314]:   File "/usr/lib/keyboard-center/lib/hid.py", line 154, in __h>
Dec 27 05:45:43 paulusesarch python3[3312314]:     raise HIDException(err)
Dec 27 05:45:43 paulusesarch python3[3312314]: lib.hid.HIDException: Success
Dec 27 05:45:43 paulusesarch python3[3312314]: During handling of the above exception, another exception occu>
Dec 27 05:45:43 paulusesarch python3[3312314]: Traceback (most recent call last):
Dec 27 05:45:43 paulusesarch python3[3312314]:   File "/usr/lib/python3.10/asyncio/base_events.py", line 595,>
Dec 27 05:45:43 paulusesarch python3[3312314]:     self._run_once()
Dec 27 05:45:43 paulusesarch python3[3312314]:   File "/usr/lib/python3.10/asyncio/base_events.py", line 1881>
Dec 27 05:45:43 paulusesarch python3[3312314]:     handle._run()
Dec 27 05:45:43 paulusesarch python3[3312314]:   File "/usr/lib/python3.10/asyncio/events.py", line 80, in _r>
Dec 27 05:45:43 paulusesarch python3[3312314]:     self._context.run(self._callback, *self._args)
Dec 27 05:45:43 paulusesarch python3[3312314]:   File "/usr/lib/keyboard-center/service.py", line 224, in usb>
Dec 27 05:45:43 paulusesarch python3[3312314]:     sys.exit(1)
Dec 27 05:45:43 paulusesarch python3[3312314]: SystemExit: 1
Dec 27 05:45:43 paulusesarch systemd[4451]: keyboard-center.service: Main process exited, code=exited, status>
Dec 27 05:45:43 paulusesarch systemd[4451]: keyboard-center.service: Failed with result 'exit-code'.
Dec 27 05:45:43 paulusesarch systemd[4451]: keyboard-center.service: Consumed 13min 14.871s CPU time.
Dec 27 05:46:08 paulusesarch systemd[4451]: Started background service for the Keyboard-Center application.
Dec 27 05:46:08 paulusesarch python3[3353743]: INFO: ------------ starting -------------
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: Config file location: /home/paulh/.config/keyboard-cent>
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: loading config file /home/paulh/.config/keyboard-center>
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: config loaded: {'mappings': {'MEMORY_1': {'MACRO_1': {'>
Dec 27 05:46:08 paulusesarch python3[3353743]: INFO: searching for supported keyboard...
Dec 27 05:46:08 paulusesarch python3[3353743]: INFO: keyboard found: Logitech G910 Orion Spectrum
Dec 27 05:46:08 paulusesarch python3[3353743]: INFO: saving deviceID into config
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: saving into config file /home/paulh/.config/keyboard-ce>
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: time taken to find keyboard in ms: 0.022220373153686523
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: requesting USB endpoint...
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: Searching for HIDraw endpoint...
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: HIDraw disable endpoint found: /dev/hidraw7
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: HIDraw read endpoint found: /dev/hidraw8
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: HIDraw read endpoint found: /dev/hidraw8
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: HIDraw read endpoint found: /dev/hidraw8
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: HIDraw read endpoint found: /dev/hidraw8
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: Checking for HID availability...
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: Connected to /dev/hidraw8
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: creating uinput device...
Dec 27 05:46:08 paulusesarch python3[3353743]: INFO: starting service...
[... cut ...]
Dec 27 17:41:58 paulusesarch python3[3356047]:   File "/usr/lib/keyboard-center/service.py", line 224, in usb>
Dec 27 17:41:58 paulusesarch python3[3356047]:     sys.exit(1)
Dec 27 17:41:58 paulusesarch python3[3356047]: SystemExit: 1
Dec 27 17:41:58 paulusesarch systemd[4451]: keyboard-center.service: Main process exited, code=exited, status>
Dec 27 17:41:58 paulusesarch systemd[4451]: keyboard-center.service: Failed with result 'exit-code'.
Dec 27 17:41:58 paulusesarch systemd[4451]: keyboard-center.service: Consumed 24min 27.504s CPU time.

poiu477, 2021-12-28:
I noticed that during it, it was cycling my rgb profile even on my motherboard controlled lights, it is the same program

poiu477, 2021-12-28:
I also looked away and just noticed it JUST crashed here is log 

-- Journal begins at Mon 2021-12-27 04:26:44 EST, ends at Mon 2021-12-27 22:35:51 EST. --
Dec 27 05:45:43 paulusesarch python3[3312314]: DEBUG: HIDerror: probably exiting? (Success)
Dec 27 05:45:43 paulusesarch python3[3312314]: DEBUG: HIDerror: probably exiting? (Success)
Dec 27 05:45:43 paulusesarch python3[3312314]: DEBUG: HIDerror: probably exiting? (Success)
Dec 27 05:45:43 paulusesarch python3[3312314]: DEBUG: HIDerror: probably exiting? (Success)
Dec 27 05:45:43 paulusesarch python3[3312314]: DEBUG: HIDerror: probably exiting? (Success)
Dec 27 05:45:43 paulusesarch python3[3312314]: DEBUG: HIDerror: probably exiting? (Success)
Dec 27 05:45:43 paulusesarch python3[3312314]: DEBUG: HIDerror: probably exiting? (Success)
Dec 27 05:45:43 paulusesarch python3[3312314]: DEBUG: HIDerror: probably exiting? (Success)
Dec 27 05:45:43 paulusesarch python3[3312314]: DEBUG: HIDerror: probably exiting? (Success)
Dec 27 05:45:43 paulusesarch python3[3312314]: DEBUG: HIDerror: probably exiting? (Success)
Dec 27 05:45:43 paulusesarch python3[3312314]: DEBUG: HIDerror: probably exiting? (Success)
Dec 27 05:45:43 paulusesarch python3[3312314]: DEBUG: HIDerror: probably exiting? (Success)
Dec 27 05:45:43 paulusesarch python3[3312314]: ERROR: Task exception was never retrieved
Dec 27 05:45:43 paulusesarch python3[3312314]: future: <Task finished name='Task-1' coro=<usbListener() done,>
Dec 27 05:45:43 paulusesarch python3[3312314]: Traceback (most recent call last):
Dec 27 05:45:43 paulusesarch python3[3312314]:   File "/usr/lib/keyboard-center/service.py", line 213, in usb>
Dec 27 05:45:43 paulusesarch python3[3312314]:     fromKeyboard = hdev.read(
Dec 27 05:45:43 paulusesarch python3[3312314]:   File "/usr/lib/keyboard-center/lib/hid.py", line 171, in read
Dec 27 05:45:43 paulusesarch python3[3312314]:     size = self.__hidcall(
Dec 27 05:45:43 paulusesarch python3[3312314]:   File "/usr/lib/keyboard-center/lib/hid.py", line 154, in __h>
Dec 27 05:45:43 paulusesarch python3[3312314]:     raise HIDException(err)
Dec 27 05:45:43 paulusesarch python3[3312314]: lib.hid.HIDException: Success
Dec 27 05:45:43 paulusesarch python3[3312314]: During handling of the above exception, another exception occu>
Dec 27 05:45:43 paulusesarch python3[3312314]: Traceback (most recent call last):
Dec 27 05:45:43 paulusesarch python3[3312314]:   File "/usr/lib/python3.10/asyncio/base_events.py", line 595,>
Dec 27 05:45:43 paulusesarch python3[3312314]:     self._run_once()
Dec 27 05:45:43 paulusesarch python3[3312314]:   File "/usr/lib/python3.10/asyncio/base_events.py", line 1881>
Dec 27 05:45:43 paulusesarch python3[3312314]:     handle._run()
Dec 27 05:45:43 paulusesarch python3[3312314]:   File "/usr/lib/python3.10/asyncio/events.py", line 80, in _r>
Dec 27 05:45:43 paulusesarch python3[3312314]:     self._context.run(self._callback, *self._args)
Dec 27 05:45:43 paulusesarch python3[3312314]:   File "/usr/lib/keyboard-center/service.py", line 224, in usb>
Dec 27 05:45:43 paulusesarch python3[3312314]:     sys.exit(1)
Dec 27 05:45:43 paulusesarch python3[3312314]: SystemExit: 1
Dec 27 05:45:43 paulusesarch systemd[4451]: keyboard-center.service: Main process exited, code=exited, status>
Dec 27 05:45:43 paulusesarch systemd[4451]: keyboard-center.service: Failed with result 'exit-code'.
Dec 27 05:45:43 paulusesarch systemd[4451]: keyboard-center.service: Consumed 13min 14.871s CPU time.
Dec 27 05:46:08 paulusesarch systemd[4451]: Started background service for the Keyboard-Center application.
Dec 27 05:46:08 paulusesarch python3[3353743]: INFO: ------------ starting -------------
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: Config file location: /home/paulh/.config/keyboard-cent>
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: loading config file /home/paulh/.config/keyboard-center>
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: config loaded: {'mappings': {'MEMORY_1': {'MACRO_1': {'>
Dec 27 05:46:08 paulusesarch python3[3353743]: INFO: searching for supported keyboard...
Dec 27 05:46:08 paulusesarch python3[3353743]: INFO: keyboard found: Logitech G910 Orion Spectrum
Dec 27 05:46:08 paulusesarch python3[3353743]: INFO: saving deviceID into config
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: saving into config file /home/paulh/.config/keyboard-ce>
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: time taken to find keyboard in ms: 0.022220373153686523
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: requesting USB endpoint...
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: Searching for HIDraw endpoint...
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: HIDraw disable endpoint found: /dev/hidraw7
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: HIDraw read endpoint found: /dev/hidraw8
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: HIDraw read endpoint found: /dev/hidraw8
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: HIDraw read endpoint found: /dev/hidraw8
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: HIDraw read endpoint found: /dev/hidraw8
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: Checking for HID availability...
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: Connected to /dev/hidraw8
Dec 27 05:46:08 paulusesarch python3[3353743]: DEBUG: creating uinput device...
Dec 27 05:46:08 paulusesarch python3[3353743]: INFO: starting service...
[... cut ...]
Dec 27 17:41:58 paulusesarch python3[3356047]:   File "/usr/lib/keyboard-center/service.py", line 224, in usb>
Dec 27 17:41:58 paulusesarch python3[3356047]:     sys.exit(1)
Dec 27 17:41:58 paulusesarch python3[3356047]: SystemExit: 1
Dec 27 17:41:58 paulusesarch systemd[4451]: keyboard-center.service: Main process exited, code=exited, status>
Dec 27 17:41:58 paulusesarch systemd[4451]: keyboard-center.service: Failed with result 'exit-code'.
Dec 27 17:41:58 paulusesarch systemd[4451]: keyboard-center.service: Consumed 24min 27.504s CPU time.

zocker-160, 2021-12-28:
ok so I don't actually see any issue caused by keyboard-center here.

It is crashing because some other application takes over `/dev/hidraw8` which leads to a HIDError, keyboard-center not being able to read from it anymore and then crashing.

What other applications are you running that also try to access the keyboard?

> I noticed that during it, it was cycling my rgb profile even on my motherboard controlled lights, it is the same program

Keyboard-center does not touch any of this, all it does is call an openRGB profile if that is set up.

zocker-160, 2022-05-04:
closing because of inactivity, please reopen or create new issue there is any more information that could help finding the cause of this issue.

closed 2022-05-04 by zocker-160