zocker-160/keyboard-center #19: Service starting to fast??? Maybe?

Refrained-LM, 2021-12-29:
I Had to add a 10 sec delay at start up for keyboard central .service file to detect my g910 keyboard. I'm running manjaro latest (as of time of post) was wondering if that delay should be standard to prevent any problems? Or I accept that fact that my setup is configured wrong. Just a thought ( I know I uninstalled programs that didn't work right off the bat when I started linux)

zocker-160, 2021-12-30:
could you provide the log from the background service?

It would be interesting to know what exactly is going on. If the issue is just that the keyboard does not get detected, then that should be solvable by adding a retry count and only giving up after x amount of tries.

Refrained-LM, 2021-12-30:
Yes the keyboard was not being detected and I would have to reboot keyboard Central and then it would find the g910. I will attached log when I get home.

Refrained-LM, 2021-12-30:
keyboard-center.service - background service for the Keyboard-Center application
     Loaded: loaded (/usr/lib/systemd/user/keyboard-center.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Thu 2021-12-30 12:43:44 EST; 26s ago
    Process: 898 ExecStart=python3 /usr/lib/keyboard-center/service.py /home/***** (code=exited, status=1/FAILURE)
   Main PID: 898 (code=exited, status=1/FAILURE)
        CPU: 108ms

Dec 30 12:43:43 R***** systemd[887]: Started background service for the Keyboard-Center application.
Dec 30 12:43:43 R***** python3[898]: INFO: ------------ starting -------------
Dec 30 12:43:43 R***** python3[898]: DEBUG: Config file location: /home/*****/.config/keyboard-center/settings.yml
Dec 30 12:43:43 R***** python3[898]: DEBUG: loading config file /home/*****/.config/keyboard-center/settings.yml
Dec 30 12:43:43 R***** python3[898]: DEBUG: config loaded: {'mappings': {'MEMORY_1': {'MACRO_1': {'gamemode': False, 'name': 'Copy', 'string': ['Ctrl', 'C'], 'type': 'combo', 'value': [[1, 29], [1, 46]]}, 'MACR>
Dec 30 12:43:43 R***** python3[898]: INFO: searching for supported keyboard...
Dec 30 12:43:44 R***** python3[898]: CRITICAL: no supported keyboard found
Dec 30 12:43:44 R***** python3[898]: DEBUG: saving into config file /home/*****/.config/keyboard-center/settings.yml
Dec 30 12:43:44 R***** systemd[887]: keyboard-center.service: Main process exited, code=exited, status=1/FAILURE
Dec 30 12:43:44 R***** systemd[887]: keyboard-center.service: Failed with result 'exit-code'.
lines 1-17/17 (END)

Refrained-LM, 2021-12-30:
but if I add "ExecStartPre=/bin/sleep 10" before
"ExecStart=python3 /usr/lib/keyboard-center/service.py %h" in the .service file I have no errors and picks up my G910 perfectly. Along with loading a profile automatically before my KDE desktop finishes loading.

zocker-160, 2021-12-30:
Understood, thank you for the report and log. 

I will add a retry loop so that it does not give up the first time it cannot find a keyboard, which should solve this issue.

zocker-160, 2022-01-04:
I just pushed 0.1.22 which adds a retry procedure in case no keyboard is found on the first try.

Would you mind trying it out and give feedback if that fixes your issue @Refrained-LM ?

Refrained-LM, 2022-01-05:
Nope got this instead.   Also lost functionality  Could not force start like bfgore.
   ~  systemctl --user status keyboard-center.service                                                                                                                                              ✔  14s  
○ keyboard-center.service - background service for the Keyboard-Center application
     Loaded: loaded (/usr/lib/systemd/user/keyboard-center.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Wed 2022-01-05 07:15:32 EST; 1min 22s ago
   Main PID: 896 (code=exited, status=1/FAILURE)
        CPU: 40ms

Jan 05 07:15:32 R e systemd[872]: Started background service for the Keyboard-Center application.
Jan 05 07:15:32 R  python3[896]: Traceback (most recent call last):
Jan 05 07:15:32 R python3[896]:   File "/usr/lib/keyboard-center/service.py", line 14, in <module>
Jan 05 07:15:32 R  python3[896]:     import uinput
Jan 05 07:15:32 R  python3[896]: ModuleNotFoundError: No module named 'uinput'
Jan 05 07:15:32 R  systemd[872]: keyboard-center.service: Main process exited, code=exited, status=1/FAILURE
Jan 05 07:15:32 R systemd[872]: keyboard-center.service: Failed with result 'exit-code'

zocker-160, 2022-01-05:
hmmm `ModuleNotFoundError: No module named 'uinput'`

that is a completely different issue though, I did not change anything in that regard, did you try a reboot, sometimes that helps when kernel modules are missing. (uinput is such a kernel module)

Refrained-LM, 2022-01-05:
Ok had to reinstall openRGB and restart, but everything works perfectly now!. Thank you so much!

zocker-160, 2022-01-05:
awesome glad to hear :ok_hand:

closed 2022-01-05 by zocker-160