zocker-160/keyboard-center #20: Keyboard Center crashes on single key on G-Keys

xerhaxs, 2022-01-06:
Found the following problem:
If I don't put a key combination in the GUI of the Keyboard-Center, but only a single key on the G-keys, the GUI crashes. I can restart the GUI afterwards, but if I select the profile of the G-key, on which I placed a single key before, the Keyboard-Center crashes again immediately. I can only remove the key via settings.yml and then use the keyboard center again without restrictions. All other G-Keys still work, but the single key does not.
When the GUI crashes, the following error message appears when I start the keyboard center via the console:

```
Traceback (most recent call last):
  File "/usr/lib/keyboard-center/main.py", line 241, in setCurrMacro
    if load: self.loadData()
  File "/usr/lib/keyboard-center/main.py", line 279, in loadData
    d, orgb = self.configparser.loadForGui(
  File "/usr/lib/keyboard-center/lib/configparser.py", line 138, in loadForGui
    return ConfigEntry.fromConfig(data), openRGB
  File "/usr/lib/keyboard-center/lib/configtypes.py", line 68, in fromConfig
    if name: result.name = name
AttributeError: 'NoneType' object has no attribute 'name'
/usr/bin/keyboard-center: Zeile 3:  2551 Abgebrochen           python3 /usr/lib/keyboard-center/main.py "$@"
```

aurtzy, 2022-01-06:
Just tested this out; looks like I'm having the same issue.

Some additional info:

The macro configuration config values get kind of messed up when this happens. I'm guessing the NoneType error has to do with grabbing those unexpected values.

Here's a snippet of what an older working single-key configuration looks like:
```
MACRO_2:
      name: GameMenu
      string: '-'
      type: key
      value: [1, 12]
```
versus the one I created just now through the GUI:
```
MACRO_6:
      gamemode: 0
      name: [1, 12]
      string: key
      type: ''
      value: '-'
```

However, the interface will still crash if I visit any "fixed" single-key profile more than once, giving the same error message as @xerhaxs posted.

In case it might be useful, here's the log for me trying to view the same single-key profile twice:
```
trigger clicked 1
saving
None
{'MEMORY_1': {'MACRO_2': {'gamemode': 0, 'name': 'Menu', 'string': '-', 'type': 'key', 'value': [1, 12]}, 'MACRO_3': {'gamemode': 0, 'name': 'AltMenu', 'string': '=', 'type': 'key', 'value': [1, 13]}, 'MACRO_4': {'name': 'Alt', 'string': ';', 'type': 'key', 'value': [1, 39]}, 'MACRO_5': {'name': 'Extra', 'string': "'", 'type': 'key', 'value': [1, 40]}, 'MACRO_6': {'gamemode': 0, 'name': 'Menu', 'string': '-', 'type': 'key', 'value': [1, 12]}}}
{'mappings': {'MEMORY_1': {'MACRO_2': {'gamemode': 0, 'name': 'Menu', 'string': '-', 'type': 'key', 'value': [1, 12]}, 'MACRO_3': {'gamemode': 0, 'name': 'AltMenu', 'string': '=', 'type': 'key', 'value': [1, 13]}, 'MACRO_4': {'name': 'Alt', 'string': ';', 'type': 'key', 'value': [1, 39]}, 'MACRO_5': {'name': 'Extra', 'string': "'", 'type': 'key', 'value': [1, 40]}, 'MACRO_6': {'gamemode': 0, 'name': 'Menu', 'string': '-', 'type': 'key', 'value': [1, 12]}}}, 'openRGB': {}, 'settings': {'retryCount': 5, 'usbDeviceID': 0, 'usbTimeout': 1000}}
loading 0 1
AAAAA <lib.configtypes.Key object at 0x7f62489bd5a0> Menu 
trigger clicked 0
saving
<lib.configtypes.Key object at 0x7f62489bdcc0>
{'MEMORY_1': {'MACRO_2': {'gamemode': 0, 'name': 'Menu', 'string': '-', 'type': 'key', 'value': [1, 12]}, 'MACRO_3': {'gamemode': 0, 'name': 'AltMenu', 'string': '=', 'type': 'key', 'value': [1, 13]}, 'MACRO_4': {'name': 'Alt', 'string': ';', 'type': 'key', 'value': [1, 39]}, 'MACRO_5': {'name': 'Extra', 'string': "'", 'type': 'key', 'value': [1, 40]}, 'MACRO_6': {'gamemode': 0, 'name': 'Menu', 'string': '-', 'type': 'key', 'value': [1, 12]}}}
{'name': (1, 12), 'gamemode': 0, 'type': '', 'string': 'key', 'value': '-'}
{'mappings': {'MEMORY_1': {'MACRO_2': {'name': (1, 12), 'gamemode': 0, 'type': '', 'string': 'key', 'value': '-'}, 'MACRO_3': {'gamemode': 0, 'name': 'AltMenu', 'string': '=', 'type': 'key', 'value': [1, 13]}, 'MACRO_4': {'name': 'Alt', 'string': ';', 'type': 'key', 'value': [1, 39]}, 'MACRO_5': {'name': 'Extra', 'string': "'", 'type': 'key', 'value': [1, 40]}, 'MACRO_6': {'gamemode': 0, 'name': 'Menu', 'string': '-', 'type': 'key', 'value': [1, 12]}}}, 'openRGB': {}, 'settings': {'retryCount': 5, 'usbDeviceID': 0, 'usbTimeout': 1000}}
loading 0 0
trigger clicked 1
saving
None
{'MEMORY_1': {'MACRO_2': {'name': (1, 12), 'gamemode': 0, 'type': '', 'string': 'key', 'value': '-'}, 'MACRO_3': {'gamemode': 0, 'name': 'AltMenu', 'string': '=', 'type': 'key', 'value': [1, 13]}, 'MACRO_4': {'name': 'Alt', 'string': ';', 'type': 'key', 'value': [1, 39]}, 'MACRO_5': {'name': 'Extra', 'string': "'", 'type': 'key', 'value': [1, 40]}, 'MACRO_6': {'gamemode': 0, 'name': 'Menu', 'string': '-', 'type': 'key', 'value': [1, 12]}}}
{'mappings': {'MEMORY_1': {'MACRO_2': {'name': (1, 12), 'gamemode': 0, 'type': '', 'string': 'key', 'value': '-'}, 'MACRO_3': {'gamemode': 0, 'name': 'AltMenu', 'string': '=', 'type': 'key', 'value': [1, 13]}, 'MACRO_4': {'name': 'Alt', 'string': ';', 'type': 'key', 'value': [1, 39]}, 'MACRO_5': {'name': 'Extra', 'string': "'", 'type': 'key', 'value': [1, 40]}, 'MACRO_6': {'gamemode': 0, 'name': 'Menu', 'string': '-', 'type': 'key', 'value': [1, 12]}}}, 'openRGB': {}, 'settings': {'retryCount': 5, 'usbDeviceID': 0, 'usbTimeout': 1000}}
loading 0 1
```
which leads to that same error after this line.

zocker-160, 2022-01-06:
uhh that is a nasty bug, sorry about that, I will fix asap

zocker-160, 2022-01-06:
so I pushed 0.1.24, which fixes this bug.

Please remove the broken single key configs from the configuration file and let it regenerate with the GUI.

Sorry for the inconvenience that was a really stupid bug. -.-

xerhaxs, 2022-01-06:
Yep, the bug is fixed.

closed 2022-01-06 by xerhaxs