zocker-160/keyboard-center #21: Cannot make it work on Linux Mint 20.3 64-bit

Corionis, 2022-01-17:
Installed OpenRGB, all the dependencies, then the .deb of keyboard-center_0.1.24-1_amd64.deb on a fresh install of Mint 20.3 with a Logitech G910 Orion Spectrum keyboard.

The GUI comes up and I can define keys and see them in settings.yml.  Get notifications, and systemctl status says:
```
● keyboard-center.service - background service for the Keyboard-Center application
     Loaded: loaded (/usr/lib/systemd/user/keyboard-center.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2022-01-17 22:25:47 MST; 6h left
   Main PID: 2009 (python3)
     CGroup: /user.slice/user-1000.slice/user@1000.service/keyboard-center.service
             └─2009 /usr/bin/python3 /usr/lib/keyboard-center/service.py /home/trh

Jan 17 22:25:48 clavius python3[2009]: DEBUG: requesting USB endpoint...
Jan 17 22:25:48 clavius python3[2009]: DEBUG: Searching for HIDraw endpoint...
Jan 17 22:25:48 clavius python3[2009]: DEBUG: HIDraw disable endpoint found: /dev/hidraw2
Jan 17 22:25:48 clavius python3[2009]: DEBUG: HIDraw read endpoint found: /dev/hidraw3
Jan 17 22:25:48 clavius python3[2009]: DEBUG: Checking for HID availability...
Jan 17 22:25:48 clavius python3[2009]: DEBUG: Connected to /dev/hidraw3
Jan 17 22:25:48 clavius python3[2009]: DEBUG: creating uinput device...
Jan 17 22:25:48 clavius python3[2009]: INFO: starting service...
Jan 17 22:25:48 clavius python3[2009]: DEBUG: Using selector: EpollSelector
Jan 17 22:25:48 clavius python3[2009]: DEBUG: Sending sequence to disable G keys

```
Note the last line that is disabling G keys.  Odd that.  But most of the G keys generate a tilde ~ and not as defined.

Have checked everything I know to, rebooted, etc. but no joy.

Nice software and idea. I'm a software engineer but not into Python.  Will help however I can.

Corionis, 2022-01-18:
Output, with G1-G9 pressed:
```
journalctl --user -f -u keyboard-center
-- Logs begin at Sat 2022-01-15 12:33:33 MST. --
Jan 18 08:24:15 clavius python3[97807]: DEBUG: requesting USB endpoint...
Jan 18 08:24:15 clavius python3[97807]: DEBUG: Searching for HIDraw endpoint...
Jan 18 08:24:15 clavius python3[97807]: DEBUG: HIDraw disable endpoint found: /dev/hidraw2
Jan 18 08:24:15 clavius python3[97807]: DEBUG: HIDraw read endpoint found: /dev/hidraw3
Jan 18 08:24:15 clavius python3[97807]: DEBUG: Checking for HID availability...
Jan 18 08:24:15 clavius python3[97807]: DEBUG: Connected to /dev/hidraw3
Jan 18 08:24:15 clavius python3[97807]: DEBUG: creating uinput device...
Jan 18 08:24:15 clavius python3[97807]: INFO: starting service...
Jan 18 08:24:15 clavius python3[97807]: DEBUG: Using selector: EpollSelector
Jan 18 08:24:15 clavius python3[97807]: DEBUG: Sending sequence to disable G keys
^[OP^[OQ^[OR^[OS^[[15~^[[17~^[[18~^[[19~^[[20~
```

For hidraw2, hidraw3 and uinput from the output it is read/write permissions:
```
crw-rw----+ 1 root root 242, 0 Jan 17 22:25 /dev/hidraw0
crw-rw----+ 1 root root 242, 1 Jan 17 22:25 /dev/hidraw1
crw-rw-rw-+ 1 root root 242, 2 Jan 17 22:25 /dev/hidraw2
crw-rw-rw-+ 1 root root 242, 3 Jan 17 22:25 /dev/hidraw3
crw-rw----+ 1 root root 242, 4 Jan 17 22:25 /dev/hidraw4
crw-rw----+ 1 root root 242, 5 Jan 17 22:25 /dev/hidraw5
crw-rw----+ 1 root root 242, 7 Jan 17 22:25 /dev/hidraw7
crw-rw----+ 1 root root 242, 8 Jan 17 22:25 /dev/hidraw8

crw-rw-rw-+ 1 root users 10, 223 Jan 17 22:25 /dev/uinput
```
What should the ownership and permissions be?

Thank you for taking a look at this.  Maybe we can broaden the distros supported.

zocker-160, 2022-01-19:
Thanks for this additional information, to me it all looks as it should.

The main question here is, if the issue is that the disable sequence does not work or if writing to the hidraw endpoint doesn't work.

I will give you a small debugging script tomorrow, which should help us to find that out. \
(I guess we are in "opposite" time zones since it is almost morning over here xD)

Corionis, 2022-01-19:
While looking around I found there is also a /dev/usb directory:
```
total 0
drwxr-xr-x   2 root root    140 Jan 18  2022 .
drwxr-xr-x  20 root root   5040 Jan 18  2022 ..
crw-------   1 root root 180, 0 Jan 18  2022 hiddev0
crw-------   1 root root 180, 1 Jan 18  2022 hiddev1
crw-rw-rw-+  1 root root 180, 2 Jan 18  2022 hiddev2
crw-------   1 root root 180, 3 Jan 18  2022 hiddev3
crw-------   1 root root 180, 4 Jan 18  2022 hiddev4
```
Not sure if it's different than expected.

(Yes, I'm in Denver, Colorado. Have worked with people from your timezone before. I'm sitting here now (6:25PM) watching a movie. If want me to run something now I can pause it :)  )

zocker-160, 2022-01-20:
Sorry for the small delay.

I have updated the debugging script for your keyboard, it would be very useful if you could run it and see if you get any  data from the keyboard at all.

Corionis, 2022-01-22:
Apologies for the late reply. Distracted looking for work.

python3 src/newDeviceDebugger.py --info  Output:
```
DEBUG: Searching for keyboard...
DEBUG: Keyboard found! :)
DEBUG: requesting USB endpoint...
DEBUG: check and detach kernel driver if active
###
DEVICE ID 046d:c335 on Bus 003 Address 008 =================
 bLength                :   0x12 (18 bytes)
 bDescriptorType        :    0x1 Device
 bcdUSB                 :  0x200 USB 2.0
 bDeviceClass           :    0x0 Specified at interface
 bDeviceSubClass        :    0x0
 bDeviceProtocol        :    0x0
 bMaxPacketSize0        :   0x40 (64 bytes)
 idVendor               : 0x046d
 idProduct              : 0xc335
 bcdDevice              :  0x602 Device 6.02
 iManufacturer          :    0x1 Logitech
 iProduct               :    0x2 Gaming Keyboard G910
 iSerialNumber          :    0x3 117B385F3336
 bNumConfigurations     :    0x1
  CONFIGURATION 1: 500 mA ==================================
   bLength              :    0x9 (9 bytes)
   bDescriptorType      :    0x2 Configuration
   wTotalLength         :   0x3b (59 bytes)
   bNumInterfaces       :    0x2
   bConfigurationValue  :    0x1
   iConfiguration       :    0x4 U106.02_B0007
   bmAttributes         :   0xa0 Bus Powered, Remote Wakeup
   bMaxPower            :   0xfa (500 mA)
    INTERFACE 0: Human Interface Device ====================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x1
     bInterfaceClass    :    0x3 Human Interface Device
     bInterfaceSubClass :    0x1
     bInterfaceProtocol :    0x1
     iInterface         :    0x0 
      ENDPOINT 0x81: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :    0x8 (8 bytes)
       bInterval        :    0x1
    INTERFACE 1: Human Interface Device ====================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x1
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x1
     bInterfaceClass    :    0x3 Human Interface Device
     bInterfaceSubClass :    0x0
     bInterfaceProtocol :    0x0
     iInterface         :    0x0 
      ENDPOINT 0x82: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x82 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x1
###
```

python3 src/newDeviceDebugger.py  Output pressing G1-G9, Ctrl-C:
```
DEBUG: Searching for keyboard...
DEBUG: Keyboard found! :)
DEBUG: requesting USB endpoint...
DEBUG: check and detach kernel driver if active
INFO: starting service...
DEBUG: Using selector: EpollSelector
DEBUG: listening to USB Interface (1, 0)
^[OP^[OQ^[OR^[OS^[[15~^[[17~^[[18~^[[19~^[[20~^CINFO: stopping...
```

zocker-160, 2022-01-22:
I pushed an update to the script, could you please pull from master branch again and run the last command again?
(I don't need the `--info` output, that looks as it should)

Corionis, 2022-01-22:
Output with G1-G9 pressed:
```
DEBUG: Searching for keyboard...
DEBUG: Keyboard found! :)
DEBUG: requesting USB endpoint...
DEBUG: check and detach kernel driver if active
INFO: starting service...
DEBUG: Using selector: EpollSelector
DEBUG: Connection using HIDAPI...
DEBUG: Sending sequence to disable G keys
DEBUG: listening to USB Interface (1, 0)
got data from keyboard: b'\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x02\x00'
got data from keyboard: b'\x11\xff\x08\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x11\xff\x08.\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x11\xff\t\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x11\xff\n\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x11\xff\x08\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x11\xff\x08\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x11\xff\x08\x00\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x11\xff\x08\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x11\xff\x08\x00\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x11\xff\x08\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x11\xff\x08\x00\x08\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x11\xff\x08\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x11\xff\x08\x00\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x11\xff\x08\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x11\xff\x08\x00 \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x11\xff\x08\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x11\xff\x08\x00@\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x11\xff\x08\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x11\xff\x08\x00\x80\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x11\xff\x08\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x11\xff\x08\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x11\xff\x08\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
^CINFO: stopping...

```

zocker-160, 2022-01-22:
ooh this looks way better!

Can you now try to run with `--reset` and then start the background service again to see if it haves differently this time?

Corionis, 2022-01-22:
That's IT !!
Odd though ... the output still says "Sending sequence to disable G keys" ... but it works.
```
$ systemctl --user status keyboard-center.service
● keyboard-center.service - background service for the Keyboard-Center application
     Loaded: loaded (/usr/lib/systemd/user/keyboard-center.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2022-01-21 17:57:24 MST; 7s ago
   Main PID: 6763 (python3)
     CGroup: /user.slice/user-1000.slice/user@1000.service/keyboard-center.service
             └─6763 /usr/bin/python3 /usr/lib/keyboard-center/service.py /home/trh

Jan 21 17:57:24 clavius python3[6763]: DEBUG: requesting USB endpoint...
Jan 21 17:57:24 clavius python3[6763]: DEBUG: Searching for HIDraw endpoint...
Jan 21 17:57:24 clavius python3[6763]: DEBUG: HIDraw disable endpoint found: /dev/hidraw2
Jan 21 17:57:24 clavius python3[6763]: DEBUG: HIDraw read endpoint found: /dev/hidraw3
Jan 21 17:57:24 clavius python3[6763]: DEBUG: Checking for HID availability...
Jan 21 17:57:24 clavius python3[6763]: DEBUG: Connected to /dev/hidraw3
Jan 21 17:57:24 clavius python3[6763]: DEBUG: creating uinput device...
Jan 21 17:57:24 clavius python3[6763]: INFO: starting service...
Jan 21 17:57:24 clavius python3[6763]: DEBUG: Using selector: EpollSelector
Jan 21 17:57:24 clavius python3[6763]: DEBUG: Sending sequence to disable G keys

```

zocker-160, 2022-01-22:
ok I found the issue then, it is using the wrong endpoint for sending the disable sequence, would you mind testing a new version with this a patch for this and retesting again?

Corionis, 2022-01-22:
Sure ... I'm sitting here paying with a shell script  :)

zocker-160, 2022-01-22:
I am not sure though, if your keyboard needs hidraw2 or 3, so we have a 50% chance to get this right on the first try xD

Corionis, 2022-01-22:
Hmm ... I've rebooted and see these permissions.  Are they still OK?
```
root@clavius:/etc/rc5.d# ls -al /dev/hidraw*
crw-rw----+ 1 root root 242, 0 Jan 21 17:22 /dev/hidraw0
crw-rw----+ 1 root root 242, 1 Jan 21 17:22 /dev/hidraw1
crw-rw-rw-+ 1 root root 242, 2 Jan 21 17:22 /dev/hidraw2
crw-rw-rw-+ 1 root root 242, 3 Jan 21 17:54 /dev/hidraw3
crw-rw----+ 1 root root 242, 4 Jan 21 17:22 /dev/hidraw4
crw-rw----+ 1 root root 242, 5 Jan 21 17:22 /dev/hidraw5
crw-rw----+ 1 root root 242, 6 Jan 21 17:22 /dev/hidraw6
crw-rw----+ 1 root root 242, 7 Jan 21 17:22 /dev/hidraw7
crw-rw----+ 1 root root 242, 8 Jan 21 17:22 /dev/hidraw8
root@clavius:/etc/rc5.d# ls -al /dev/usb/
total 0
drwxr-xr-x   2 root root    140 Jan 21 17:54 .
drwxr-xr-x  20 root root   5040 Jan 21 17:54 ..
crw-------   1 root root 180, 0 Jan 21 17:22 hiddev0
crw-------   1 root root 180, 1 Jan 21 17:22 hiddev1
crw-rw-rw-+  1 root root 180, 2 Jan 21 17:54 hiddev2
crw-------   1 root root 180, 3 Jan 21 17:22 hiddev3
crw-------   1 root root 180, 4 Jan 21 17:22 hiddev4
root@clavius:/etc/rc5.d# ls -al /dev/uinput 
crw-rw----+ 1 root users 10, 223 Jan 21 17:22 /dev/uinput

```

zocker-160, 2022-01-22:
they should be exactly the same as before and to me it looks as it should

zocker-160, 2022-01-22:
Alright here the version with the patch.
[keyboard-center_0.1.25-1_amd64.zip](https://github.com/zocker-160/keyboard-center/files/7917348/keyboard-center_0.1.25-1_amd64.zip)

steps:
- stop the background service
- install new version
- start again
- check logs

Corionis, 2022-01-22:
Done.  THANK YOU!

Note that I got a GDebi install warning it was the same version.

zocker-160, 2022-01-22:
So it works now for you?

Does the log print the buttons pressed as it should?

> Note that I got a GDebi install warning it was the same version.

Yeah I just noticed I forgot to update the version inside the deb, but it should not matter really, I will publish the proper deb from CI CD, once you confirmed that it is working.

Corionis, 2022-01-22:
**_Yes_**, it works as I have it configured!  Very cool.

You mean this sort of output?
```
Jan 21 18:17:37 clavius python3[10042]: DEBUG: MACRO_2 pressed
Jan 21 18:17:42 clavius python3[10042]: DEBUG: MACRO_7 pressed
```

zocker-160, 2022-01-22:
> You mean this sort of output?

Yes, awesome, thanks for the help! 

Will publish as 0.1.25 then.

closed 2022-01-22 by zocker-160

Corionis, 2022-01-22:
So now it supports Linux Mint Cinnamon 20.3 !

btw - I came-up with a way for it to execute a program using custom Cinnamon Keyboard Shortcuts.

1. Define a G key in Keyboard Center for an odd combination such as Meta, F1.
2. Define a custom keyboard shortcut to execute a program on that key.

It works.

zocker-160, 2022-01-22:
command execution is on my TODO list, so stay tuned ;)

Corionis, 2022-01-25:
Argh ... it stopped working after a reboot.  Commenting here in case you wish to reopen this issue.

Been through the procedures here with no joy.  As far as I can tell nothing has changed:

```
$ d /dev/hi*
crw-rw----+ 1 root root 242, 0 Jan 25 09:03 /dev/hidraw0
crw-rw----+ 1 root root 242, 1 Jan 25 09:03 /dev/hidraw1
crw-rw-rw-+ 1 root root 242, 2 Jan 25 09:03 /dev/hidraw2
crw-rw-rw-+ 1 root root 242, 3 Jan 25 09:03 /dev/hidraw3
crw-rw----+ 1 root root 242, 4 Jan 25 09:03 /dev/hidraw4
crw-rw----+ 1 root root 242, 5 Jan 25 09:03 /dev/hidraw5
crw-rw----+ 1 root root 242, 7 Jan 25 09:03 /dev/hidraw7
crw-rw----+ 1 root root 242, 8 Jan 25 09:03 /dev/hidraw8

trh@clavius:~$ d /dev/uinput 
crw-rw-rw-+ 1 root users 10, 223 Jan 25 09:03 /dev/uinput

trh@clavius:~$ d /dev/usb
total 0
crw-------  1 root root 180, 0 Jan 25 09:03 hiddev0
crw-------  1 root root 180, 1 Jan 25 09:03 hiddev1
crw-rw-rw-+ 1 root root 180, 2 Jan 25 09:03 hiddev2
crw-------  1 root root 180, 3 Jan 25 09:03 hiddev3

trh@clavius:~$ systemctl --user status keyboard-center
● keyboard-center.service - background service for the Keyboard-Center application
     Loaded: loaded (/usr/lib/systemd/user/keyboard-center.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2022-01-25 09:03:29 MST; 2min 3s ago
   Main PID: 1974 (python3)
     CGroup: /user.slice/user-1000.slice/user@1000.service/keyboard-center.service
             └─1974 /usr/bin/python3 /usr/lib/keyboard-center/service.py /home/trh

Jan 25 09:03:30 clavius python3[1974]: DEBUG: requesting USB endpoint...
Jan 25 09:03:30 clavius python3[1974]: DEBUG: Searching for HIDraw endpoint...
Jan 25 09:03:30 clavius python3[1974]: DEBUG: HIDraw read endpoint found: /dev/hidraw3
Jan 25 09:03:30 clavius python3[1974]: DEBUG: HIDraw disable endpoint found: /dev/hidraw3
Jan 25 09:03:30 clavius python3[1974]: DEBUG: Checking for HID availability...
Jan 25 09:03:30 clavius python3[1974]: DEBUG: Connected to /dev/hidraw3
Jan 25 09:03:30 clavius python3[1974]: DEBUG: creating uinput device...
Jan 25 09:03:30 clavius python3[1974]: INFO: starting service...
Jan 25 09:03:30 clavius python3[1974]: DEBUG: Using selector: EpollSelector
Jan 25 09:03:30 clavius python3[1974]: DEBUG: Sending sequence to disable G keys

trh@clavius:~$ journalctl --user -f -u keyboard-center
-- Logs begin at Sat 2022-01-15 12:33:33 MST. --
Jan 25 09:03:30 clavius python3[1974]: DEBUG: requesting USB endpoint...
Jan 25 09:03:30 clavius python3[1974]: DEBUG: Searching for HIDraw endpoint...
Jan 25 09:03:30 clavius python3[1974]: DEBUG: HIDraw read endpoint found: /dev/hidraw3
Jan 25 09:03:30 clavius python3[1974]: DEBUG: HIDraw disable endpoint found: /dev/hidraw3
Jan 25 09:03:30 clavius python3[1974]: DEBUG: Checking for HID availability...
Jan 25 09:03:30 clavius python3[1974]: DEBUG: Connected to /dev/hidraw3
Jan 25 09:03:30 clavius python3[1974]: DEBUG: creating uinput device...
Jan 25 09:03:30 clavius python3[1974]: INFO: starting service...
Jan 25 09:03:30 clavius python3[1974]: DEBUG: Using selector: EpollSelector
Jan 25 09:03:30 clavius python3[1974]: DEBUG: Sending sequence to disable G keys
^[OP^[OQ^[OR^[OS^[[15~^[[17~^[[18~^[[19~^[[20~^C
```
The last system change was installing 0.1.25-1
Let me know if there is any further information or testing you need.

Corionis, 2022-01-25:
After rebooting again then running ``systemctl --user restart keyboard-center.service`` it came back.
Weird.
I will keep an eye on things and try to find a pattern to it.

zocker-160, 2022-01-25:
Yeah this is weird indeed.

Is is possible that you are running a dual boot setup with Windows and having the Logitech Ghub installed there?
From experience the official Logitech driver does sometimes put the keyboard in a weird state where it does not accept any commands until replugged.

zocker-160, 2022-01-25:
could you next time you do that switch try to "unpower" the keyboard for a few seconds before booting Linux just to see if that makes any difference?

Corionis, 2022-01-25:
Good idea.  Will try that this evening.

Corionis, 2022-02-11:
Whatever M$ and/or Logitech is doing to the G910 is weird and a real pain.

When switching back to Linux after playing Windows games:
 1. Stop boot by going to BIOS setup screen.
 2. Unplug the keyboard from the hub, wait 5 seconds, reconnect. It should be all blue keys.
 3. Boot Linux.
 4. Login.
 5. Run in terminal:  systemctl --user restart keyboard-center.service

I have OpenRGB installed but not configured in Keyboard Center, if that makes any difference.

Also, I'm still using 0.1.25-1 because it (generally) works.

Question:  Does Keyboard Center require re-install after a kernel update?

Thanks again for all your hard work.

zocker-160, 2022-02-11:
Yeah I know it is a pain, the Windows driver puts the keyboard in a super weird state, where it simply does not react to any commands you send to it.

The only solution that I could think of is to enforce a hard reset of the keyboard, but that sadly causes (at least in my testing) the Windows driver to not recognize the keyboard anymore. (and ppl will blame keyboard-center for killing their keyboard on Windows -.-)

> Login.
> Run in terminal: systemctl --user restart keyboard-center.service

After you login, it does only work when manually restarting the service? Can you try to just start the GUI and close it again after login?

> I have OpenRGB installed but not configured in Keyboard Center, if that makes any difference.

That is purely optional, it it does not change the behavior towards the keyboard over USB.

> Question: Does Keyboard Center require re-install after a kernel update?

No it does not as long as the new kernel has all kernel modules installed and active that are needed (which on Ubuntu based systems should always be the case as long as you don't compile your own kernel without them)

zocker-160, 2022-02-14:
I will create a new issue for that, can you please tell me what Logitech software you are using on Windows?

Is it the old Logitech Keyboard Software or the new GHub?

I will investigate this problem, but I need to reproduce it in order to do an attempt of fixing it.

Corionis, 2022-02-14:
Understood.  I'll help however I can.
I'm using the latest version of the older "Logitech Gaming Software", not the newer GHub.
I'm willing to install and test whatever you need as time allows.

zocker-160, 2022-02-14:
I created a separate issue https://github.com/zocker-160/keyboard-center/issues/24.

All updates regarding this issue will be there.

zocker-160, 2022-03-29:
@Corionis I sadly cannot ping you from the new issue I opened, so I will do it here and politely ask you for following my instructions here https://github.com/zocker-160/keyboard-center/issues/24#issuecomment-1082049533

(and pls answer there :) )