zocker-160/keyboard-center #23: Numpad keys on G815 can't be added to macro.

Aditeya, 2022-01-25:
If I try to add a key, from the numpad, on to the macro input box, nothing happens. It accepts `NumLock` but won't accept any keys from the numpad. I'm running this on Arch Linux with the bspwm TWM.

zocker-160, 2022-01-25:
Can confirm, does not work for me too, I will look into it

zocker-160, 2022-01-25:
Issue found, QT does interpret the keys on the numpad as modifiers (for whatever reason) and they therefore get ignored.
I will try to find a workaround and push an update.

zocker-160, 2022-01-25:
I just pushed 0.1.26 which should solve this issue, can you confirm it working?

Aditeya, 2022-01-25:
Yes, It works but it seems to accept multiple keys as input on the numpad (with numlock on and off). However, it only uses the last key entered when triggered.

Aditeya, 2022-01-25:
additionally if I make a macro such type `123` from the numpad and the numlock is off then it triggers the alternate keys (ie `End`, `Down`, `Page Down`)

zocker-160, 2022-01-25:
> Yes, It works but it seems to accept multiple keys as input on the numpad (with numlock on and off). However, it only uses the last key entered when triggered.

uuh I see, yes that is another bug which is a side effect of this (stupid) workaround for the numpad. will fix that

> additionally if I make a macro such type 123 from the numpad and the numlock is off then it triggers the alternate keys (ie End, Down, Page Down)

That is something I cannot influence or change. Keyboard Center does simulate the keypress as if you would physically press it. So it will behave depending on the state of the numlock.

zocker-160, 2022-01-25:
oops did not mean to close it

zocker-160, 2022-01-25:
pushed fix for the multiple keys issue as 0.1.27

the second issue is a wontfix, since I am simulating the physical button press and therefore cannot take the state of numlock into account.

Aditeya, 2022-01-26:
Yep, it works as expected. Also to detect numlock state, you could use `numlockx`. However, it wouldn't work on wayland since it is a X11 program.

zocker-160, 2022-01-26:
ok thanks for the feedback, will close this issue then.

> Also to detect numlock state, you could use numlockx. However, it wouldn't work on wayland since it is a X11 program.

Yes I know that there are multiple options to implement that, but I want to stay as desktop environment agnostic as possible, which is why I use the uinput and udev kernel modules instead of any X11 or Wayland based solution.

It would be possible to make the application aware of the numlock state, but that would add complexity that I frankly do not want to maintain :/.

commit 317e6bd0 (2022-01-25): fix numpad workaround breaking keylimit

in the attempt to fix #23 using a workaround the keylimit check got broken which allowed to add more than one button to the macro
closed 2022-01-26 by zocker-160