zocker-160/keyboard-center #24: Keyboard fails to disable default GKey mapping when rebooting from Windows

zocker-160, 2022-02-14:
Driver used on Windows is `Logitech Gaming Software` but it probably also affects GHub users.
Keyboard affected: G910

zocker-160, 2022-03-29:
(no idea why this got closed)

Corionis, 2022-03-29:
It appears to be working pretty well with 1.28-1.

A couple of things:

1. From Windows back to Linux worked the first time.
2. As a test I powered-off / unplugged everything and did a cold boot - oddly that failed.
3. Booted Windows, then shutdown (but did not power off), and booted Linux - that worked.

Instead of multi-booting hard drives I use an internal SSD dock and physically swap boot drives. Had Windows stomp on my Linux, twice.

Question: To be safe I uninstalled the prior version before installing 1.28-1.  Just wondering if that's necessary or whether the .deb package will upgrade itself?

I'll live with it a while and let you know if there are any consistent problems.

Thank you very much for fixing this and your hard work on this cool tool.

zocker-160, 2022-03-29:
thank you for the feedback!

Glad the soft reset seems to work most of the time, why it failed for the cold boot is weird, I will try to reproduce that.

> Question: To be safe I uninstalled the prior version before installing 1.28-1. Just wondering if that's necessary or whether the .deb package will upgrade itself?

you don't need to, apt will handle all of that for you and detect that you are upgrading in case on older version is already installed (so you can just do `apt install ./xxx.deb`)

> Question: To be safe I uninstalled the prior version before installing 1.28-1. Just wondering if that's necessary or whether the .deb package will upgrade itself?

yes pls let me know if there are issues, I sadly cannot do long term testing :)

zocker-160, 2022-05-04:
since there has not been any command or complaint about this for quite some time now, I will close this.

Feel free to reopen or create a new issue, if you still encounter problems.

commit 86e73bfd (2022-03-29): send soft reset command to G910 on startup

should (hopefully) resolve #24
closed 2022-05-04 by zocker-160