zocker-160/keyboard-center #26: trying to configure g815

mcasaccio, 2022-04-09:
Hello!!!
I'm trying to configure memory 1 to move between desktops (I'm assigning ctrl+ f1) but I can't get it to work.... it doesn't respond to the configurations

`mappings:
  MEMORY_1:
    MACRO_1:
      gamemode: 0
      name: CustomProfile
      string:
      - [Ctrl, F1]
      - [Ctrl, F2]
      - [Ctrl, F3]
      type: macro
      value:
      - - [1, 29]
        - [1, 59]
      - - [1, 29]
        - [1, 60]
      - - [1, 29]
        - [1, 61]
    MACRO_2:
      name: example-cut
      string:
      - [Ctrl, X]
      - [Ctrl, V]
      type: macro
      value:
      - - [1, 29]
        - [1, 45]
      - - [1, 29]
        - [1, 47]
    MACRO_3:
      name: example-nice
      string:
      - [N]
      - [I]
      - [C]
      - [E]
      type: macro
      value:
      - - [1, 49]
      - - [1, 23]
      - - [1, 46]
      - - [1, 18]
  MEMORY_2: {}
openRGB: {}
settings: {retryCount: 5, usbDeviceID: 3, usbTimeout: 1000}
`

zocker-160, 2022-04-10:
hi, thank you for the report. What desktop environment are you using? I would like to try to reproduce this issue.

mcasaccio, 2022-04-11:
> hola, gracias por el informe. ¿Qué entorno de escritorio estás usando? Me gustaría intentar reproducir este problema.
> 
> EDITAR: acabo de hacer algunas pruebas en KDE y funciona para mí, puedo cambiar entre espacios de trabajo sin problemas.

I'm using manjaro with kde plasma

zocker-160, 2022-04-11:
> Is there any way that the G keys do not write me the numbers:

This does not happen for me. Sadly I don't have a way to test on Manjaro.

Could you please send me your output for: `journalctl --user -f -u keyboard-center`

mcasaccio, 2022-04-11:
> > Is there any way that the G keys do not write me the numbers:
> 
> This does not happen for me. Sadly I don't have a way to test on Manjaro.
> 
> Could you please send me your output for: `journalctl --user -u keyboard-center > kc.log` and post the log file here?

yes, of course
`abr 08 10:24:14 matias systemd[803]: Started background service for the Keyboard-Center application.
abr 08 10:24:15 matias python3[20735]: INFO: ------------ starting -------------
abr 08 10:24:15 matias python3[20735]: DEBUG: Config file location: /home/matias/.config/keyboard-center/settings.yml
abr 08 10:24:15 matias python3[20735]: DEBUG: creating confFolder /home/matias/.config/keyboard-center
abr 08 10:24:15 matias python3[20735]: INFO: copying template config file
abr 08 10:24:15 matias python3[20735]: DEBUG: source: /usr/lib/keyboard-center/config/testconfig-example.yaml
abr 08 10:24:15 matias python3[20735]: DEBUG: destinatin: /home/matias/.config/keyboard-center/settings.yml
abr 08 10:24:15 matias python3[20735]: DEBUG: loading config file /home/matias/.config/keyboard-center/settings.yml
abr 08 10:24:15 matias python3[20735]: DEBUG: config loaded: {'mappings': {'MEMORY_1': {'MACRO_1': {'name': 'example-A', 'string': 'A', 'type': 'key', 'value': [1, 30]}, 'MACRO_2': {'name': 'example-cut', 'string': [['Ctrl', 'X'], ['Ctrl', 'V']], 'type': 'macro', 'value': [[[1, 29], [1, 45]], [[1, 29], [1, 47]]]}, 'MACRO_3': {'name': 'example-nice', 'string': [['N'], ['I'], ['C'], ['E']], 'type': 'macro', 'value': [[[1, 49]], [[1, 23]], [[1, 46]], [[1, 18]]]}}}, 'settings': {'usbTimeout': 1000, 'retryCount': 5}}
abr 08 10:24:15 matias python3[20735]: INFO: searching for supported keyboard...
abr 08 10:24:15 matias python3[20735]: INFO: keyboard found: Logitech G815
abr 08 10:24:15 matias python3[20735]: INFO: saving deviceID into config
abr 08 10:24:15 matias python3[20735]: DEBUG: saving into config file /home/matias/.config/keyboard-center/settings.yml
abr 08 10:24:15 matias python3[20735]: DEBUG: time taken to find keyboard in ms: 0.008833646774291992
abr 08 10:24:15 matias python3[20735]: DEBUG: requesting USB endpoint...
abr 08 10:24:15 matias python3[20735]: DEBUG: Searching for HIDraw endpoint...
abr 08 10:24:15 matias python3[20735]: DEBUG: HIDraw read endpoint found: /dev/hidraw3
abr 08 10:24:15 matias python3[20735]: DEBUG: HIDraw disable endpoint found: /dev/hidraw3
abr 08 10:24:15 matias python3[20735]: DEBUG: HIDraw read endpoint found: /dev/hidraw3
abr 08 10:24:15 matias python3[20735]: DEBUG: HIDraw disable endpoint found: /dev/hidraw3
abr 08 10:24:15 matias python3[20735]: DEBUG: HIDraw read endpoint found: /dev/hidraw3
abr 08 10:24:15 matias python3[20735]: DEBUG: HIDraw disable endpoint found: /dev/hidraw3
abr 08 10:24:15 matias python3[20735]: DEBUG: HIDraw read endpoint found: /dev/hidraw3
abr 08 10:24:15 matias python3[20735]: DEBUG: HIDraw disable endpoint found: /dev/hidraw3
abr 08 10:24:15 matias python3[20735]: DEBUG: HIDraw read endpoint found: /dev/hidraw3
abr 08 10:24:15 matias python3[20735]: DEBUG: HIDraw disable endpoint found: /dev/hidraw3
abr 08 10:24:15 matias python3[20735]: DEBUG: HIDraw read endpoint found: /dev/hidraw3
abr 08 10:24:15 matias python3[20735]: DEBUG: HIDraw disable endpoint found: /dev/hidraw3
abr 08 10:24:15 matias python3[20735]: DEBUG: Checking for HID availability...
abr 08 10:24:15 matias python3[20735]: DEBUG: Connected to /dev/hidraw3
abr 08 10:24:15 matias python3[20735]: DEBUG: creating uinput device...
abr 08 10:24:15 matias python3[20735]: INFO: starting service...
abr 08 10:24:15 matias python3[20735]: /usr/lib/keyboard-center/service.py:420: DeprecationWarning: There is no current event loop
abr 08 10:24:15 matias python3[20735]:   evLoop = asyncio.get_event_loop()
abr 08 10:24:15 matias python3[20735]: DEBUG: Using selector: EpollSelector
abr 08 10:24:15 matias python3[20735]: DEBUG: Sending sequence to disable G keys
abr 08 10:24:35 matias python3[20735]: INFO: config changed - reloading...
abr 08 10:24:35 matias python3[20735]: DEBUG: loading config file /home/matias/.config/keyboard-center/settings.yml
abr 08 10:24:35 matias python3[20735]: DEBUG: config loaded: {'mappings': {'MEMORY_1': {'MACRO_1': {'gamemode': 0, 'name': 'example-A', 'string': ['Ctrl', 'Alt', 'Shift', 'Meta', 'A'], 'type': 'combo', 'value': [[1, 29], [1, 56], [1, 42], [1, 125], [1, 30]]}, 'MACRO_2': {'name': 'example-cut', 'string': [['Ctrl', 'X'], ['Ctrl', 'V']], 'type': 'macro', 'value': [[[1, 29], [1, 45]], [[1, 29], [1, 47]]]}, 'MACRO_3': {'name': 'example-nice', 'string': [['N'], ['I'], ['C'], ['E']], 'type': 'macro', 'value': [[[1, 49]], [[1, 23]], [[1, 46]], [[1, 18]]]}}}, 'openRGB': {}, 'settings': {'retryCount': 5, 'usbDeviceID': 3, 'usbTimeout': 1000}}
abr 08 10:25:26 matias python3[20735]: INFO: config changed - reloading...
abr 08 10:25:26 matias python3[20735]: DEBUG: loading config file /home/matias/.config/keyboard-center/settings.yml
abr 08 10:25:26 matias python3[20735]: DEBUG: config loaded: {'mappings': {'MEMORY_1': {'MACRO_1': {'gamemode': 0, 'name': 'example-A', 'string': ['Ctrl', 'Alt', 'Shift', 'Meta', '1', 'A'], 'type': 'combo', 'value': [[1, 29], [1, 56], [1, 42], [1, 125], [1, 2], [1, 30]]}, 'MACRO_2': {'name': 'example-cut', 'string': [['Ctrl', 'X'], ['Ctrl', 'V']], 'type': 'macro', 'value': [[[1, 29], [1, 45]], [[1, 29], [1, 47]]]}, 'MACRO_3': {'name': 'example-nice', 'string': [['N'], ['I'], ['C'], ['E']], 'type': 'macro', 'value': [[[1, 49]], [[1, 23]], [[1 [... cut ...]
abr 11 11:12:22 matias python3[762]: DEBUG: MACRO_3 pressed
abr 11 13:07:57 matias python3[762]: DEBUG: MACRO_2 pressed
abr 11 13:07:58 matias python3[762]: DEBUG: MACRO_1 pressed
abr 11 13:07:59 matias python3[762]: DEBUG: MACRO_2 pressed
abr 11 13:07:59 matias python3[762]: DEBUG: MACRO_3 pressed
abr 11 13:08:00 matias python3[762]: DEBUG: MACRO_2 pressed
abr 11 13:08:00 matias python3[762]: DEBUG: MACRO_1 pressed
abr 11 13:08:03 matias python3[762]: DEBUG: MACRO_1 pressed
abr 11 13:08:04 matias python3[762]: DEBUG: MACRO_1 pressed
`

zocker-160, 2022-04-11:
thank you for the log, sadly I cannot find anything wrong going on, it all looks at it should.

zocker-160, 2022-04-12:
it probably is something Manjaro specific that I cannot test I am afraid.

mcasaccio, 2023-03-30:
I just installed it. It works perfectly!

closed 2023-03-30 by mcasaccio