zocker-160/keyboard-center #27: Support for G510

Flying--Dutchman, 2022-04-18:
Hello, I followed your steps in the README to include the (old) G510 Keyboard. 
I pressed all the memory (M-keys), makro (G-keys) and every other non-standard key I could find. The last key was the volume wheel.
Output:

USB ID: 046d:c22d

**python3 src/newDeviceDebugger.py:**
```
DEBUG: Searching for keyboard...
DEBUG: Keyboard found! :)
DEBUG: requesting USB endpoint...
DEBUG: HIDraw read endpoint found: /dev/hidraw1
DEBUG: HIDraw disable endpoint found: /dev/hidraw1
DEBUG: HIDraw read endpoint found: /dev/hidraw1
DEBUG: HIDraw disable endpoint found: /dev/hidraw1
DEBUG: Checking for HID availability...
DEBUG: Connected to /dev/hidraw1
INFO: starting listener...
/home/flyingdutchman/PycharmProjects/keyboard-center/src/newDeviceDebugger.py:181: DeprecationWarning: There is no current event loop
  evLoop = asyncio.get_event_loop()
DEBUG: Using selector: EpollSelector
DEBUG: Connection using HIDAPI...
DEBUG: Sending sequence to disable G keys
0: b'\x11\x01\x11\x1a\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
1: b'\x11\x01\n*\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
2: b'\x11\x01\x0fZ\x01\x03\x07\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
3: b'\x11\x01\x0f\x1a\x00\x02\x00\x00\x00\x00\x00\x084d\x00\x00\x01\x00\x00\x00'
4: b'\x11\x01\x0f\x1a\x01\x04\x00\x00\x00\x00\x00\x004\x01d\x08\x01\x00\x00\x00'
DEBUG: listening to USB Interface (1, 0) | b'/dev/hidraw1'
got data from keyboard: b'\x03\x00\x00\x10\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00 \x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00@\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x80\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x03\x01\x00\x00\x00'
^[OPgot data from keyboard: b'\x01\x00:\x00\x00\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x01\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x03\x02\x00\x00\x00'
^[OQgot data from keyboard: b'\x01\x00;\x00\x00\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x01\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x03\x04\x00\x00\x00'
^[ORgot data from keyboard: b'\x01\x00<\x00\x00\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x01\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x03\x08\x00\x00\x00'
^[OSgot data from keyboard: b'\x01\x00=\x00\x00\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x01\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x03\x10\x00\x00\x00'
^[[15~got data from keyboard: b'\x01\x00>\x00\x00\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x01\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x03 \x00\x00\x00'
^[[17~got data from keyboard: b'\x01\x00?\x00\x00\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x01\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x03@\x00\x00\x00'
^[[18~got data from keyboard: b'\x01\x00@\x00\x00\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x01\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x03\x80\x00\x00\x00'
^[[19~got data from keyboard: b'\x01\x00A\x00\x00\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x01\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x03\x00\x01\x00\x00'
[... 146 lines cut ...]
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x1
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x2
     bInterfaceClass    :    0x3 Human Interface Device
     bInterfaceSubClass :    0x0
     bInterfaceProtocol :    0x0
     iInterface         :    0x0 
      ENDPOINT 0x82: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x82 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :    0x8 (8 bytes)
       bInterval        :    0x4
      ENDPOINT 0x3: Interrupt OUT ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x3 OUT
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x4
###

```

zocker-160, 2022-04-18:
thank you very much for this output, that looks very useful. I do have a few questions:

- do you use this keyboard on Windows as well (sth like dual boot or VM)? If yes are you using LGS or GHub?
- does the keyboard press any keys (like F1 etc) when pressing any of the GKeys? (you can check with `evtest` for example)

I am asking, because in your output it seems like that there is a default mapping going on (which all modern Logitech keyboards do), but that means that the GKeyMapping disable function did not work - so we will have to do a fair bit of additional testing.

Flying--Dutchman, 2022-04-18:
I have to thank you for your work on this great tool! :+1: 

1. I do have a windows system with dual boot. I'm currently not using it, but I could install LGS.
2. Yes, it does. G1 - G12 are mapped with F1 - F12 and G13 - G18 are mapped with the numbers 1 - 6. To this day I still don't know why I thought I would use 18 x 3 macro keys.... 

The G510 is rather old (ca. 2010), and I would guess that there aren't a lot of people using it, especially not with Linux. 
So, don't put too much effort into it :wink:

zocker-160, 2022-04-19:
understood

> So, don't put too much effort into it

well it is more up to you if you are willing to test stuff for me, since you are the only one that can do that :)

the first thing you could try is go into the `src/newDeviceDebugger.py` file and try to use the `disableGKeys` variable from the G710+ and G910 and see if you manage to disable the default mapping this way.

If you are unfamiliar with Python, you can just commend and uncomment a line with `#`, so you need to comment lines 41 - 47 and uncomment 32 and then run it.

If we can find out the command to disable the mapping, then it should "just work"

Flying--Dutchman, 2022-04-19:
Unfortunately that didn't work. 
Also, with `usbUseWrite` set to False, I get an exception.

zocker-160, 2022-04-19:
I pushed an update to the script, please try again with that. Also as previously try with `disableGKeysInterface` set to 1 and 0.

Flying--Dutchman, 2022-04-19:
With `disableGKeysInterface = 0` I get an exception:

With `usbUseWrite = True`: `keyboard-center/src/newDeviceDebugger.py:122> exception=HIDException('Broken pipe')`
With `usbUseWrite = False`: `keyboard-center/src/newDeviceDebugger.py:122> exception=HIDException('ioctl (SFEATURE): Broken pipe')`

With `disableGKeysInterface = 1` The programm hangs after:
```
DEBUG: Sending sequence to disable G keys
0: b'\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
```

zocker-160, 2022-04-19:
ooh awesome!

alright I pushed another update to the script, could you do the button pressing again (like shown in the readme)?

It would be helpful to add some empty lines after each button press, so I can see what the keyboard is sending with each press.

zocker-160, 2022-04-22:
any update @Flying--Dutchman? You are the only one that can help me add support for your keyboard :)

Flying--Dutchman, 2022-04-23:
Hey, some stuff came up and I am currently on vacation. Will post the log here next Saturday or Sunday 👍

Flying--Dutchman, 2022-05-02:
Keys pressed:
M1, M2, M3, MR
G1 - G18
Game slider (on and off)
Headphone sound off
Microphone mute
LCD - App switch
LCD - Buttons 1-4
Keyboard light on/off
Play/Pause
Stop
Previous
Next
Sound off
Volume wheel (both directions)

```
DEBUG: listening to USB Interface (1, 0) | b'/dev/hidraw1'
got data from keyboard: b'\x03\x00\x00\x10\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x01\x00(\x00\x00\x00\x00\x00'

got data from keyboard: b'\x01\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00 \x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x01\x00(\x00\x00\x00\x00\x00'

got data from keyboard: b'\x01\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00@\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x01\x00(\x00\x00\x00\x00\x00'

got data from keyboard: b'\x01\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x80\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x01\x00(\x00\x00\x00\x00\x00'

got data from keyboard: b'\x01\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x03\x01\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x01\x00(\x00\x00\x00\x00\x00'

got data from keyboard: b'\x01\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x03\x02\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x01\x00(\x00\x00\x00\x00\x00'

got data from keyboard: b'\x01\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x03\x04\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x01\x00(\x00\x00\x00\x00\x00'

got data from keyboard: b'\x01\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x03\x08\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x01\x00(\x00\x00\x00\x00\x00'

got data from keyboard: b'\x01\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x03\x10\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x01\x00(\x00\x00\x00\x00\x00'

got data from keyboard: b'\x01\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x03 \x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x01\x00(\x00\x00\x00\x00\x00'

got data from keyboard: b'\x01\x00\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x03@\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
[... 132 lines cut ...]
got data from keyboard: b'\x02 '
got data from keyboard: b'\x02\x00'
got data from keyboard: b'\x02 '
got data from keyboard: b'\x02\x00'
got data from keyboard: b'\x02 '
got data from keyboard: b'\x02\x00'
got data from keyboard: b'\x02 '
got data from keyboard: b'\x02\x00'
got data from keyboard: b'\x02@'
got data from keyboard: b'\x02\x00'
got data from keyboard: b'\x02@'
got data from keyboard: b'\x02\x00'
got data from keyboard: b'\x02@'
got data from keyboard: b'\x02\x00'
got data from keyboard: b'\x02@'
got data from keyboard: b'\x02\x00'
got data from keyboard: b'\x02@'
got data from keyboard: b'\x02\x00'
got data from keyboard: b'\x02@'
got data from keyboard: b'\x02\x00'
got data from keyboard: b'\x01\x01\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x01\x01\x06\x00\x00\x00\x00\x00'
^CHIDerrpr: Interrupted system call

```

zocker-160, 2022-05-02:
thank you very much for this, that is very helpful

I wonder, do the media buttons (start stop etc) do anything when you press them? Do they work at all in Linux?
If not I could add support for those as well and make them work.

Flying--Dutchman, 2022-05-02:
I'm the one who should be thankful, not you. :smile: 

The media buttons are working on my system, without any problem.
The headphone and microphone buttons also work, but only when the device is connected with the keyboard unfortunately. (Same as on windows)
Game switch and keyboard light also work without any problem, the logic for those are probably directly on the keyboard.

zocker-160, 2022-05-02:
> The media buttons are working on my system, without any problem.
>  Game switch and keyboard light also work without any problem, the logic for those are probably directly on the keyboard.

ok nice sounds good, I will then just ignore those inputs.

zocker-160, 2022-05-03:
>  Volume wheel (both directions)

did you scroll up or down first? :smile:

zocker-160, 2022-05-03:
alright I have a first version for testing, what system are you using @Flying--Dutchman? Can I give you a DEB?

Flying--Dutchman, 2022-05-03:
> did you scroll up or down first? smile
Good question.... I think I scrolled down first ^^ But that control also already works.

> alright I have a first version for testing, what system are you using @Flying--Dutchman? Can I give you a DEB?
I'm running Manjaro, but you can give me a deb file

> EDIT: and please let me know how the UI handles 18 macro keys, I have not tested it with so many buttons xD
Will do :smile:

zocker-160, 2022-05-03:
alright there you go @Flying--Dutchman 
[keyboard-center-git.zip](https://github.com/zocker-160/keyboard-center/files/8609647/keyboard-center-git.zip)

Flying--Dutchman, 2022-05-04:
I tested the G3-example, which works -> nice
But some of the F-keys are still mapped. When pressing G11, it triggers the F11-Key for example.
I will check if I can narrow it down to a line of code.

> EDIT: and please let me know how if the UI handles 18 macro keys, I have not tested it with so many buttons xD
The ui has no problem with the 18 keys and works great! :+1:

zocker-160, 2022-05-04:
hmm weird, so we still have the mapping issue ._.

Flying--Dutchman, 2022-05-10:
Yes, and I cant figure out why... 
Will try again with the "NewDeviceDebugger", as soon when I have time

Flying--Dutchman, 2022-05-12:
Ok, so I tried it with the sources from 19. April and the current main branch.
The only thing I needed to do in `newDeviceDebugger` was to change the following line:
`usbUseWrite = True` to `usbUseWrite = False`

zocker-160, 2022-05-12:
@Flying--Dutchman alright I pushed 0.1.31 with useWrite disabled for G510

let me know if it works this time ;)

Flying--Dutchman, 2022-05-13:
[image]

Flying--Dutchman, 2022-05-13:
Works like a charm, thank you very much! :+1:

closed 2022-05-13 by Flying--Dutchman

zocker-160, 2022-05-13:
nice, very good :ok_hand: