zocker-160/keyboard-center #29: can't set alt-tab

recallmenot, 2022-05-03:
Hi,
I'm trying to create a macro that contains ALT+tab but when I press tab in the "press shortcut" field or "custom modifier" field it switches away to the next GUI element instead of registering that I pressed tab.
Can you help me solve this?
I'm on a G710+

zocker-160, 2022-05-03:
uh yeah thanks for the report, this could be a tricky one to fix, I will look into it.

recallmenot, 2022-05-03:
meanwhile for anyone looking to use alt-tab, I found a workaround:

keyboard-center uses [these](https://github.com/torvalds/linux/blob/master/include/uapi/linux/input-event-codes.h) codes for the key:

create the macro without the alt+tab in it first, then close keyboard-center, saving it

backup your ~/.config/keyboard-center/settings.yml, then edit it manually and insert into the macro you created:
in "string"
`      - [Alt, Tab]`
in "value"
```
      - - [1, 56]
        - [1, 15]
```
56 is code for ALT
15 is code for TAB
`
systemctl --user restart keyboard-center`

zocker-160, 2022-05-04:
fix is available now in 0.1.29

closed 2022-05-04 by zocker-160

recallmenot, 2022-05-04:
my god that was fast, works now 10/10
thank you very much!
also, I love the AUR :D