zocker-160/keyboard-center #30: ability to send modifiers by themselves

recallmenot, 2022-05-04:
Hi again, sorry to be annoying. :D
I'm using virt-manager and want to quickly alt tab to another program on the host.
To release the focus from the guest I'd need to send L_CTRL+L_ALT without anything else (followed by ALT+TAB of course but that works perfectly now).
Currently keyboard-center when saving throws the error "No key assigned!".
Maybe it would it be possible if you made that just an ignorable warning?

zocker-160, 2022-05-04:
ah yeah that makes sense, adding an ignore button to the warning might be a solution, though not one I would like.
I will look into that as well ^^

zocker-160, 2022-05-05:
this will require a hacky solution, because sadly `QKeySequenceEdit` does not support inputting modifiers alone with no additional key.

recallmenot, 2022-05-05:
hi,
I found a workaround again
If you look at the [input-event-codes](https://github.com/torvalds/linux/blob/master/include/uapi/linux/input-event-codes.h), #0 is bound to "reserved". as far as I can tell, sending that works fine, though I had to put it manually into the settings.yml
maybe tell QKeySequenceEdit to send that
my string line is `      - [Ctrl, Alt, nullkey]`
my value lines for L-alt and L-Ctrl are
```
      - - [1, 29]
        - [1, 56]
        - [1, 0]
```

zocker-160, 2022-05-05:
hmm yeah it could work, but it would be limited to this very specific use case.

I think it would be better to have some type of "RAW" input mode which bypasses the `QKeySequenceEdit`, this would also give the user the ability to differentiate between `L_CTRL` and `R_CTRL` which currently is not possible.

zocker-160, 2022-05-07:
@recallmenot I pushed to 0.1.30 which adds raw input and a nullkey, could you please give me some feedback, I sadly could not test this in depth since I do not have any use case for such shortcuts.

recallmenot, 2022-05-10:
Hi Zocker, sorry for the wait.
I tested it, it works, awesome! And it's easy to use!
Only sometimes when sending CTRL+ALT, the VM receives ALT and the guest receives CTRL + ALT.
That seems to be a bug in the Virt-man window, probably nothing you can do anything about.
The program sends
```
- - [1, 29]
  - [1, 56]
  - [1, 0]
```
Thank you for making this great program!

closed 2022-05-10 by recallmenot

zocker-160, 2022-05-10:
nice glad to hear, thank you for the feedback @recallmenot