zocker-160/keyboard-center #32: keyboard-center not starting, KeyError: 'usbDeviceID'

Aditeya, 2022-06-26:
I'm was trying to start keyboard-center but it wouldn't start up and would show this error:
[image]

Running it in the terminal gives this error:
```
[aditeya@nemo ~/.config/keyboard-center]$ keyboard-center
[root] INFO: 26.06.2022 04:01:31 PM ------------ starting -------------
[Configparser] DEBUG: 26.06.2022 04:01:31 pm Config file location: /home/aditeya/.config/keyboard-center/settings.yml
[Configparser] DEBUG: 26.06.2022 04:01:31 pm loading config file /home/aditeya/.config/keyboard-center/settings.yml
[Configparser] DEBUG: 26.06.2022 04:01:31 pm config loaded: {'mappings': {'MEMORY_1': {'MACRO_1': {'name': 'example-A', 'string': 'A', 'type': 'key', 'value': [1, 30]}, 'MACRO_2': {'name': 'example-cut', 'string': [['Ctrl', 'X'], ['Ctrl', 'V']], 'type': 'macro', 'value': [[[1, 29], [1, 45]], [[1, 29], [1, 47]]]}, 'MACRO_3': {'name': 'example-nice', 'string': [['N'], ['I'], ['C'], ['E']], 'type': 'macro', 'value': [[[1, 49]], [[1, 23]], [[1, 46]], [[1, 18]]]}}}, 'settings': {'usbTimeout': 1000, 'retryCount': 5}}
[BGService] INFO: 26.06.2022 04:01:31 pm setting up service...
[BGService] INFO: 26.06.2022 04:01:31 pm searching for supported keyboard...
[BGService] INFO: 26.06.2022 04:01:31 pm keyboard found: Logitech G815
[Configparser] DEBUG: 26.06.2022 04:01:31 pm setting deviceID to 3
[root] ERROR: 26.06.2022 04:01:44 pm 'usbDeviceID'
Traceback (most recent call last):
  File "/usr/lib/keyboard-center/main.py", line 55, in <module>
    window = MainWindow(app, devmode)
  File "/usr/lib/keyboard-center/mainUi.py", line 140, in __init__
    self.initBackgroundService()
  File "/usr/lib/keyboard-center/mainUi.py", line 230, in initBackgroundService
    service = BackgroundService(self.configparser)
  File "/usr/lib/keyboard-center/service.py", line 89, in __init__
    self._initKeyboard()
  File "/usr/lib/keyboard-center/service.py", line 103, in _initKeyboard
    self.config.setAndSaveDeviceID(i)
  File "/usr/lib/keyboard-center/lib/configparser.py", line 142, in setAndSaveDeviceID
    if self.getDeviceID() != id:
  File "/usr/lib/keyboard-center/lib/configparser.py", line 80, in getDeviceID
    return self.getSettings()["usbDeviceID"]
KeyError: 'usbDeviceID'
Traceback (most recent call last):
  File "/usr/lib/keyboard-center/main.py", line 55, in <module>
    window = MainWindow(app, devmode)
  File "/usr/lib/keyboard-center/mainUi.py", line 140, in __init__
    self.initBackgroundService()
  File "/usr/lib/keyboard-center/mainUi.py", line 230, in initBackgroundService
    service = BackgroundService(self.configparser)
  File "/usr/lib/keyboard-center/service.py", line 89, in __init__
    self._initKeyboard()
  File "/usr/lib/keyboard-center/service.py", line 103, in _initKeyboard
    self.config.setAndSaveDeviceID(i)
  File "/usr/lib/keyboard-center/lib/configparser.py", line 142, in setAndSaveDeviceID
    if self.getDeviceID() != id:
  File "/usr/lib/keyboard-center/lib/configparser.py", line 80, in getDeviceID
    return self.getSettings()["usbDeviceID"]
KeyError: 'usbDeviceID'
```

This error showed up after I tried resetting the config. I did the reset because the application wasn't working and if I try to start it, a transparent window just opens.

zocker-160, 2022-06-26:
did you reboot after the update specified in the release notes?

if you wish to reset the config, please just delete it, keyboard-center should generate a new file.

Aditeya, 2022-06-27:
I believe I ran keyboard-center once after rebooting. I was only aware of rebooting after that when i checked release notes. 
I also deleted the `.config/keyboard-center` directory and rebooted, but it gives this error despite the file being present: 

[image]

terminal output: 
```
[aditeya@nemo ~]$ keyboard-center
[root] INFO: 27.06.2022 09:34:43 AM ------------ starting -------------
[Configparser] DEBUG: 27.06.2022 09:34:43 am Config file location: /home/aditeya/.config/keyboard-center/settings.yml
[Configparser] INFO: 27.06.2022 09:34:43 am copying template config file
[Configparser] DEBUG: 27.06.2022 09:34:43 am source: /usr/lib/keyboard-center/config/testconfig.yaml.example
[Configparser] DEBUG: 27.06.2022 09:34:43 am destinatin: /home/aditeya/.config/keyboard-center/settings.yml
[root] ERROR: 27.06.2022 09:35:35 am [Errno 2] No such file or directory: '/usr/lib/keyboard-center/config/testconfig.yaml.example'
Traceback (most recent call last):
  File "/usr/lib/keyboard-center/main.py", line 55, in <module>
    window = MainWindow(app, devmode)
  File "/usr/lib/keyboard-center/mainUi.py", line 139, in __init__
    self.configparser = self.getConfiguration()
  File "/usr/lib/keyboard-center/mainUi.py", line 211, in getConfiguration
    configparser = Configparser(TEMPLATE_LOCATION, silent=False)
  File "/usr/lib/keyboard-center/lib/configparser.py", line 41, in __init__
    self._getConfigLocation(locConfTemplate)
  File "/usr/lib/keyboard-center/lib/configparser.py", line 63, in _getConfigLocation
    self._copyConfig(confTemplate, confLoc)
  File "/usr/lib/keyboard-center/lib/configparser.py", line 248, in _copyConfig
    with open(src, "r") as s:
FileNotFoundError: [Errno 2] No such file or directory: '/usr/lib/keyboard-center/config/testconfig.yaml.example'
Traceback (most recent call last):
  File "/usr/lib/keyboard-center/main.py", line 55, in <module>
    window = MainWindow(app, devmode)
  File "/usr/lib/keyboard-center/mainUi.py", line 139, in __init__
    self.configparser = self.getConfiguration()
  File "/usr/lib/keyboard-center/mainUi.py", line 211, in getConfiguration
    configparser = Configparser(TEMPLATE_LOCATION, silent=False)
  File "/usr/lib/keyboard-center/lib/configparser.py", line 41, in __init__
    self._getConfigLocation(locConfTemplate)
  File "/usr/lib/keyboard-center/lib/configparser.py", line 63, in _getConfigLocation
    self._copyConfig(confTemplate, confLoc)
  File "/usr/lib/keyboard-center/lib/configparser.py", line 248, in _copyConfig
    with open(src, "r") as s:
FileNotFoundError: [Errno 2] No such file or directory: '/usr/lib/keyboard-center/config/testconfig.yaml.example'
```

I Installed this through the AUR (keyboard-center)

zocker-160, 2022-06-27:
ok I see the issue, sorry about that, but I messed up on two places at the same time with this release

I will push a fix asap

zocker-160, 2022-06-27:
I pushed an update, which should fix both of the issues.

Please let me know if that fixes it for you.

Aditeya, 2022-06-27:
Yup its working now, but upon startup theres an invisible/transparend window with a border:

[image]

Its barely visible, but you can see the border. Clicking unhide & then hide fixes the issue.
Another one is after i hide and unhide and change workspaces, (like say to 1 where the browser is) it overlays the entire space. 

I'm not sure if this is a quirk with bspwm or if its an issue with the program tho

zocker-160, 2022-06-27:
thanks for the feedback

That is very weird. Does changing the setting "minimize on start" have any influence on that?

I think one workaround would be to minimize the application instead hiding to tray. It seems like hiding the window does cause issues for you.

I sadly cannot reproduce that, I am just using QT's `window.hide()` function ._.

Aditeya, 2022-06-28:
Yes, disabling minimize on start works well. However it keeps the window open on start up. Would it be possible to add a CLI option or flag, so it just runs in the background. That way I can just add it to a startup script, but I can start up the GUI to configure if need be. Sort of like how the systemd background process worked.

zocker-160, 2022-06-28:
The problem with the tray icon is sadly not easily solvable. 

I need it, in order to make the command feature work properly. 

If I give the user the option to hide the tray icon as well, the user will not be able to reactivate the window anymore. If the user just opens the application up again, it will be a new instance with a new process, completely independent from the first.

Sadly on Wayland there is no way for me to maximize a window of a different process unless I start implementing Unix or network sockets for IPC, which is quite ugly.

If you do not care about the command feature, feel free to revert back to a 0.1.x version, the PKGBUILD is in the AUR, you can pull it and just install it with `makepkg -si`

Aditeya, 2022-06-29:
Ah I see, I guess I'm okay with this current implementation then. Thank you!

zocker-160, 2022-06-29:
If you have an idea how to solve this, feel free to let me know, I am open to suggestions and I fully understand if you find the icon in the tray annoying.

I will look into implementing a global hotkey for reactivation, I think that could be a potential solution for power users.

Since the original issue is solved, I will close this issue.

closed 2022-06-29 by zocker-160