zocker-160/keyboard-center #34: On Laptop wake up

HekTheLastSavior, 2022-07-04:
Hey,
Great project thank you for all your efforts.
I'm having an issue where the deamon previously and now the taskbar stop working when the laptop wake up.
I wrote a script that kill & start the deamon on previous version but it's not optimal because I had to run it manually.
Any suggestion to handle this issue ?
Thanks

HekTheLastSavior, 2022-07-04:
Sure,
log localtion `~/.var/log/`

`[root] INFO: 04.07.2022 06:15:58 PM ------------ starting -------------
[Configparser] DEBUG: 04.07.2022 06:15:58 PM Config file location: /home/XXX/.config/keyboard-center/settings.yml
[Configparser] DEBUG: 04.07.2022 06:15:58 PM loading config file /home/XXX/.config/keyboard-center/settings.yml
[Configparser] DEBUG: 04.07.2022 06:15:58 PM config loaded: {.....}
[BGService] INFO: 04.07.2022 06:15:58 PM setting up service...
[BGService] INFO: 04.07.2022 06:15:58 PM searching for supported keyboard...
[BGService] INFO: 04.07.2022 06:15:58 PM keyboard found: Logitech G910 Orion Spectrum
[Configparser] DEBUG: 04.07.2022 06:15:58 PM setting deviceID to 0
[BGService] DEBUG: 04.07.2022 06:15:58 PM time taken to find keyboard in ms: 6.9637298583984375
[BGService] DEBUG: 04.07.2022 06:15:58 PM requesting USB endpoint...
[BGService] DEBUG: 04.07.2022 06:15:58 PM searching HIDpaths...
[BGService] DEBUG: 04.07.2022 06:15:58 PM HIDraw read endpoint found: /dev/hidraw2
[BGService] DEBUG: 04.07.2022 06:15:58 PM HIDraw write endpoint found: /dev/hidraw2
[BGService] DEBUG: 04.07.2022 06:15:58 PM HIDraw read endpoint found: /dev/hidraw2
[BGService] DEBUG: 04.07.2022 06:15:58 PM HIDraw write endpoint found: /dev/hidraw2
[BGService] DEBUG: 04.07.2022 06:15:58 PM HIDraw read endpoint found: /dev/hidraw2
[BGService] DEBUG: 04.07.2022 06:15:58 PM HIDraw write endpoint found: /dev/hidraw2
[BGService] DEBUG: 04.07.2022 06:15:58 PM HIDraw read endpoint found: /dev/hidraw2
[BGService] DEBUG: 04.07.2022 06:15:58 PM HIDraw write endpoint found: /dev/hidraw2
[BGService] DEBUG: 04.07.2022 06:15:58 PM Checking for HID availability...
[BGService] DEBUG: 04.07.2022 06:15:58 PM Connected to /dev/hidraw2
[BGService] DEBUG: 04.07.2022 06:15:58 PM creating uinput device
[QT] DEBUG: 04.07.2022 06:15:58 PM start health check...
[BGService] INFO: 04.07.2022 06:15:58 PM starting service loop...
[BGService] DEBUG: 04.07.2022 06:15:58 PM Sending sequence to disable G keys
[QT] DEBUG: 04.07.2022 06:15:58 PM setting up GUI...
[Configparser] DEBUG: 04.07.2022 06:15:58 PM loading config file /home/XXX/.config/keyboard-center/settings.yml
[Configparser] DEBUG: 04.07.2022 06:15:58 PM config loaded: {....}
[BGService] DEBUG: 04.07.2022 06:16:20 PM MACRO_1 pressed`

HekTheLastSavior, 2022-07-05:
The tray is always available and the software is still accessible.

My keyboard simply reset. Currently I start the software again in order to gain back my keyboard settings.
Mind you that is the issue was in the deamon as well. 

It might need a way to stay alive, or listen to state of the pc. 
I did a quick research about Linux state and I found [acpid](https://wiki.archlinux.org/title/Acpid), it seems to be somewhat universal in all Linux flavors.

I'm not good with Python, I would've added some code to this, my suggestion if it's too annoying to integrate in code is to have a script that restart the service, the previous deamon version was fine by me.

Thank you and keep up the good work :)

zocker-160, 2022-07-05:
do you suspend to RAM or suspend to disk?
I did some testing yesterday and I sadly cannot reproduce it, but I only tested suspend to RAM.

I am willing to look into this, but I need to understand first what is going on, which I sadly don't right now.

What do you mean with "reset"? Do the macro keys work? Do the M-keys work? Does the application register any keyboard input?
Do you use it with openRGB?

You can follow the log with `tail -f ~/.var/log/keyboardCenter.log`

As for the deamon, if you do not need the command feature, you can always revert back to an older 0.1.x version.

HekTheLastSavior, 2022-07-06:
> do you suspend to RAM or suspend to disk? I did some testing yesterday and I sadly cannot reproduce it, but I only tested suspend to RAM.
Either depending on battery state but mostly to ram.

> I am willing to look into this, but I need to understand first what is going on, which I sadly don't right now.
Great

> What do you mean with "reset"? Do the macro keys work? Do the M-keys work? Does the application register any keyboard input? Do you use it with openRGB?
I mean by "Reset" that as if your software is not running and the keyboard is on default state with memory buttons off as in not lid. I'm not using openRGB

> You can follow the log with `tail -f ~/.var/log/keyboardCenter.log`
I did a quick experiment, so when laptop resume from sleep there's no activity in the log , the taskbar icons is active and the software is active but no actual result on the keyboard ( default state as mentioned above) if I keep the software running and open it again , the keyboard is active but the G keys are run twice. So I have to close one of them.
Regarding log output is similar to my previous log dump.

> As for the deamon, if you do not need the command feature, you can always revert back to an older 0.1.x version.
I prefer to be use the latest version :)

zocker-160, 2022-07-06:
ok thanks for the explanation, I think I get what is going on, the application goes get a SIGINT I think.

What distro are you on, would you be willing to try a debug version for testing? I sadly still cannot reproduce, it works just fine after suspend using Ubuntu 20.04 LTS.

HekTheLastSavior, 2022-07-07:
No worries, I'm running Manjero with kde .
I can do a debug test, let me know what I should do

zocker-160, 2022-07-08:
I sadly have bad news, I just did some testing on a Manjaro KDE laptop and I still cannot recreate the issue. 

I did suspend and resume multiple times and the macro keys keep working for me after resume every time. Sometimes it just takes a few seconds for the driver to kick in, but it works.

So I am sadly out of ideas on this one, the only thing I could think of is adding a force restart service option in the menu.

HekTheLastSavior, 2022-07-08:
It's fine , it's better if it's can been done from the terminal it will be easier to trigger it .
Thank you :)

zocker-160, 2022-07-09:
I do not think I can offer you that though, since that would require some IPC, which I already tried (see issue 33) and it did not work.

Whatever Arch does, it does not work and I am not really willing to invest a lot of time into implementing something like DBus.

HekTheLastSavior, 2022-07-09:
I see, no worries. Thank you though and keep up the good work :)

closed 2022-07-09 by HekTheLastSavior

zocker-160, 2022-07-12:
I added a force restart option to the tray icon in 0.2.7, this is the best I can do for you for now.