zocker-160/keyboard-center #35: Feature Request - Ability to paste strings

Amon-X, 2022-07-14:
Please add the ability to paste strings instead of having to add each character individually. Something similar to ydotool type 'Sentence to be type.'

zocker-160, 2022-07-14:
thanks for this suggestion, I will look into it.

zocker-160, 2022-08-04:
@Amon-X  alright so I did think about this for a while and I am hesitant to implement this, because I would need to add a new button with a new implementation of something that is already possible using the command field like this:

[image]

Amon-X, 2022-08-04:
I only created this request at your suggestion (reddit pms). That xdotool type is a far more elegant solution to what I was using. Thank you for that.  The command I am using is` /bin/bash -c 'echo -n "some crazy text!!!" | xsel && xdotool click 2'` which pastes all the text at once but takes over the middle mouse click.

zocker-160, 2022-08-04:
> I only created this request at your suggestion (reddit pms)

Yes and I thank you for doing that, because you are not the only one requesting this.

So I will keep the issue open for now, it is not off the table, but I do dislike to add xdotool / ydotool as a dependency of keybard-center.

Amon-X, 2022-08-05:
What about capturing multiple/a series of keys at once instead of just a single key? Would that be a way of doing it without adding dependencies?

zocker-160, 2022-08-05:
agreed, that is a possibility, I will keep that in mind

zocker-160, 2023-03-17:
So I finally found the time to look into this again and decided to not implement this, as there are way better tools out there that can achieve the same (and more).

The main reason is that Wayland sadly makes things very complicated, since there is no equivalent to `xdotool`. Any other replacement uses `uinput` Kernel module, which means you need a background service running (or be happy with 1+ seconds of lag).

TL;DR:

- if you are on X11, use `xdotool type "your string"`
- if you are on Wayland, use any of the alternatives, the most common are:
  - [ydotool](https://github.com/ReimuNotMoe/ydotool)
  - [dotool](https://git.sr.ht/~geb/dotool)
  - [wtype](https://github.com/atx/wtype)

As for the input recording, that is out of scope for now, since I would need to introduce a completely new window for something that is already kinda possible to do with the manual delays.

closed 2023-03-17 by zocker-160