zocker-160/keyboard-center #37: keyboard-center.service did not install on Linux Mint 20.3

Corionis, 2022-09-27:
Stopped the service, uninstalled 1.28.1 (very old) then ran:  apt install keyboard-center_0.2.8-1_amd64.deb

Running  systemctl --user start keyboard-center.service  caused an error of not found.

Copied the service to /lib/systemd/user/ from source and then it worked.

Upgrade instructions would be useful.

zocker-160, 2022-09-27:
There is no systemd service anymore since 0.2.1, see the [updated instructions in the readme](https://github.com/zocker-160/keyboard-center#manage-background-service).

I have replaced it with a tray icon, so it can be managed more easily by ppl with no terminal experience.

Corionis, 2022-09-27:
Confirmed. Thank you.

closed 2022-09-27 by Corionis