zocker-160/keyboard-center #38: Trying to record Key Mappings for Logitech G613

computermillerj, 2023-01-06:
For some reason when I enter `lsusb` my keyboard shows up as Logitech G631, but it actually is a Logitech G613?

But the main issue is when I run the script `python3 src/newDeviceDebugger.py` the below is output

```
DEBUG: Searching for keyboard...
DEBUG: Keyboard found! :)
DEBUG: requesting USB endpoint...
DEBUG: HIDraw read endpoint found: /dev/hidraw5
DEBUG: HIDraw disable endpoint found: /dev/hidraw5
DEBUG: HIDraw read endpoint found: /dev/hidraw5
DEBUG: HIDraw disable endpoint found: /dev/hidraw5
DEBUG: HIDraw read endpoint found: /dev/hidraw5
DEBUG: HIDraw disable endpoint found: /dev/hidraw5
DEBUG: HIDraw read endpoint found: /dev/hidraw5
DEBUG: HIDraw disable endpoint found: /dev/hidraw5
DEBUG: HIDraw read endpoint found: /dev/hidraw5
DEBUG: HIDraw disable endpoint found: /dev/hidraw5
DEBUG: Checking for HID availability...
DEBUG: Connected to /dev/hidraw5
INFO: starting listener...
DEBUG: Using selector: EpollSelector
DEBUG: listening to USB Interface (1, 0) | b'/dev/hidraw5'
```

Then when I press G1 I get `^[OP` which is the same as when I press F1. Similar for G2 I get `^[OQ` which is the same as F2 and so on. What do I need to do to separate out the G keys from the F keys so I can record?

zocker-160, 2023-01-06:
> For some reason when I enter lsusb my keyboard shows up as Logitech G631, but it actually is a Logitech G613?

Do you have it plugged in via a USB? The G915 has a similar thing that the USB ID changes to a G815 when plugging it in via USB.

> Then when I press G1 I get ^[OP which is the same as when I press F1. Similar for G2 I get ^[OQ which is the same as F2 and so on. What do I need to do to separate out the G keys from the F keys so I can record?

Yes this is a common issue with Logitech keyboards, because they map the G-Keys to the F-keys by default and that mapping has to be disabled.

If you look into the script, there is a variable named `disableGKeys` which contains the command for disabling this exact mapping.
I would recommend you to try the ones out that are commented out, maybe you are lucky and your keyboard uses the same command as other already supported keyboards.

Looking at images, the G613 seems to look similar to my G815, so I would try the G915 disable keys command in the script.

You could also try the one from the G815:
```py
disableGKeys = [
    b'\x11\xff\x11\x1a\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00', 
    b'\x11\xff\n*\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
    b'\x11\xff\x0fZ\x01\x03\x07\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
    b'\x11\xff\x0f\x1a\x00\x02\x00\x00\x00\x00\x00\x084d\x00\x00\x01\x00\x00\x00',
    b'\x11\xff\x0f\x1a\x01\x04\x00\x00\x00\x00\x00\x004\x01d\x08\x01\x00\x00\x00',
    ]
```

computermillerj, 2023-01-06:
> Do you have it plugged in via a USB? 

Yes it is wireless via LIGHTSPEED receiver then plugged in via USB.

> EDIT: looking at your log, I wonder why it does not show the log entry Sending sequence to disable G keys, this could also be an issue.

This was my bad. I had already messed with line 130/131 because I was getting an error and wasnt sure if it was OK to try the disable commands for another keyboard. So I put it back to the original way and tried both the G915 and G815 and disableGKeysInterface = 1 and 0.

Below is what I get for all combinations of the above.
```
DEBUG: Searching for keyboard...
DEBUG: Keyboard found! :)
DEBUG: requesting USB endpoint...
DEBUG: HIDraw disable endpoint found: /dev/hidraw4
DEBUG: HIDraw read endpoint found: /dev/hidraw5
DEBUG: HIDraw read endpoint found: /dev/hidraw5
DEBUG: HIDraw read endpoint found: /dev/hidraw5
DEBUG: HIDraw read endpoint found: /dev/hidraw5
DEBUG: HIDraw read endpoint found: /dev/hidraw5
DEBUG: Checking for HID availability...
DEBUG: Connected to /dev/hidraw5
INFO: starting listener...
DEBUG: Using selector: EpollSelector
DEBUG: Connection using HIDAPI...
DEBUG: Sending sequence to disable G keys
0: b'\x11\x01\x11\x1a\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
ERROR: Task exception was never retrieved
future: <Task finished name='Task-1' coro=<usbListener() done, defined at /home/jem/repos/keyboard-center/src/newDeviceDebugger.py:135> exception=HIDException('Broken pipe')>
Traceback (most recent call last):
  File "/home/jem/repos/keyboard-center/src/newDeviceDebugger.py", line 144, in usbListener
    await disableGkeyMapping(HIDpathDisable)
  File "/home/jem/repos/keyboard-center/src/newDeviceDebugger.py", line 126, in disableGkeyMapping
    hdev.write(data)
  File "/home/jem/repos/keyboard-center/src/lib/hid.py", line 163, in write
    return self.__hidcall(hidapi.hid_write, self.__dev, data, len(data))
  File "/home/jem/repos/keyboard-center/src/lib/hid.py", line 154, in __hidcall
    raise HIDException(err)
lib.hid.HIDException: Broken pipe
```

Other bits of info that may/may not be helpful.

Script settings for my keyboard
```
## Logitech, Inc. G613
usbVendor = 0x046d
usbProduct = 0xc53d

```

Output from `python3 src/newDeviceDebugger.py --info`
```
DEBUG: Searching for keyboard...
DEBUG: Keyboard found! :)
DEBUG: requesting USB endpoint...
###
DEVICE ID 046d:c53d on Bus 001 Address 005 =================
 bLength                :   0x12 (18 bytes)
 bDescriptorType        :    0x1 Device
 bcdUSB                 :  0x200 USB 2.0
 bDeviceClass           :    0x0 Specified at interface
 bDeviceSubClass        :    0x0
 bDeviceProtocol        :    0x0
 bMaxPacketSize0        :   0x20 (32 bytes)
 idVendor               : 0x046d
 idProduct              : 0xc53d
 bcdDevice              : 0x4301 Device 67.01
 iManufacturer          :    0x1 Error Accessing String
 iProduct               :    0x2 Error Accessing String
 iSerialNumber          :    0x0 
 bNumConfigurations     :    0x1
  CONFIGURATION 1: 98 mA ===================================
   bLength              :    0x9 (9 bytes)
   bDescriptorType      :    0x2 Configuration
   wTotalLength         :   0x54 (84 bytes)
   bNumInterfaces       :    0x3
   bConfigurationValue  :    0x1
   iConfiguration       :    0x4 Error Accessing String
   bmAttributes         :   0xa0 Bus Powered, Remote Wakeup
   bMaxPower            :   0x31 (98 mA)
    INTERFACE 0: Human Interface Device ====================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x1
     bInterfaceClass    :    0x3 Human Interface Device
     bInterfaceSubClass :    0x1
     bInterfaceProtocol :    0x1
     iInterface         :    0x0 
      ENDPOINT 0x81: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :    0xc (12 bytes)
       bInterval        :    0x1
    INTERFACE 1: Human Interface Device ====================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x1
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x1
     bInterfaceClass    :    0x3 Human Interface Device
     bInterfaceSubClass :    0x1
     bInterfaceProtocol :    0x2
     iInterface         :    0x0 
      ENDPOINT 0x82: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x82 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x20 (32 bytes)
       bInterval        :    0x1
    INTERFACE 2: Human Interface Device ====================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x2
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x1
     bInterfaceClass    :    0x3 Human Interface Device
     bInterfaceSubClass :    0x0
     bInterfaceProtocol :    0x0
     iInterface         :    0x0 
      ENDPOINT 0x83: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x83 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x20 (32 bytes)
       bInterval        :    0x1
###
```

I am running on Kubuntu 22.04 and as far as I know have all the necessary packages. Can get version number if that would help.

zocker-160, 2023-01-07:
ok understood, the error in your log looks like you are not allowed to send data to the keyboard (which is normal without the udev rules in place). As noted in the readme, you might have to run it as root.

If you are already running as root, then the HID endpoint might be wrong, but to me it looks correct.

The main problem is, as with any new keyboard, we need to find out the command to disable the Gkey mapping and sadly there is no way for me to find that out by guessing.

All the commands for the keyboards so far I have found out by reversing the windows driver and Logitech GHub to see what data it sends when plugging it in.

computermillerj, 2023-01-07:
I seem to get this error message regardless of what I do, even as root. Thanks so much for the help here, I really appreciate it.

closed 2023-01-07 by computermillerj

zocker-160, 2023-01-07:
Do you have the same issue when plugging the keyboard in via USB directly?

I do remember having a very similar issue with the wireless G915

computermillerj, 2023-01-07:
This keyboard does not seem to have a hard-wired option at all. It is Lightspeed or Bluetooth only unless I am missing something...

zocker-160, 2023-01-07:
ah ok, I did not know that

I don't think there is a simple solution then I am afraid.

computermillerj, 2023-01-07:
No problem. Thanks for trying!!