zocker-160/keyboard-center #40: openrgb

HekTheLastSavior, 2023-03-29:
hey 
hope all is well.
is it possible to kill the openrgb process then executing again after restarting keyboard center...
maybe giving a few seconds delay between the two.
it will help fix my issue when my laptop wakes up and I need to do this process manually in order to get it to work,

thank you !!

zocker-160, 2023-03-29:
Which part of the application stops working after you wake from sleep?

In the UI you can select File > Force restart driver, which does completely reset the keyboard and the mappings but does also restart openRGB in case it was started internally.

I know that openRGB itself is a bit tricky as it doesn't always survive a sleep / wake up cycle and gets hung up. Currently I am sending a SIGINT signal to the OpenRGB process, if that is not enough, I could look into upping that to a SIGKILL.

HekTheLastSavior, 2023-03-29:
I'm aware of the force restart and I use it all the time.

But now I'm using openrgb and I have to exit from keyboard-center, killall -9 openrgb then wait like 5-10 seconds before opening keyboard-center and everything work accordingly

zocker-160, 2023-03-29:
Are you running an instance of OpenRGB separately? Is that instance getting hung up when you wake from sleep?

HekTheLastSavior, 2023-03-29:
not running a separate instance.
when the laptop resume from sleeping keyboard-center is active but not working so I need to force restart it .
I need to exit keyboard-center in order to kill openrgb that keyboard-center forked.

zocker-160, 2023-03-29:
The G-Keys do not work anymore after you wake from sleep?

Could you share the log file after you woke from sleep and triggered a force restart? It should reset the openrgb subprocess as well.

HekTheLastSavior, 2023-03-30:
G-keys stops working after the pc wake up.
we already had a discussion about this a while back and we couldn't figure out the problem.

regarding openrgb it's process is still available after closing keyboard-center. And do note that's I'm not running it myself.

zocker-160, 2023-03-30:
> we already had a discussion about this a while back and we couldn't figure out the problem.

Yes I remember now and I think that is why I added the reset option, but it clearly does not work as it should, so I wonder why it does not reset properly.

I would guess that it gets stuck at either `deleting BGService` or `stopping openRGB server...` in the log.

zocker-160, 2023-03-30:
yeah I know it is not great, but there is not really much I can do, since the device connection does not seem to resume correctly and running `kill` to get rid of the openrgb zombie would require root.

zocker-160, 2023-03-31:
I can actually reproduce the issue with openrgb background process not closing, I will look into it.

zocker-160, 2023-03-31:
fix is available now as 1.0.2

HekTheLastSavior, 2023-04-03:
no worries with the sleep issue...
thank you for your prompt update and help solving the issue.
kudos !!

closed 2023-04-03 by HekTheLastSavior