zocker-160/keyboard-center #42: Arch update broke the 'uinput'

lesu1, 2023-05-05:
Hi,
Not sure what did break the system (new Python update?), but I have some logs to share.  I hope they will be helpful to understand.

System: 
- Arch, 6.3.1-zen1-1-zen
- Python 3.11.3
- Keyboard G910

05.05.2023 05:52	systemd	Started Keyboard Center.
05.05.2023 05:52	plasmashell	Traceback (most recent call last):
05.05.2023 05:52	plasmashell	  File "/usr/lib/keyboard-center/main.py", line 8, in <module>
05.05.2023 05:52	plasmashell	    from mainUi import MainWindow
05.05.2023 05:52	plasmashell	  File "/usr/lib/keyboard-center/mainUi.py", line 23, in <module>
05.05.2023 05:52	plasmashell	    from devices.keyboard import SUPPORTED_DEVICES, KeyboardInterface
05.05.2023 05:52	plasmashell	  File "/usr/lib/keyboard-center/devices/keyboard.py", line 1, in <module>
05.05.2023 05:52	plasmashell	    import uinput
05.05.2023 05:52	plasmashell	ModuleNotFoundError: No module named 'uinput'

So I see that 'uinput' module is not found. 
I've tried to install it via: **pip install python-uinput**

And here's the log after that:
05.05.2023 05:55	systemd	Started Keyboard Center.
05.05.2023 05:55	plasmashell	Traceback (most recent call last):
05.05.2023 05:55	plasmashell	  File "/usr/lib/keyboard-center/main.py", line 8, in <module>
05.05.2023 05:55	plasmashell	    from mainUi import MainWindow
05.05.2023 05:55	plasmashell	  File "/usr/lib/keyboard-center/mainUi.py", line 23, in <module>
05.05.2023 05:55	plasmashell	    from devices.keyboard import SUPPORTED_DEVICES, KeyboardInterface
05.05.2023 05:55	plasmashell	  File "/usr/lib/keyboard-center/devices/keyboard.py", line 1, in <module>
05.05.2023 05:55	plasmashell	    import uinput
05.05.2023 05:55	plasmashell	  File "/home/jk/.local/lib/python3.11/site-packages/uinput/__init__.py", line 86, in <module>
05.05.2023 05:55	plasmashell	    _libsuinput_path = os.path.abspath(os.path.join(os.path.dirname(__file__), "..", "_libsuinput" + sysconfig.get_config_var("SO")))
05.05.2023 05:55	plasmashell	                                                                                     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
05.05.2023 05:55	plasmashell	TypeError: can only concatenate str (not "NoneType") to str

At this stage, my *nix knowledge is over. Let me know if I can help somehow.
But for now, it's a great time for a timemachine. :-)

Thanks, lesu1

lesu1, 2023-05-05:
A few cents more, after a timeshift restore, I have a Python 3.10.10 version, all good.
I will hold on to the Python update for now. 
lesu1

Aditeya, 2023-05-05:
I'm experiencing the same with Python 3.11.3

zocker-160, 2023-05-05:
thanks for reporting but it is a known issue with Python 3.11, see https://github.com/tuomasjjrasanen/python-uinput/pull/41

I need to check what I can do about it, since that module I (sadly) heavily rely on is dead. 
A short term solution would be to create a patch, long term I will have to move away from that dependency.

(I sadly won't have time to work on a solution of this until next week, sorry :( )

lesu1, 2023-05-10:
Interesting, today everything is working fine. I've booted my machine and wow. The Keyboard Center is working fine.
Still the same Python 3.11.3 in my system.
@Aditeya how about you?

lesu1

Aditeya, 2023-05-10:
I got an update for `python-uinput` from the [aur pkg](https://aur.archlinux.org/packages/python-uinput). Looks like they patched it, so it works!

zocker-160, 2023-05-10:
AUR maintainer of python-uinput finally got active and added the patch containing the PR needed to add support for Python 3.11.

However please leave this issue open, as I need to add this patch to the Ubuntu package as well.

zocker-160, 2023-05-12:
patched `python3-uinput` now also available for Debian and Ubuntu LTS

closed 2023-05-12 by zocker-160