zocker-160/keyboard-center #44: stopped working on Garuda Linux a couple weeks ago

justTravis, 2023-05-17:
No errors shown.  The window tries to appear for a moment then disappears without ending up in the tray.

I have a Logitech G910 keyboard.

Kernel: 6.3.1-zen2-1-zen
DE: Plasma 5.27.5
WM: KWin

Python 3.11.3

if I run "keyboard-center" from the terminal, I see this:

Traceback (most recent call last):
  File "/usr/lib/keyboard-center/main.py", line 8, in <module>
    from mainUi import MainWindow
  File "/usr/lib/keyboard-center/mainUi.py", line 23, in <module>
    from devices.keyboard import SUPPORTED_DEVICES, KeyboardInterface
  File "/usr/lib/keyboard-center/devices/keyboard.py", line 1, in <module>
    import uinput
  File "/usr/lib/python3.11/site-packages/python_uinput-0.11.2-py3.11-linux-x86_64.egg/uinput/__init__.py", line 86, in <module>
    _libsuinput_path = os.path.abspath(os.path.join(os.path.dirname(__file__), "..", "_libsuinput" + sysconfig.get_config_var("SO")))
                                                                                     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
TypeError: can only concatenate str (not "NoneType") to str

zocker-160, 2023-05-17:
This is a duplicate of #42.

How did you install it? AUR already has the patch included, rebuilding python-uinput should fix it (see the linked issue).

justTravis, 2023-05-17:
> This is a duplicate of #42.
> 
> How did you install it? AUR already has the patch included, rebuilding python-uinput should fix it (see the linked issue).

thank you!  a simple "yay" command got it working!

closed 2023-05-17 by zocker-160