zocker-160/keyboard-center #46: HIDpath error

justTravis, 2023-05-17:
I am getting the following error:

[root] ERROR: 17.05.2023 12:23:27 PM cannot access local variable 'HIDpath' where it is not associated with a value
Traceback (most recent call last):
  File "/usr/lib/keyboard-center/main.py", line 52, in <module>
    window = MainWindow(app, devmode, not bgmode)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/keyboard-center/mainUi.py", line 68, in __init__
    self.initBackgroundService()
  File "/usr/lib/keyboard-center/mainUi.py", line 153, in initBackgroundService
    self.service = BackgroundService(self.configparser, not self.devmode)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/keyboard-center/service.py", line 85, in __init__
    self._initKeyboard()
  File "/usr/lib/keyboard-center/service.py", line 126, in _initKeyboard
    self._getHIDpaths()
  File "/usr/lib/keyboard-center/service.py", line 166, in _getHIDpaths
    if HIDpath and not __HIDavailable(HIDpath, numTries):
       ^^^^^^^
UnboundLocalError: cannot access local variable 'HIDpath' where it is not associated with a value
Traceback (most recent call last):
  File "/usr/lib/keyboard-center/main.py", line 52, in <module>
    window = MainWindow(app, devmode, not bgmode)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/keyboard-center/mainUi.py", line 68, in __init__
    self.initBackgroundService()
  File "/usr/lib/keyboard-center/mainUi.py", line 153, in initBackgroundService
    self.service = BackgroundService(self.configparser, not self.devmode)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/keyboard-center/service.py", line 85, in __init__
    self._initKeyboard()
  File "/usr/lib/keyboard-center/service.py", line 126, in _initKeyboard
    self._getHIDpaths()
  File "/usr/lib/keyboard-center/service.py", line 166, in _getHIDpaths
    if HIDpath and not __HIDavailable(HIDpath, numTries):
       ^^^^^^^
UnboundLocalError: cannot access local variable 'HIDpath' where it is not associated with a value

I have a Logitech G910 keyboard.

Kernel: 6.3.1-zen2-1-zen
DE: Plasma 5.27.5
WM: KWin

Python 3.11.3

zocker-160, 2023-05-17:
This is a new one, I have never seen this error before :V

It did work for you in the past, can you maybe say a bit more about when this happened? Did you unplug the keyboard or go to sleep / hibernate?

justTravis, 2023-05-17:
I added myself to the "docker" group so I wouldn't have to sudo docker commands.  It didn't seem to work, so I rebooted to see if it would, and it did, BUT....  this error appeared with keyboard-center.

Docker was already installed, I just wasn't in the group to allow using it without sudo.

zocker-160, 2023-05-17:
Can you please check with `id` command in terminal, that you are still part of the `plugdev` group?

The exception above is a bug, it should show a proper error message, but libhidraw not working is a permission issue on your side.

justTravis, 2023-05-17:
hmmm, no I am not.  

and the group doesn't even exist - should I just manually add it and then add myself to it?

zocker-160, 2023-05-17:
If the group does not exist, then creating it will probably not fix the issue.

Can you check if running as root also gives that error? If not then this is 100% a permission issue.

Please check that the udev rules are still in place. I sadly do not know why docker would break something like this, but it did work in the past, so something took away the read permission from your user.

You could also try reinstalling as the install does trigger udevadm, so it could fix it.

justTravis, 2023-05-17:
sudo gives the same error.  I tried uninstalling and reinstalling, no dice.

zocker-160, 2023-05-18:
Did you install any other software which does access keyboard?

I need the output of `sudo ls -l /dev/hid*` and `sudo evtest`

justTravis, 2023-05-18:
I didn't install anything in between when it worked and when it stopped working.  I do have another piece of software called input-remapper that works with a Razer Tartarus, but it has never interfered before that I know of.

The output of the ls command is:
crw-rw----+ 1 root root 240,  0 May 17 16:44 /dev/hidraw0
crw-rw----+ 1 root root 240,  1 May 17 16:44 /dev/hidraw1
crw-rw----+ 1 root root 240, 10 May 17 16:44 /dev/hidraw10
crw-------  1 root root 240, 12 May 17 16:44 /dev/hidraw12
crw-rw----+ 1 root root 240,  2 May 17 16:44 /dev/hidraw2
crw-rw----+ 1 root root 240,  3 May 17 16:44 /dev/hidraw3
crw-rw----+ 1 root root 240,  4 May 17 16:44 /dev/hidraw4
crw-------  1 root root 240,  5 May 17 16:44 /dev/hidraw5
crw-------  1 root root 240,  6 May 17 16:44 /dev/hidraw6
crw-rw----+ 1 root root 240,  7 May 17 16:44 /dev/hidraw7
crw-rw----+ 1 root root 240,  8 May 17 16:44 /dev/hidraw8
crw-rw----+ 1 root root 240,  9 May 17 16:44 /dev/hidraw9

And the output of the evtest is:
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      Sleep Button
/dev/input/event1:      Power Button
/dev/input/event10:     HDA NVidia HDMI/DP,pcm=7
/dev/input/event11:     HDA NVidia HDMI/DP,pcm=8
/dev/input/event12:     HDA NVidia HDMI/DP,pcm=9
/dev/input/event13:     python-uinput
/dev/input/event14:     Valve Software Steam Controller
/dev/input/event15:     HID 0c45:7403
/dev/input/event16:     HID 0c45:7403
/dev/input/event17:     Razer Razer Tartarus V2
/dev/input/event18:     Razer Razer Tartarus V2 Keyboard
/dev/input/event19:     Razer Razer Tartarus V2
/dev/input/event2:      Power Button
/dev/input/event20:     Razer Razer Tartarus V2
/dev/input/event21:     Logitech Gaming Keyboard G910
/dev/input/event22:     NexiGo HD Webcam: NexiGo HD Web
/dev/input/event23:     Kingston HyperX Cloud Stinger Wireless Consumer Control
/dev/input/event24:     Kingston HyperX Cloud Stinger Wireless
/dev/input/event25:     Kingston HyperX Cloud Stinger Wireless
/dev/input/event26:     Kingston HyperX Cloud Stinger Wireless
/dev/input/event3:      Eee PC WMI hotkeys
/dev/input/event4:      HDA Intel PCH Front Mic
/dev/input/event5:      HDA Intel PCH Rear Mic
/dev/input/event6:      HDA Intel PCH Line
/dev/input/event7:      HDA Intel PCH Line Out
/dev/input/event8:      HDA Intel PCH Front Headphone
/dev/input/event9:      HDA NVidia HDMI/DP,pcm=3

Thank you so much for the effort!

zocker-160, 2023-05-18:
I pushed a new version, it will not fix your issue, but it should fix the broken error message.

As for your issue, I need even more information, as the shown above looks ok to me, can you pls provide the output of this script:
```bash
#!/bin/bash

FILES=/dev/hidraw*
for f in $FILES
do
  FILE=${f##*/}
  DEVICE="$(cat /sys/class/hidraw/${FILE}/device/uevent | grep HID_NAME | cut -d '=' -f2)"
  printf "%s \t %s\n" $FILE "$DEVICE"
done
```
It maps the HID endpoint to a physical device, this one should show which endpoint is related to the G910 keyboard.

justTravis, 2023-05-18:
I installed the new version and as you said it still fails.  The error message is now:
[root] ERROR: 18.05.2023 07:41:00 AM
Traceback (most recent call last):
  File "/usr/lib/keyboard-center/main.py", line 52, in <module>
    window = MainWindow(app, devmode, not bgmode)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/keyboard-center/mainUi.py", line 68, in __init__
    self.initBackgroundService()
  File "/usr/lib/keyboard-center/mainUi.py", line 153, in initBackgroundService
    self.service = BackgroundService(self.configparser, not self.devmode)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/keyboard-center/service.py", line 85, in __init__
    self._initKeyboard()
  File "/usr/lib/keyboard-center/service.py", line 126, in _initKeyboard
    self._getHIDpaths()
  File "/usr/lib/keyboard-center/service.py", line 171, in _getHIDpaths
    raise NoEndpointException()
service.NoEndpointException
Traceback (most recent call last):
  File "/usr/lib/keyboard-center/main.py", line 52, in <module>
    window = MainWindow(app, devmode, not bgmode)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/keyboard-center/mainUi.py", line 68, in __init__
    self.initBackgroundService()
  File "/usr/lib/keyboard-center/mainUi.py", line 153, in initBackgroundService
    self.service = BackgroundService(self.configparser, not self.devmode)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/keyboard-center/service.py", line 85, in __init__
    self._initKeyboard()
  File "/usr/lib/keyboard-center/service.py", line 126, in _initKeyboard
    self._getHIDpaths()
  File "/usr/lib/keyboard-center/service.py", line 171, in _getHIDpaths
    raise NoEndpointException()
service.NoEndpointException

And the output of your script is:
hidraw0          Valve Software Steam Controller
hidraw1          Valve Software Steam Controller
hidraw10         Logitech Gaming Keyboard G910
hidraw12         Kingston HyperX Cloud Stinger Wireless
hidraw2          Valve Software Steam Controller
hidraw3          Valve Software Steam Controller
hidraw4          Valve Software Steam Controller
hidraw5          HID 0c45:7403
hidraw6          HID 0c45:7403
hidraw7          Razer Razer Tartarus V2
hidraw8          Razer Razer Tartarus V2
hidraw9          Razer Razer Tartarus V2

zocker-160, 2023-05-18:
Thanks for the output, yeah there is a bug somewhere which prevents the script of finding the right endpoint, which is weird since it seemed to be working in the past.

Also could you send me your log file? (should be located in `~/.var/log/keyboardCenter.log`)

I sadly am a bit out of ideas, I need to add a more logging to find out what is going on.

I know it might sound weird, but could you try to reboot without the Steam Controller plugged in? 

I have a suspicion that I have a bug in the code which fails to detect the keyboard when the hidraw id is > 9

justTravis, 2023-05-18:
hmmm I can't reboot just yet due to I'm in the middle of meetings at work.  but I will later!  and here is the log you asked for.

[keyboardCenter.log](https://github.com/zocker-160/keyboard-center/files/11508778/keyboardCenter.log)

justTravis, 2023-05-19:
sadly the "reboot without the steam controller" didn't help.

I re-ran the script you gave and got this output:
hidraw0          HID 0c45:7403
hidraw1          HID 0c45:7403
hidraw2          Razer Razer Tartarus V2
hidraw3          Razer Razer Tartarus V2
hidraw4          Razer Razer Tartarus V2
hidraw5          Logitech Gaming Keyboard G910
hidraw7          Kingston HyperX Cloud Stinger Wireless

and running (as sudo) the keyboard-center app shows this:
[root] ERROR: 19.05.2023 07:03:33 AM
Traceback (most recent call last):
  File "/usr/lib/keyboard-center/main.py", line 52, in <module>
    window = MainWindow(app, devmode, not bgmode)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/keyboard-center/mainUi.py", line 68, in __init__
    self.initBackgroundService()
  File "/usr/lib/keyboard-center/mainUi.py", line 153, in initBackgroundService
    self.service = BackgroundService(self.configparser, not self.devmode)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/keyboard-center/service.py", line 85, in __init__
    self._initKeyboard()
  File "/usr/lib/keyboard-center/service.py", line 126, in _initKeyboard
    self._getHIDpaths()
  File "/usr/lib/keyboard-center/service.py", line 171, in _getHIDpaths
    raise NoEndpointException()
service.NoEndpointException
Traceback (most recent call last):
  File "/usr/lib/keyboard-center/main.py", line 52, in <module>
    window = MainWindow(app, devmode, not bgmode)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/keyboard-center/mainUi.py", line 68, in __init__
    self.initBackgroundService()
  File "/usr/lib/keyboard-center/mainUi.py", line 153, in initBackgroundService
    self.service = BackgroundService(self.configparser, not self.devmode)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/keyboard-center/service.py", line 85, in __init__
    self._initKeyboard()
  File "/usr/lib/keyboard-center/service.py", line 126, in _initKeyboard
    self._getHIDpaths()
  File "/usr/lib/keyboard-center/service.py", line 171, in _getHIDpaths
    raise NoEndpointException()
service.NoEndpointException

zocker-160, 2023-05-20:
Thank you for this, so my original suspicion did not materialize. I will create a special debugging build for you.

I would however be interested in the following output to see what kernel log says while registering the keyboard:
`sudo dmesg | grep G910`

justTravis, 2023-05-21:
that command returns nothing.  if I ignore the grep and just scan the output there is no mention of the G910 or Logitech or anything identifiable to me.

as a side note. I am now dual booting into ArcoLinux and it works fine even with all updates.  But I experimented with a few other distros, and it is worth noting that NONE of the debian flavors work.  Kubuntu, Mint, MX, and some others... I've not yet seen the .DEB file work at all.  This is what is keeping me on Arch based distros even though I think I may not be the target audience for the instability.

zocker-160, 2023-05-22:
> that command returns nothing. if I ignore the grep and just scan the output there is no mention of the G910 or Logitech or anything identifiable to me.

This is weird, I am so confused. On all of my machines I can see how the keyboard is being detected and registered in the hidraw system by the kernel. It should look something like this:

```
[    2.796164] usb 5-4.3: Product: G815 RGB MECHANICAL GAMING KEYBOARD
[    5.005368] input: Logitech G815 RGB MECHANICAL GAMING KEYBOARD as /devices/pci0000:00/0000:00:08.1/0000:0c:00.3/usb5/5-4/5-4.3/5-4.3:1.0/0003:046D:C33F.0001/input/input2
[    5.066078] hid-generic 0003:046D:C33F.0001: input,hidraw0: USB HID v1.11 Keyboard [Logitech G815 RGB MECHANICAL GAMING KEYBOARD] on usb-0000:0c:00.3-4.3/input0
[    5.066188] input: Logitech G815 RGB MECHANICAL GAMING KEYBOARD Keyboard as /devices/pci0000:00/0000:00:08.1/0000:0c:00.3/usb5/5-4/5-4.3/5-4.3:1.1/0003:046D:C33F.0002/input/input3
[    5.130071] input: Logitech G815 RGB MECHANICAL GAMING KEYBOARD Mouse as /devices/pci0000:00/0000:00:08.1/0000:0c:00.3/usb5/5-4/5-4.3/5-4.3:1.1/0003:046D:C33F.0002/input/input4
[    5.130156] hid-generic 0003:046D:C33F.0002: input,hiddev0,hidraw1: USB HID v1.11 Keyboard [Logitech G815 RGB MECHANICAL GAMING KEYBOARD] on usb-0000:0c:00.3-4.3/input1
[   80.865259] input: Logitech G815 (keyboard-center) as /devices/virtual/input/input25
```

Regarding the deb, it is build for Ubuntu LTS and I am personally running it on Kubuntu. Do you have any error etc when trying to run on Ubuntu LTS? It is possible that I messed up the deb and needs fixing asap.

I cannot speak of derivates like Mint or MX as it would be very time consuming having a test setup for all of them.

justTravis, 2023-06-20:
I switched to Arco Linux and to XFCE and it's working fine.  Some games run less well than they did on Garuda but keyboard center works so that's a better deal for me :)  I have just started a new job so have not yet had time to make a side install of Ubuntu or any Debian based system to test again with.  Now that I'm reading this again, I recall I do have a laptop around with Kubuntu.  I'll try to give that a shot today.

justTravis, 2023-06-23:
It is working fine on my Kubuntu.  It seems like the problem is one-off on my particular Garuda installation :(  Probably not worth more of your time.  I do very much appreciate your efforts though!

zocker-160, 2023-06-23:
thank you for testing!

closing this issue then