zocker-160/keyboard-center #47: Keyboard Center Fails to Launch.

Refrained-LM, 2023-06-05:
I try to launch Keyboard Center with the command "keyboard-center %f " and this is the error that I got below

Traceback (most recent call last):
  File "/usr/lib/keyboard-center/main.py", line 8, in <module>
    from mainUi import MainWindow
  File "/usr/lib/keyboard-center/mainUi.py", line 23, in <module>
    from devices.keyboard import SUPPORTED_DEVICES, KeyboardInterface
  File "/usr/lib/keyboard-center/devices/keyboard.py", line 1, in <module>
    import uinput
  File "/usr/lib/python3.11/site-packages/uinput/__init__.py", line 86, in <module>
    _libsuinput_path = os.path.abspath(os.path.join(os.path.dirname(__file__), "..", "_libsuinput" + sysconfig.get_config_var("SO")))
                                                                                     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
TypeError: can only concatenate str (not "NoneType") to str

Im running.

OS: Manjaro 23.0.0 UltimaThule
Kernel: x86_64 Linux 5.15.114-2-MANJARO
Shell: zsh 5.9
Resolution: 5360x1440
DE: KDE 5.106.0 / Plasma 5.27.5
WM: KWin
Disk: 8.3T / 14T (64%)
CPU: AMD Ryzen 5 5600X 6-Core @ 12x 3.693GHz
GPU: Advanced Micro Devices, Inc. [AMD/ATI] Navi 21 [Radeon RX 6800/6800 XT / 6900 XT] (rev c1)
RAM: 4211MiB / 64208MiB                                

was working last week I don't know what happened or how to fix.  I have rebuilt/reinstalled but that didn't work please let me know of any more info is needed

zocker-160, 2023-06-05:
This is a duplicate of #42.

Manjaro moved to Python 3.11 yesterday, which is why you are seeing this error.
You need to rebuild the `python-uinput` AUR package, then it should work again.

Refrained-LM, 2023-06-05:
Yup All good like nothing ever happened.

closed 2023-06-05 by zocker-160