zocker-160/keyboard-center #48: Error during init of keyboard

stefexec, 2023-06-16:
Thanks for the great work although i am getting this error:

[image]

I have installed 

https://github.com/JSubelj/g910-gkey-macro-support

previously but it is currently disabled as i discovered your amazing softeware after ive already installed that one.
Could this be related? What information do you need from me?

zocker-160, 2023-06-16:
Please try a reboot and don't start `g910-gkey-macro-support`.

The error most likely is cause by the fact that `g910-gkey-macro-support` unloads the kernel driver which causes keyboard-center to not be able to find the keyboard generating the error above.

A reboot will re-enable the kernel driver and should make your keyboard detectable again.

stefexec, 2023-06-17:
I love you.

closed 2023-06-17 by stefexec