zocker-160/keyboard-center #49: Software crashing on start

HekTheLastSavior, 2023-06-20:
I'm not sure what changed, perhaps python was updated on my system, but keyboard center is no longer starting and I'm getting the following error.

```zsh
❯ keyboard-center

Traceback (most recent call last):

  File "/usr/lib/keyboard-center/main.py", line 8, in <module>

    from mainUi import MainWindow

  File "/usr/lib/keyboard-center/mainUi.py", line 23, in <module>

    from devices.keyboard import SUPPORTED_DEVICES, KeyboardInterface

  File "/usr/lib/keyboard-center/devices/keyboard.py", line 1, in <module>

    import uinput

  File "/usr/lib/python3.11/site-packages/uinput/__init__.py", line 86, in <module>

    _libsuinput_path = os.path.abspath(os.path.join(os.path.dirname(__file__), "..", "_libsuinput" + sysconfig.get_config_var("SO")))

                                                                                     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

TypeError: can only concatenate str (not "NoneType") to str

```

Any suggestions ?
Thank you!

zocker-160, 2023-06-20:
If you are running Arch based system, then this is a duplicate of #42 (and #44, #47 for that matter)

HekTheLastSavior, 2023-06-21:
Thank you for the reply, yes I'm running Manajaro. 
I got it fixed after reading the other issues by installing python-uinput from AUR instead of the official package by manjaro. The weird part is that they are both updated recently.
Thank you for the support and keep up the good work!

closed 2023-06-21 by HekTheLastSavior

zocker-160, 2023-06-21:
> The weird part is that they are both updated recently.

Interesting, I need to take a look, but the AUR version does contain a fix for the issue above, the one in the official Manjaro repository probably does not have.