zocker-160/keyboard-center #50: TypeError: can only concatenate str

recallmenot, 2023-06-30:
Hi, I just updated to 1.0.3-1 (Manjaro) and when opening keyboard center, I get the following error message:
```
keyboard-center
Traceback (most recent call last):
  File "/usr/lib/keyboard-center/main.py", line 8, in <module>
    from mainUi import MainWindow
  File "/usr/lib/keyboard-center/mainUi.py", line 23, in <module>
    from devices.keyboard import SUPPORTED_DEVICES, KeyboardInterface
  File "/usr/lib/keyboard-center/devices/keyboard.py", line 1, in <module>
    import uinput
  File "/usr/lib/python3.11/site-packages/uinput/__init__.py", line 86, in <module>
    _libsuinput_path = os.path.abspath(os.path.join(os.path.dirname(__file__), "..", "_libsuinput" + sysconfig.get_config_var("SO")))
                                                                                     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
TypeError: can only concatenate str (not "NoneType") to str
```
Maybe another package got updated too? I just ran system updates..

recallmenot, 2023-06-30:
sorry duplicate of https://github.com/zocker-160/keyboard-center/issues/49

closed 2023-06-30 by recallmenot

recallmenot, 2023-06-30:
so I figured it out:
`pacman -Rdd python-uinput`
then open pamac (Software), python-uinput from AUR (which includes a fix) will now be available where it was previously hidden due to identical name with official repo version
now keyboard-center works again

zocker-160, 2023-06-30:
@recallmenot I just want to add, that after reporting the issue to Manjaro, they have moved that package from community to extra and have included the needed patch. (sadly it is still in unstable)

So once it moves to stable, you should be able to switch to the repository package.

recallmenot, 2023-06-30:
awesome, thanks for the info!