zocker-160/keyboard-center #51: please add support for MX-Linux

justTravis, 2023-07-11:
MX is based on Debian so it seems like it should work, but it doesn't.  

The error it fails with is that it cannot find the module "package"

zocker-160, 2023-07-11:
Is that error during installation or when trying to run it?

justTravis, 2023-07-12:
When trying to run it.  The install seems to go fine.  Running it by double-clicking the icon just seems to "do nothing."  Running "sudo keyboard-center" gives the error I mentioned.

justTravis, 2023-07-12:
and FYI the same problem happens in LMDE (Linux Mint Debian Edition).

zocker-160, 2023-07-12:
Would it be possible for you to give me an actual terminal output, because I sadly currently really don't know what `cannot find the module "package"` is supposed to mean since "package" could be anything :(.

I will try to reproduce this later today on LMDE.

justTravis, 2023-07-13:
Yes.  My apologies.  I've tried now on numerous Debian based distributions (trying to wean myself from Arch based) and all in common have this same error message:

Traceback (most recent call last):
  File "/usr/lib/keyboard-center/main.py", line 8, in <module>
    from mainUi import MainWindow
  File "/usr/lib/keyboard-center/mainUi.py", line 5, in <module>
    from packaging import version
ModuleNotFoundError: No module named 'packaging'

zocker-160, 2023-07-13:
ahhh now I understand the issue, thank you.

Will fix asap.

zocker-160, 2023-07-13:
I pushed a new version, which removes the dependency from packaging.version. Let me know if that works for you.

justTravis, 2023-07-14:
works for me now!  thanks!!!