zocker-160/keyboard-center #53: Feature Request: Cycle Commands

Doomer-Cat, 2023-10-29:
1. press key
2. 1st command executes
3. press key again
4. 2nd command executes
5. press key again
6. 1st command executes
... and so on 

Maybe like a checkbox or something

zocker-160, 2023-10-29:
Understood, this should be doable. 

Only minor issue is, that I am not sure what you as a user would expect to happen if one or more of the actions is a delay, skip it?

Doomer-Cat, 2023-11-07:
> Understood, this should be doable.
> 
> Only minor issue is that I am not sure what you as a user would expect to happen if one or more of the actions is a delay, skip it?
I'd say skip it, if one would really want to add one there is still the `sleep .3s && command`  option right

zocker-160, 2024-09-05:
I know it has been a while my implementation of this request is not a checkbox as it would make things very complicated for me if I add a separate checkbox for each possible variation requested.

So with version v2 you can now use lua scripting to implement what you are asking for, I have created an example script [here](https://github.com/zocker-160/keyboard-center/blob/master/lua_examples/05-globalregister.lua).

closed 2024-09-05 by zocker-160