zocker-160/keyboard-center #54: Using Super Key as Custom Modifier Deletes All Settings

scott-carrion, 2023-11-23:
Attempting to use the super (Windows) key as a custom modifier for a macro results in the settings.yml file being completely deleted.

Using Debian Bookworm and i3 (with GNOME installed as well), keyboard-center version 1.0.5.

I don't know if this has anything to do with it, but I have a Logitech G815 keyboard.

[image]

zocker-160, 2023-11-24:
oh god thanks for reporting, this is really bad, will fix asap.

The Window / Super key should not be allowed int the custom modifier in the first place as it is already under the primary modifiers under "Meta". 
Maybe I should rename that button to make it more clear.

zocker-160, 2023-12-02:
fixed in 1.0.6

closed 2023-12-02 by zocker-160

scott-carrion, 2023-12-06:
Just wanted to confirm that this issue is indeed fixed in 1.0.6 (great turnaround time by the way!)

I retested and now see that the "Meta/Super" key is clearly labeled now, and that upon hovering over the button, it now reads "aka the Windows key".

Ignoring this and trying to use the Super (aka Windows) key as a custom modifier is correctly denied: Nothing happens when you press it.