zocker-160/keyboard-center #56: Question: Missing explanation of the config file

Yaakoss, 2024-01-01:
Hi,
i'm trying the software and i'd like to setup G1-G9 to be used as F13-F21.'
I looked at the config file and it seems to be pretty simple, but there is one thing i do not understand.
what is value for?

Thanks in advance

zocker-160, 2024-01-01:
value is either a tuple or a list of tuples where the first value is the type of the key / entry and the second one is the keycode as defined in `linux/input-event-codes.h` of your Linux kernel.

For mine (5.15) F13 - F24 is defined like this:
```
#define KEY_F13                 183
#define KEY_F14                 184
#define KEY_F15                 185
#define KEY_F16                 186
#define KEY_F17                 187
#define KEY_F18                 188
#define KEY_F19                 189
#define KEY_F20                 190
#define KEY_F21                 191
#define KEY_F22                 192
#define KEY_F23                 193
#define KEY_F24                 194
```

Yaakoss, 2024-01-01:
Thank you very much. It works now:) I need to use Xmodmap to remap them but its is doing what i want.

closed 2024-01-01 by Yaakoss