zocker-160/keyboard-center #58: Support for G510 - 2011 (046d:c22e)

JoZ3, 2024-02-15:
Hi I have a Logitech G510 from 2011, but it is different from the one sent by @Flying--Dutchman, because they have different USB ID, so the application does not detect it.

The USB ID is the following: **046d:c22e**

**python3 src/newDeviceDebugger.py --info:**

```
DEBUG: Searching for keyboard...
DEBUG: Keyboard found! :)
DEBUG: requesting USB endpoint...
###
DEVICE ID 046d:c22e on Bus 005 Address 003 =================
 bLength                :   0x12 (18 bytes)
 bDescriptorType        :    0x1 Device
 bcdUSB                 :  0x200 USB 2.0
 bDeviceClass           :    0x0 Specified at interface
 bDeviceSubClass        :    0x0
 bDeviceProtocol        :    0x0
 bMaxPacketSize0        :    0x8 (8 bytes)
 idVendor               : 0x046d
 idProduct              : 0xc22e
 bcdDevice              :  0x165 Device 1.65
 iManufacturer          :    0x1 Error Accessing String
 iProduct               :    0x2 Error Accessing String
 iSerialNumber          :    0x0 
 bNumConfigurations     :    0x1
  CONFIGURATION 1: 500 mA ==================================
   bLength              :    0x9 (9 bytes)
   bDescriptorType      :    0x2 Configuration
   wTotalLength         :  0x122 (290 bytes)
   bNumInterfaces       :    0x5
   bConfigurationValue  :    0x1
   iConfiguration       :    0x0 
   bmAttributes         :   0xa0 Bus Powered, Remote Wakeup
   bMaxPower            :   0xfa (500 mA)
    INTERFACE 0: Human Interface Device ====================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x1
     bInterfaceClass    :    0x3 Human Interface Device
     bInterfaceSubClass :    0x1
     bInterfaceProtocol :    0x1
     iInterface         :    0x0 
      ENDPOINT 0x81: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :    0x8 (8 bytes)
       bInterval        :    0x4
    INTERFACE 1: Human Interface Device ====================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x1
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x2
     bInterfaceClass    :    0x3 Human Interface Device
     bInterfaceSubClass :    0x0
     bInterfaceProtocol :    0x0
     iInterface         :    0x0 
      ENDPOINT 0x82: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x82 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :    0x8 (8 bytes)
       bInterval        :    0x4
      ENDPOINT 0x3: Interrupt OUT ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x3 OUT
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x4
    INTERFACE 2: Audio =====================================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x2
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x0
     bInterfaceClass    :    0x1 Audio
     bInterfaceSubClass :    0x1
     bInterfaceProtocol :    0x0
     iInterface         :    0x0 
    INTERFACE 3: Audio =====================================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x3
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x0
     bInterfaceClass    :    0x1 Audio
     bInterfaceSubClass :    0x2
     bInterfaceProtocol :    0x0
     iInterface         :    0x0 
    INTERFACE 3, 1: Audio ==================================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x3
     bAlternateSetting  :    0x1
     bNumEndpoints      :    0x1
     bInterfaceClass    :    0x1 Audio
     bInterfaceSubClass :    0x2
     bInterfaceProtocol :    0x0
     iInterface         :    0x0 
      ENDPOINT 0x5: Isochronous OUT ========================
       bLength          :    0x9 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x5 OUT
       bmAttributes     :    0x9 Isochronous
       wMaxPacketSize   :   0x64 (100 bytes)
       bInterval        :    0x1
    INTERFACE 3, 2: Audio ==================================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x3
     bAlternateSetting  :    0x2
     bNumEndpoints      :    0x1
     bInterfaceClass    :    0x1 Audio
     bInterfaceSubClass :    0x2
     bInterfaceProtocol :    0x0
     iInterface         :    0x0 
      ENDPOINT 0x5: Isochronous OUT ========================
       bLength          :    0x9 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x5 OUT
       bmAttributes     :    0x9 Isochronous
       wMaxPacketSize   :   0xc8 (200 bytes)
       bInterval        :    0x1
    INTERFACE 4: Audio =====================================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x4
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x0
     bInterfaceClass    :    0x1 Audio
     bInterfaceSubClass :    0x2
     bInterfaceProtocol :    0x0
     iInterface         :    0x0 
    INTERFACE 4, 1: Audio ==================================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x4
     bAlternateSetting  :    0x1
     bNumEndpoints      :    0x1
     bInterfaceClass    :    0x1 Audio
     bInterfaceSubClass :    0x2
     bInterfaceProtocol :    0x0
     iInterface         :    0x0 
      ENDPOINT 0x86: Isochronous IN ========================
       bLength          :    0x9 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x86 IN
       bmAttributes     :    0x1 Isochronous
       wMaxPacketSize   :   0x80 (128 bytes)
       bInterval        :    0x1
###
```

But when I run **python3 src/newDeviceDebugger.py** I get an error:

```
DEBUG: Searching for keyboard...
DEBUG: Keyboard found! :)
DEBUG: requesting USB endpoint...
DEBUG: Checking for HID availability...
INFO: starting listener...
DEBUG: Using selector: EpollSelector
DEBUG: Connection using HIDAPI...
ERROR: Task exception was never retrieved
future: <Task finished name='Task-1' coro=<usbListener() done, defined at /home/myuser/keyboard-center/src/newDeviceDebugger.py:131> exception=ValueError('specify vid/pid or path')>
Traceback (most recent call last):
  File "/home/myuser/keyboard-center/src/newDeviceDebugger.py", line 140, in usbListener
    await disableGkeyMapping(HIDpathDisable)
  File "/home/myuser/keyboard-center/src/newDeviceDebugger.py", line 116, in disableGkeyMapping
    with HIDDevice(path=HIDpath) as hdev:
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/myuser/keyboard-center/src/lib/hid.py", line 135, in __init__
    raise ValueError('specify vid/pid or path')
ValueError: specify vid/pid or path
```

I tried to comment out the default G815 **disableGKeys** lines and uncomment the G510 lines, but I get the same error.

zocker-160, 2024-02-15:
very weird, it is failing to connect to the hid endpoint, have you tried running as root? 

Either you are missing the needed udev rules or you have some other application running, which is requesting exclusive access to the USB device, in which case this won't work either.

zocker-160, 2024-02-15:
I have updated the debugger script, please retry with that using root.

JoZ3, 2024-02-15:
Hi, thanks for the fast reply, I run the script from root and get the following error:

```
DEBUG: Searching for keyboard...
DEBUG: Keyboard found! :)
DEBUG: requesting USB endpoint...
Traceback (most recent call last):
  File "/home/myuser/keyboard-center/src/newDeviceDebugger.py", line 230, in <module>
    main()
  File "/home/myuser/keyboard-center/src/newDeviceDebugger.py", line 210, in main
    HIDpath, HIDpathDisable = getHIDpaths()
                              ^^^^^^^^^^^^^
  File "/home/myuser/keyboard-center/src/newDeviceDebugger.py", line 99, in getHIDpaths
    raise RuntimeError("Failed to find HIDpath (does other app have exclusive access?)")
RuntimeError: Failed to find HIDpath (does other app have exclusive access?)
```

I don't know if it will influence, but I have g15daemon on my system, I will uninstall it and run the script again.

zocker-160, 2024-02-15:
yeah ok with the new error message it seems like there is something else having exclusive access to the USB device.

You don't actually need to uninstall it, doing a reboot and making sure it does not start should be enough.

zocker-160, 2024-02-15:
I just checked the source code of g15daemon and its dependency libg15 and found two notable things:

first it does indeed use libusb which is probably why libHIDraw is unable to see the keyboard:
https://gitlab.com/menelkir/libg15/-/blob/master/libg15.c?ref_type=heads#L168

and I also found the disable key sequence, which is the same as I use for the other G510 model, so that should work:
https://gitlab.com/menelkir/libg15/-/blob/master/libg15.c?ref_type=heads#L124

JoZ3, 2024-02-15:
Hi, I disabled the g15daemon service and restarted the system, the script worked fine, I pressed the keys in the order @Flying--Dutchman did: 

```
M1, M2, M3, MR
G1 - G18
Game slider (on and off)
Headphone sound off
Microphone mute
LCD - App switch
LCD - Buttons 1-4
Keyboard light on/off
Play/Pause
Stop
Previous
Next
Sound off
Volume wheel (both directions)
```

I noticed something, I don't know if it is normal or there is a problem with my keyboard, when I press the keys M1, M2, M3 and MR, it doesn't shows information, the information shows from the G1 key.

```
DEBUG: Searching for keyboard...
DEBUG: Keyboard found! :)
DEBUG: requesting USB endpoint...
DEBUG: HIDraw read endpoint found: /dev/hidraw2
DEBUG: HIDraw disable endpoint found: /dev/hidraw2
DEBUG: HIDraw read endpoint found: /dev/hidraw2
DEBUG: HIDraw disable endpoint found: /dev/hidraw2
DEBUG: Checking for HID availability...
DEBUG: Connected to /dev/hidraw2
DEBUG: Connected to /dev/hidraw2
INFO: starting listener...
DEBUG: Using selector: EpollSelector
DEBUG: Connection using HIDAPI...
DEBUG: Sending sequence to disable G keys
0: b'\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
1: b'\x07\x03\x00'
2: b'\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
DEBUG: listening to USB Interface (1, 0) | b'/dev/hidraw2'
got data from keyboard: b'\x03\x01\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x03\x02\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x03\x04\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x03\x08\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x03\x10\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x03 \x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x03@\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x03\x80\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x03\x00\x01\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x03\x00\x02\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x03\x00\x04\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x03\x00\x08\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x03\x00\x10\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x03\x00 \x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x03\x00@\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
[... 94 lines cut ...]
got data from keyboard: b'\x02\x00'
got data from keyboard: b'\x02 '
got data from keyboard: b'\x02\x00'
got data from keyboard: b'\x02 '
got data from keyboard: b'\x02 '
got data from keyboard: b'\x02\x00'
got data from keyboard: b'\x02 '
got data from keyboard: b'\x02\x00'
got data from keyboard: b'\x02 '
got data from keyboard: b'\x02@'
got data from keyboard: b'\x02 '
got data from keyboard: b'\x02@'
got data from keyboard: b'\x02 '
got data from keyboard: b'\x02\x00'
got data from keyboard: b'\x02 '
got data from keyboard: b'\x02\x00'
got data from keyboard: b'\x01\x01\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x01\x01\x06\x00\x00\x00\x00\x00'
^CHIDerrpr: Interrupted system call
got data from keyboard: b'\x01\x01\x00\x00\x00\x00\x00\x00'
got data from keyboard: b'\x01\x00\x00\x00\x00\x00\x00\x00'
INFO: stopping...
ERROR: Task was destroyed but it is pending!
task: <Task pending name='Task-1' coro=<usbListener() running at /home/myuser/keyboard-center/src/newDeviceDebugger.py:160>>
```

zocker-160, 2024-02-15:
oh nice this looks better now, mapping the buttons right now.

Are you sure that G1 and M1 are the same?

zocker-160, 2024-02-15:
ok no it looks similar, but they are different:

M1:  `b'\x03\x01\x00\x00\x00'`
G1: `b'\x03\x10\x00\x00\x00'`

zocker-160, 2024-02-15:
so I think I have almost all buttons mapped, I am not sure about G16 - 18 though, does it send 3 lines for one button press?

JoZ3, 2024-02-15:
G16 - 18

```
got data from keyboard: b'\x03\x00\x80\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x01\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x02\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'

```

JoZ3, 2024-02-15:
> oh nice this looks better now, mapping the buttons right now.
> 
> ~Are you sure that G1 and M1 are the same?~ EDIT: they are different:
> 
> M1: `b'\x03\x01\x00\x00\x00'` G1: `b'\x03\x10\x00\x00\x00'`
> 
> But they are indeed inverted compared to the other (older?) G510, which is really strange why they did that.

I don't understand something, when I press M1, M2, M3 and MR no information appears, the one you indicate as M1, is the information  when I press G1: 
```

got data from keyboard: b'\x03\x01\x00\x00\x00'
got data from keyboard: b'\x03\x00\x00\x00\x00'
```

zocker-160, 2024-02-15:
ah damn, the Mkeys then seem to be using a different USB channel, that sucks

zocker-160, 2024-02-15:
could you please try changing the line 14 `usbInterface = (1, 0)` to `(0, 0)` or `(2, 0)` etc to see if those show something when you press M key?

JoZ3, 2024-02-15:
> could you please try changing the line 14 `usbInterface = (1, 0)` to `(0, 0)` or `(2, 0)` etc to see if those show something when you press M key?

with (0,0) nothing happend

with (2,0)  this:

```
DEBUG: Searching for keyboard...
DEBUG: Keyboard found! :)
DEBUG: requesting USB endpoint...
Traceback (most recent call last):
  File "/home/myuser/keyboard-center/src/newDeviceDebugger.py", line 230, in <module>
    main()
  File "/home/myuser/keyboard-center/src/newDeviceDebugger.py", line 191, in main
    keyboardEndpoint: core.Endpoint = keyboard\
                                      ^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/usb/core.py", line 534, in __getitem__
    return Endpoint(
           ^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/usb/core.py", line 347, in __init__
    desc = backend.get_endpoint_descriptor(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/usb/backend/libusb1.py", line 798, in get_endpoint_descriptor
    ep_desc = i.endpoint[ep]
              ~~~~~~~~~~^^^^
ValueError: NULL pointer access
```

JoZ3, 2024-02-15:
I will install a windows virtual machine and install the keyboard drivers to verify that there is no physical damage, although it seems strange to me, a long time ago the M1 - 3 and MR keys worked very well with the g15daemon and its utilities.

zocker-160, 2024-02-15:
here an initial version for testing:
[keyboard-center-git.zip](https://github.com/zocker-160/keyboard-center/files/14302375/keyboard-center-git.zip)

JoZ3, 2024-02-15:
> here an initial version for testing: [keyboard-center-git.zip](https://github.com/zocker-160/keyboard-center/files/14302375/keyboard-center-git.zip)

No device found :(

```
[root] INFO: 15.02.2024 04:29:43 PM ------------ starting -------------
[Configparser] DEBUG: 15.02.2024 04:29:43 PM Config file location: /home/XXXX/.config/keyboard-center/settings.yml
[Configparser] DEBUG: 15.02.2024 04:29:43 PM loading config file /home/XXXX/.config/keyboard-center/settings.yml
[Configparser] DEBUG: 15.02.2024 04:29:43 PM config loaded: {'mappings': {'MEMORY_1': {'MACRO_1': {'name': 'example-A', 'string': 'A', 'type': 'key', 'value': [1, 30]}, 'MACRO_2': {'name': 'example-cut', 'string': [['Ctrl', 'X'], ['Ctrl', 'V']], 'type': 'macro', 'value': [[[1, 29], [1, 45]], [[1, 29], [1, 47]]]}, 'MACRO_3': {'name': 'example-nice', 'string': [['N'], ['I'], ['C'], ['E']], 'type': 'macro', 'value': [[[1, 49]], [[1, 23]], [[1, 46]], [[1, 18]]]}}}, 'settings': {'retryCount': 5, 'usbDeviceID': 'None', 'usbTimeout': 1000}}
[BGService] INFO: 15.02.2024 04:29:43 PM setting up service...
[BGService] INFO: 15.02.2024 04:29:43 PM searching for supported keyboard...
[BGService] CRITICAL: 15.02.2024 04:29:43 PM no supported keyboard found, retrying...
[Configparser] DEBUG: 15.02.2024 04:29:43 PM setting deviceID to None
[QT] DEBUG: 15.02.2024 04:29:48 PM start health check...
[QT] DEBUG: 15.02.2024 04:29:48 PM setting up GUI...
[Configparser] DEBUG: 15.02.2024 04:29:48 PM loading config file /home/XXXX/.config/keyboard-center/settings.yml
[Configparser] DEBUG: 15.02.2024 04:29:48 PM config loaded: {'mappings': {'MEMORY_1': {'MACRO_1': {'name': 'example-A', 'string': 'A', 'type': 'key', 'value': [1, 30]}, 'MACRO_2': {'name': 'example-cut', 'string': [['Ctrl', 'X'], ['Ctrl', 'V']], 'type': 'macro', 'value': [[[1, 29], [1, 45]], [[1, 29], [1, 47]]]}, 'MACRO_3': {'name': 'example-nice', 'string': [['N'], ['I'], ['C'], ['E']], 'type': 'macro', 'value': [[[1, 49]], [[1, 23]], [[1, 46]], [[1, 18]]]}}}, 'settings': {'retryCount': 5, 'usbDeviceID': 'None', 'usbTimeout': 1000}}
Traceback (most recent call last):
  File "/opt/keyboard-center-git/mainUi.py", line 376, in closeEvent
    self.service.quit(False)
    ^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'quit'

```

And apparently the M1, M2 and M3 buttons are not working, from the virtual machine and the logitech center you can switch between M1, M2 and M3, but physically you can't, I will check it on a computer with windows 10 installed.

zocker-160, 2024-02-15:
> No device found :(

yes I did a mistake sorry :V

zocker-160, 2024-02-15:
There you go, I hope this one works
[keyboard-center-git.zip](https://github.com/zocker-160/keyboard-center/files/14302642/keyboard-center-git.zip)

zocker-160, 2024-02-15:
awesome glad to hear, I will push a new version with those changes tomorrow.

As for the remaining functionality:
- the media keys should work (start / stop, volume etc)
- for the M keys I can look into implementing a way to switch in software
- for the LCD screen, that is out of scope of this project, one would need to implement a service based on HIDraw talking to the LCD in a separate application
- I do not know if the G510 has RGB keys, but if yes it should be connectable with OpenRGB

zocker-160, 2024-02-16:
1.0.8 released, so closing.

Any of the other points mentioned above should be a separate issue if desired.

closed 2024-02-16 by zocker-160