zocker-160/keyboard-center #61: Python update broke 'uinput'

lesu1, 2024-04-30:
Hi,
I think we have an issue with Python update 3.12.3-1 and 'uinput'  again. 
Not sure what's the problem this time, but to me it looks the same:

kernel: 6.8.7-arch1-2 (64-bit)
Python: 3.12.3-1

4/30/24 12:23 PM	systemd	Starting Keyboard Center...
4/30/24 12:23 PM	systemd	Started Keyboard Center.
4/30/24 12:23 PM	keyboard-center	Traceback (most recent call last):
4/30/24 12:23 PM	keyboard-center	  File "/opt/keyboard-center/main.py", line 8, in <module>
4/30/24 12:23 PM	keyboard-center	    from mainUi import MainWindow
4/30/24 12:23 PM	keyboard-center	  File "/opt/keyboard-center/mainUi.py", line 20, in <module>
4/30/24 12:23 PM	keyboard-center	    from devices.keyboard import SUPPORTED_DEVICES, KeyboardInterface
4/30/24 12:23 PM	keyboard-center	  File "/opt/keyboard-center/devices/keyboard.py", line 1, in <module>
4/30/24 12:23 PM	keyboard-center	    import uinput
4/30/24 12:23 PM	keyboard-center	ModuleNotFoundError: No module named 'uinput'

I've tried to timeshift back to Python 3.11.8, everything is good, but i have some other issues with the other apps, not related to keyboard-center.

Maybe you could have a look into this?

Thanks,
lesu1

zocker-160, 2024-04-30:
thank you for reporting, I will take a look

zocker-160, 2024-04-30:
I guess you are on arch, so in that case the python-uinput module was transferred to a new owner / org on GitHub (https://github.com/tuomasjjrasanen/python-uinput/issues/49#issuecomment-1951352788), which should include the fix needed for Python 3.12+.

I have marked the AUR package as outdated, if that gets moved to the new version, it should (hopefully) work again.

lesu1, 2024-05-06:
This is not the case anymore, with the git version from: https://github.com/pyinput/python-uinput

Just clone the repo and

```
python setup.py build
python setup.py install
```
Probably with 'sudo'.

Thanks, lesu1

closed 2024-05-06 by lesu1

zocker-160, 2024-05-06:
yeah the AUR package needs to be updated to that version, it is already marked out of date, but no update yet.

zocker-160, 2024-06-04:
just a heads up: 

I am pushing a new version which is intentionally setting the minimum version of `python-uinput` to version 1.0.1 which is going to break the AUR package.

Please use the workaround mentioned above (https://github.com/zocker-160/keyboard-center/issues/61#issuecomment-2095591733) until the AUR package is updated (it is marked out of date, but nothing is happening).

zocker-160, 2024-07-03:
I have finally managed to take over the `python-uinput` AUR package, it should now finally work without manual intervention.