zocker-160/keyboard-center #62: Save button not working

Sa-Kage, 2024-06-03:
As the title says, the save button does do nothing for me. The only option to save to the config file (~/.config/keyboard-center/settings.yml) is closing the app and click "Yes" when ask if I want to save...

System info:
```
System:
  Kernel: 5.15.0-107-generic x86_64 bits: 64 compiler: gcc v: 11.4.0 Desktop: Cinnamon 6.0.4
    tk: GTK 3.24.33 wm: muffin vt: 7 dm: LightDM 1.30.0 Distro: Linux Mint 21.3 Virginia
    base: Ubuntu 22.04 jammy
Machine:
  Type: Desktop System: ASUS product: N/A v: N/A serial: <superuser required>
  Mobo: ASUSTeK model: PRIME H770-PLUS D4 v: Rev 1.xx serial: <superuser required>
    UEFI: American Megatrends v: 0401 date: 09/30/2022
CPU:
  Info: 16-core (8-mt/8-st) model: 13th Gen Intel Core i7-13700KF bits: 64 type: MST AMCP
    smt: enabled arch: N/A rev: 1 cache: L1: 1.4 MiB L2: 24 MiB L3: 30 MiB
  Speed (MHz): avg: 1110 high: 2242 min/max: 800/5300:5400:4200 cores: 1: 1861 2: 1100 3: 1106
    4: 1100 5: 1101 6: 1100 7: 1090 8: 1101 9: 1104 10: 1103 11: 1095 12: 1100 13: 1094 14: 1100
    15: 2242 16: 1101 17: 800 18: 802 19: 800 20: 1536 21: 800 22: 800 23: 800 24: 814
    bogomips: 164044
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
Graphics:
  Device-1: NVIDIA TU104 [GeForce RTX 2080 Rev. A] vendor: Micro-Star MSI driver: nvidia
    v: 535.171.04 pcie: speed: 2.5 GT/s lanes: 16 ports: active: none off: DP-2,DP-3,HDMI-A-1
    empty: DP-1,Unknown-1 bus-ID: 01:00.0 chip-ID: 10de:1e87 class-ID: 0300
  Display: x11 server: X.Org v: 1.21.1.4 driver: X: loaded: nvidia
    unloaded: fbdev,modesetting,nouveau,vesa gpu: nvidia display-ID: :0 screens: 1
  Screen-1: 0 s-res: 4920x1920 s-dpi: 96 s-size: 1302x508mm (51.3x20.0") s-diag: 1398mm (55")
  Monitor-1: DP-3 pos: primary,top-left res: 1080x1920 hz: 60 dpi: 93
    size: 296x527mm (11.7x20.7") diag: 604mm (23.8")
  Monitor-2: DP-5 pos: bottom-r res: 1920x1080 hz: 60 dpi: 82 size: 598x336mm (23.5x13.2")
    diag: 686mm (27")
  Monitor-3: HDMI-0 pos: primary,bottom-c res: 1920x1080 hz: 60 dpi: 82
    size: 598x336mm (23.5x13.2") diag: 686mm (27")
  OpenGL: renderer: NVIDIA GeForce RTX 2080/PCIe/SSE2 v: 4.6.0 NVIDIA 535.171.04
    direct render: Yes
Audio:
  Device-1: Intel vendor: ASUSTeK driver: snd_hda_intel v: kernel bus-ID: 00:1f.3
    chip-ID: 8086:7a50 class-ID: 0403
  Device-2: NVIDIA TU104 HD Audio vendor: Micro-Star MSI driver: snd_hda_intel v: kernel pcie:
    speed: 8 GT/s lanes: 16 bus-ID: 01:00.1 chip-ID: 10de:10f8 class-ID: 0403
  Device-3: YZ 2.4G Wireless headset type: USB driver: hid-generic,snd-usb-audio,usbhid
    bus-ID: 1-12:6 chip-ID: 7545:0897 class-ID: 0300
  Sound Server-1: ALSA v: k5.15.0-107-generic running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes
Network:
  Device-1: Realtek RTL8125 2.5GbE vendor: ASUSTeK driver: r8169 v: kernel pcie: speed: 5 GT/s
    lanes: 1 port: 4000 bus-ID: 06:00.0 chip-ID: 10ec:8125 class-ID: 0200
  IF: eno1 state: up speed: 1000 Mbps duplex: full mac: <filter>
  IF-ID-1: lxcbr0 state: down mac: <filter>
RAID:
  Hardware-1: Intel Device driver: N/A v: N/A port: N/A bus-ID: 00:0e.0 chip-ID: 8086:a77f rev:
    class-ID: 0104
Drives:
  Local Storage: total: 6.37 TiB used: 2.29 TiB (35.9%)
  ID-1: /dev/nvme0n1 vendor: Crucial model: CT1000P3SSD8 size: 931.51 GiB speed: 31.6 Gb/s
    lanes: 4 type: SSD serial: <filter> rev: P9CR30D temp: 31.9 C scheme: GPT
  ID-2: /dev/sda vendor: Western Digital model: WD30EZRX-00SPEB0 size: 2.73 TiB speed: 6.0 Gb/s
    type: HDD rpm: 5400 serial: <filter> rev: 0A80 scheme: GPT
  ID-3: /dev/sdb vendor: Hitachi model: HDS721050CLA362 size: 465.76 GiB speed: 3.0 Gb/s
    type: HDD rpm: 7200 serial: <filter> rev: A50E scheme: MBR
  ID-4: /dev/sdc vendor: Samsung model: SSD 860 EVO 500GB size: 465.76 GiB speed: 6.0 Gb/s
    type: SSD serial: <filter> rev: 1B6Q scheme: GPT
  ID-5: /dev/sdd vendor: Western Digital model: WD20EZRX-00D8PB0 size: 1.82 TiB speed: 6.0 Gb/s
    type: HDD rpm: 5400 serial: <filter> rev: 0A80 scheme: GPT
Partition:
  ID-1: / size: 58.76 GiB used: 30.07 GiB (51.2%) fs: ext4 dev: /dev/nvme0n1p4
  ID-2: /boot/efi size: 476 MiB used: 6.1 MiB (1.3%) fs: vfat dev: /dev/nvme0n1p3
  ID-3: /home size: 43.39 GiB used: 12.35 GiB (28.5%) fs: ext4 dev: /dev/nvme0n1p5
Swap:
  ID-1: swap-1 type: partition size: 36.7 GiB used: 0 KiB (0.0%) priority: -2 dev: /dev/sdd3
USB:
  Hub-1: 1-0:1 info: Hi-speed hub with single TT ports: 16 rev: 2.0 speed: 480 Mb/s
    chip-ID: 1d6b:0002 class-ID: 0900
  Device-1: 1-2:2 info: ASUSTek AURA LED Controller type: HID driver: hid-generic,usbhid
    interfaces: 2 rev: 2.0 speed: 12 Mb/s power: 16mA chip-ID: 0b05:19af class-ID: 0300
    serial: <filter>
  Device-2: 1-3:3 info: Logitech G502 SE HERO Gaming Mouse type: Mouse,HID
    driver: hid-generic,usbhid interfaces: 2 rev: 2.0 speed: 12 Mb/s power: 300mA chip-ID: 046d:c08b
    class-ID: 0300 serial: <filter>
  Device-3: 1-4:4 info: Logitech G910 Orion Spectrum Mechanical Keyboard type: Keyboard,HID
    driver: hid-generic,usbhid interfaces: 2 rev: 2.0 speed: 12 Mb/s power: 500mA chip-ID: 046d:c335
    class-ID: 0300 serial: <filter>
  Hub-2: 1-10:5 info: ASMedia ASM1074 High-Speed hub ports: 4 rev: 2.1 speed: 480 Mb/s
    power: 100mA chip-ID: 174c:2074 class-ID: 0900
  Device-1: 1-12:6 info: YZ 2.4G Wireless headset type: Audio,HID
    driver: hid-generic,snd-usb-audio,usbhid interfaces: 4 rev: 1.1 speed: 12 Mb/s power: 400mA
    chip-ID: 7545:0897 class-ID: 0300
  Hub-3: 2-0:1 info: Super-speed hub ports: 10 rev: 3.1 speed: 20 Gb/s chip-ID: 1d6b:0003
    class-ID: 0900
  Hub-4: 2-10:2 info: ASMedia ASM1074 SuperSpeed hub ports: 4 rev: 3.0 speed: 5 Gb/s power: 8mA
    chip-ID: 174c:3074 class-ID: 0900
  Hub-5: 3-0:1 info: Hi-speed hub with single TT ports: 2 rev: 2.0 speed: 480 Mb/s
    chip-ID: 1d6b:0002 class-ID: 0900
  Hub-6: 4-0:1 info: Super-speed hub ports: 4 rev: 3.1 speed: 10 Gb/s chip-ID: 1d6b:0003
    class-ID: 0900
Sensors:
  System Temperatures: cpu: 27.8 C mobo: N/A gpu: nvidia temp: 30 C
  Fan Speeds (RPM): N/A gpu: nvidia fan: 26%
Repos:
  Packages: 2576 apt: 2563 flatpak: 13
  No active apt repos in: /etc/apt/sources.list
  Active apt repos in: /etc/apt/sources.list.d/anydesk-stable.list
    1: deb http: //deb.anydesk.com/ all main
  Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
    1: deb http: //linux-mint.froonix.org virginia main upstream import backport
    2: deb http: //artfiles.org/ubuntu.com jammy main restricted universe multiverse
    3: deb http: //artfiles.org/ubuntu.com jammy-updates main restricted universe multiverse
    4: deb http: //artfiles.org/ubuntu.com jammy-backports main restricted universe multiverse
    5: deb http: //security.ubuntu.com/ubuntu/ jammy-security main restricted universe multiverse
  No active apt repos in: /etc/apt/sources.list.d/steam-beta.list
  Active apt repos in: /etc/apt/sources.list.d/steam-stable.list
    1: deb [arch=amd64,i386 signed-by=/usr/share/keyrings/steam.gpg] https: //repo.steampowered.com/steam/ stable steam
    2: deb-src [arch=amd64,i386 signed-by=/usr/share/keyrings/steam.gpg] https: //repo.steampowered.com/steam/ stable steam
  Active apt repos in: /etc/apt/sources.list.d/waydroid.list
    1: deb [signed-by=/usr/share/keyrings/waydroid.gpg] https: //repo.waydro.id/ jammy main
  Active apt repos in: /etc/apt/sources.list.d/winehq-jammy.sources
    1: deb [arch=amd64 i386] https: //dl.winehq.org/wine-builds/ubuntu jammy main
Info:
  Processes: 440 Uptime: 11m wakeups: 1 Memory: 31.15 GiB used: 2.31 GiB (7.4%) Init: systemd
  v: 249 runlevel: 5 Compilers: gcc: 11.4.0 alt: 11/12 Client: Cinnamon v: 6.0.4 inxi: 3.3.13
```

zocker-160, 2024-06-03:
can you share the log file if one exists? (`.var/log/keyboardCenter.log`)

It is anonymized.

lesu1, 2024-06-04:
My workaround for this bug is to close the main window via X and click Save. fyi

Sa-Kage, 2024-06-04:
tried adding "Ctrl + S" to macro key 1 just for testing here
confirmed it was not written to settings file
[keyboardCenter.log](https://github.com/user-attachments/files/15548581/keyboardCenter.log)

zocker-160, 2024-06-04:
This is very weird, according to the log the file is saved, but it seems like it does not load correctly. 
Could you send me your `.config/keyboard-center/settings.yml` ?

zocker-160, 2024-06-04:
I think I found the issue, I fucked up somewhere around the config cleanup, I am looking into it.

Sa-Kage, 2024-06-04:
Just in case it is still needed:
settings.yml (file type is not allowed here)
```
mappings:
  MEMORY_1:
    MACRO_2:
      gamemode: 0
      name: paste
      string: [Ctrl, V]
      type: combo
      value:
      - [1, 29]
      - [1, 47]
    MACRO_3:
      gamemode: 0
      name: copy
      string: [Ctrl, C]
      type: combo
      value:
      - [1, 29]
      - [1, 46]
    MACRO_4:
      gamemode: 0
      name: cut
      string: [Ctrl, X]
      type: combo
      value:
      - [1, 29]
      - [1, 45]
openRGB: {MEMORY_1: White-KeyBlue, White-KeyBlue: null}
settings: {minOnStart: true, retryCount: 5, showNotifications: false, usbDeviceID: 0,
  usbTimeout: 1000}
```

zocker-160, 2024-06-04:
Thank you, yep you have the same issue that I managed to reproduce, happens with your file as well.

zocker-160, 2024-06-04:
to your edit: yes it is stored in memory correctly, the saving to file fails and on the next load your change will get lost.

commit 55d04c67 (2024-06-04): fix save to file when pressing save

fixes #62
closed 2024-06-04 by zocker-160