zocker-160/keyboard-center #64: cannot access local variable 'e' where it is not associated with a value

recallmenot, 2024-06-11:
I've stumbled into a bug preventing keyboard-center from launching.
OS is manjaro, up-to-date.
Tried reinstalling keyboard-center (1.0.9-1) from AUR, using `yay -Sdd` to skip the python-uinput dependency and [installing it manually](https://github.com/zocker-160/keyboard-center/issues/61#issuecomment-2095591733).

Should work, no?
keyboard-center used to work fine on this PC.

running with sudo didn't help.

This is the lower part of my `~/.var/log/keyboardCenter.log`
```
[BGService] INFO: 11.06.2024 10:41:24  setting up service...
[BGService] INFO: 11.06.2024 10:41:24  searching for supported keyboard...
[BGService] INFO: 11.06.2024 10:41:24  keyboard found: Logitech G710+
[Configparser] DEBUG: 11.06.2024 10:41:24  setting deviceID to 1
[BGService] DEBUG: 11.06.2024 10:41:24  time taken to find keyboard in ms: 14.10675048828125
[BGService] DEBUG: 11.06.2024 10:41:24  requesting USB endpoint...
[BGService] DEBUG: 11.06.2024 10:41:24  Searching for HID endpoints...
[root] ERROR: 11.06.2024 10:41:24  cannot access local variable 'e' where it is not associated with a value
Traceback (most recent call last):
  File "/opt/keyboard-center/mainUi.py", line 142, in initBackgroundService
    self.service = BackgroundService(self.configparser, useOpenRGB)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/keyboard-center/service.py", line 85, in __init__
    self._initKeyboard()
  File "/opt/keyboard-center/service.py", line 125, in _initKeyboard
    self._getHIDpaths()
  File "/opt/keyboard-center/service.py", line 165, in _getHIDpaths
    raise NoEndpointException()
service.NoEndpointException

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/keyboard-center/main.py", line 52, in <module>
    window = MainWindow(app, devmode, not bgmode)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/keyboard-center/mainUi.py", line 55, in __init__
    self.initBackgroundService()
  File "/opt/keyboard-center/mainUi.py", line 153, in initBackgroundService
    detailText=str(e),
                   ^
UnboundLocalError: cannot access local variable 'e' where it is not associated with a value
```

Running keyboard-center form terminal gives more "error":
```
[root] ERROR: 11.06.2024 10:51:48  cannot access local variable 'e' where it is not associated with a value
Traceback (most recent call last):
  File "/opt/keyboard-center/mainUi.py", line 142, in initBackgroundService
    self.service = BackgroundService(self.configparser, useOpenRGB)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/keyboard-center/service.py", line 85, in __init__
    self._initKeyboard()
  File "/opt/keyboard-center/service.py", line 125, in _initKeyboard
    self._getHIDpaths()
  File "/opt/keyboard-center/service.py", line 165, in _getHIDpaths
    raise NoEndpointException()
service.NoEndpointException

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/keyboard-center/main.py", line 52, in <module>
    window = MainWindow(app, devmode, not bgmode)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/keyboard-center/mainUi.py", line 55, in __init__
    self.initBackgroundService()
  File "/opt/keyboard-center/mainUi.py", line 153, in initBackgroundService
    detailText=str(e),
                   ^
UnboundLocalError: cannot access local variable 'e' where it is not associated with a value
Traceback (most recent call last):
  File "/opt/keyboard-center/mainUi.py", line 142, in initBackgroundService
    self.service = BackgroundService(self.configparser, useOpenRGB)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/keyboard-center/service.py", line 85, in __init__
    self._initKeyboard()
  File "/opt/keyboard-center/service.py", line 125, in _initKeyboard
    self._getHIDpaths()
  File "/opt/keyboard-center/service.py", line 165, in _getHIDpaths
    raise NoEndpointException()
service.NoEndpointException

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/keyboard-center/main.py", line 52, in <module>
    window = MainWindow(app, devmode, not bgmode)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/keyboard-center/mainUi.py", line 55, in __init__
    self.initBackgroundService()
  File "/opt/keyboard-center/mainUi.py", line 153, in initBackgroundService
    detailText=str(e),
                   ^
UnboundLocalError: cannot access local variable 'e' where it is not associated with a value
```

zocker-160, 2024-06-11:
Probably a bug, will look into it.

Running as root is not a good idea though as the config and the log file will then also be owned by root.

zocker-160, 2024-06-11:
ok so there are two things happening here:

A NoEndPointException is thrown, which means that the keyboard was not found. During the handling of that Exception it crashes because of a stupid copy paste bug I made.

zocker-160, 2024-06-11:
NoEndpointException is thrown, because it fails to find the HIDraw endpoint of the keyboard, that is however not a bug, but rather a problem with your system.

Usually caused by other keyboard software unloading the kernel driver.

recallmenot, 2024-06-13:
thank you for looking into it!

I'm glad there actually was something wrong because I rebooted today (1W standby on modern platforms FTW) and keyboard-center now launches just fine again, without rebuilding it o.O

closed 2024-06-13 by recallmenot