zocker-160/keyboard-center #65: Feature Request: Distinguish between key-up and key-down

drunken-dwarf, 2024-07-01:
It would be nice if we could send only key up/down inputs. That way one could create macros where keys are pressed for an exact amount of time.

Example sequence:

Key-down: e
Sleep 175ms
Key-up: e

zocker-160, 2024-07-01:
Accepted, but as with the other currently pending feature requests, I sadly currently don't have the time to work on this.

Being able to differentiate between key click, down and up will also require changes to the config file format, which is already quite a mess.

I cannot promise any ETA ._.

zocker-160, 2024-09-05:
With v2 I have added lua scripting which allows you to do this. I know it is not exactly what you asked for, but if I added every single request for special behavior, I would end up with a huge confusing list of checkboxes.

You can find an example script for keyup / keydown and delay in the examples [here](https://github.com/zocker-160/keyboard-center/blob/master/lua_examples/01-simplekeyupdown.lua) and [here](https://github.com/zocker-160/keyboard-center/blob/master/lua_examples/03-emithold.lua).

closed 2024-09-05 by zocker-160