zocker-160/keyboard-center #66: Feature Request: Add settings for loop while holding, toggle loop and single run

drunken-dwarf, 2024-07-01:
The Logitech application allows you to run sequences in a loop when holding the mapped key, toggle a loop or simply run it once.
It would be nice if we also got that feature as it allows for far more versatility.

When loop-on-hold is selected, no longer pressing the mapped key results in the sequence stopping where it currently is at.

zocker-160, 2024-07-01:
This feature makes sense, a similar feature is something that I also wanted to implement some time ago, so I am accepting this, but as with all the other feature requests, I cannot promise an ETA.

zocker-160, 2024-09-05:
As stated in the other issue you opened, with lua scripting you can now do most of what you are mentioning here. 

Currently not possible is "toggle loop on keypress" as each keypress has its own state and is reset with every press.
In order to make this possible, I would need to think of a way to set global variables which persist between keypresses.

zocker-160, 2024-09-05:
Ok so I added new functions to set global registers which persist between keypresses. 

With those it is now possible to toggle a loop on and off with each key press. Also you can loop-on-hold by adding the register check after each action instead of once for each loop.

An example script can be found [here](https://github.com/zocker-160/keyboard-center/blob/master/lua_examples/06-looptoggle.lua).

closed 2024-09-05 by zocker-160