zocker-160/keyboard-center #67: Adding a build system

aurtzy, 2024-10-19:
Hi!

Is there interest in adding a build system?  I've been experimenting with adding one to keyboard-center recently for educational purposes, but I would be happy to make a pull request and discuss merging these changes.

zocker-160, 2024-10-19:
Hi,

in general I am all for a build system, but I wonder what a build target would be, since there is nothing to compile. For creating the DEB or AUR package, all I am doing is moving a few files into the correct location.

see the package script (or the PKGBUILD of the AUR package):
https://github.com/zocker-160/keyboard-center/blob/17b57e97fbfed5d3e775df3f6f111efe1b7826c0/linux_packaging/DEB-keyboard-center/lure.sh#L37

aurtzy, 2024-10-19:
Perhaps "build system" might not be the best term in the case of Python? My changes amount mostly to the addition of a `pyproject.toml` and directory structure changes to a standard format so that a given distribution's Python packaging guidelines can be ideally followed (for example, for [Arch Linux](https://wiki.archlinux.org/title/Python_package_guidelines#Standards_based_(PEP_517))).

I think this is more of an exception, but the Guix package manager (which I've [packaged](https://github.com/aurtzy/guix-config/blob/master/modules/my-guix/packages/keyboard-center.scm#L93) keyboard-center for, but haven't brought to upstream quite yet...) does actually do some sort of automatic cpython compilation/caching during the build for packages that are able to use its Python build system.

Unfortunately (fortunately?), because keyboard-center is already very easy to package by just copying some files, there's not many benefits I can list other than future-proofing or niceties like dependency checking. WDYT?

zocker-160, 2024-10-19:
> My changes amount mostly to the addition of a pyproject.toml and directory structure changes to a standard format so that a given distribution's Python packaging guidelines can be ideally followed [...]

Ah ok now I get it, yes I agree, the current situation is a _bit_ of a hackjob.

So yes I am all for it if you want to contribute that, especially if that makes packaging for other distributions easier. 
I myself was happy enough to have a working AUR let alone a GUIX one xD.

IIRC when packing the files as a wheel and installing that, I will probably also have to update the starter script, which currently expects itself to be placed in `/opt` (which yes I know is also not great.....).

closed 2024-10-26 by zocker-160