zocker-160/keyboard-center #72: Failed to load configuration file!

Chris321sirhc, 2024-12-14:
I get this error after installing via terminal on Ubuntu 24.04.01 LTS:

[Errno 2] No such file or directory: '/home/chris/.config/keyboard-center/settings.json'

[image]

Manually creating an empty settings.json file did not resolve this for me. As I am incredibly annoyed by the G5 key mapping in particular I would love to use this tool.

Looking forward to your help, thanks very much!

Kind regards,
Chris :)

zocker-160, 2024-12-14:
Thanks for this report, this is probably a bug.

Could you share the log file, if one exists? (you can find it in `~/.var/log`)

Chris321sirhc, 2024-12-14:
No worries, here is the log:

```bash
cat ~/.var/log/keyboardCenter.log 
[root] INFO: 14.12.2024 04:47:19  ------------ Keyboard Center 2.0.4 -------------
[ConfigLoader] DEBUG: 14.12.2024 04:47:19  config file location: /home/XXXX/.config/keyboard-center/settings.json
[ConfigLoader] DEBUG: 14.12.2024 04:47:19  config file not found - creating default
```

However, contrary to the message no config is created. I should note that the directory existed without me creating it, just not the file.

zocker-160, 2024-12-14:
Do you have write permission in that folder?

Until I find the issue, you can use the config file in this repository and simply copy it to that location, that should make this application at least work.

zocker-160, 2024-12-14:
Actually I think I found the issue already, I feel kinda stupid for not noticing it earlier, I am looking into it.

zocker-160, 2024-12-14:
Sadly cannot reproduce the issue, it creates the file just fine for me.

You can use this in the mean time:
[settings.json](https://github.com/user-attachments/files/18136708/settings.json)

zocker-160, 2024-12-14:
I just noticed that you are not using the latest version? I did fix config related issues between 2.0.4 and 2.0.5

Chris321sirhc, 2024-12-14:
Hello, thank you for your immediate investigation. The settings.json fixed the problem right away. I previously installed the top most entry under releases (or so I thought) but have now installed the current one.

Thank you very much! :)