zocker-160/keyboard-center #73: keyboard-center2 on AUR broken

recallmenot, 2025-01-23:
Hi,
I'd like to report that keyboard-center2 on the AUR is broken because uinput apparently can't be found:
```
yay -S keyboard-center2
AUR Explicit (1): keyboard-center2-2.0.5-3
:: PKGBUILD up to date, skipping download: keyboard-center2
  1 keyboard-center2                         (Build Files Exist)
==> Packages to cleanBuild?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> 
  1 keyboard-center2                         (Build Files Exist)
==> Diffs to show?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> 
==> Making package: keyboard-center2 2.0.5-3 (Thu 23 Jan 2025 02:18:41 PM CET)
==> Retrieving sources...
  -> Updating keyboard-center2-2.0.5 git repo...
==> WARNING: Skipping verification of source file PGP signatures.
==> Validating source files with sha256sums...
    keyboard-center2-2.0.5 ... Skipped
:: (1/1) Parsing SRCINFO: keyboard-center2
==> Making package: keyboard-center2 2.0.5-3 (Thu 23 Jan 2025 02:18:43 PM CET)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Updating keyboard-center2-2.0.5 git repo...
==> Validating source files with sha256sums...
    keyboard-center2-2.0.5 ... Skipped
==> Removing existing $srcdir/ directory...
==> Extracting sources...
  -> Creating working copy of keyboard-center2-2.0.5 git repo...
Cloning into 'keyboard-center2-2.0.5'...
done.
Switched to a new branch 'makepkg'
==> Starting prepare()...
==> Sources are ready.
==> Making package: keyboard-center2 2.0.5-3 (Thu 23 Jan 2025 02:18:45 PM CET)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> WARNING: Using existing $srcdir/ tree
==> Starting build()...
==> Entering fakeroot environment...
==> Starting package()...
==> Tidying install...
  -> Removing libtool files...
  -> Purging unwanted files...
  -> Removing static library files...
  -> Stripping unneeded symbols from binaries and libraries...
  -> Compressing man and info pages...
==> Checking for packaging issues...
==> Creating package "keyboard-center2"...
  -> Generating .PKGINFO file...
  -> Generating .BUILDINFO file...
  -> Adding install file...
  -> Generating .MTREE file...
  -> Compressing package...
==> Leaving fakeroot environment.
==> Finished making: keyboard-center2 2.0.5-3 (Thu 23 Jan 2025 02:18:48 PM CET)
==> Cleaning up...
loading packages...
resolving dependencies...
looking for conflicting packages...

Packages (1) keyboard-center2-2.0.5-3

Total Installed Size:  0.39 MiB

:: Proceed with installation? [Y/n] 
(1/1) checking keys in keyring                                           [########################################] 100%
(1/1) checking package integrity                                         [########################################] 100%
[... 181 lines cut ...]
Architecture    : x86_64
URL             : https://github.com/zocker-160/keyboard-center
Licenses        : GPL3
Groups          : None
Provides        : None
Depends On      : python>=3.12  python-pyqt5>=5.15  python-uinput>=1.0.1  python-pyusb>=1.0.2  python-lupa>=1.14
                  python-setuptools  hidapi>=0.10  libnotify>=0.7.9  lua>=5.3
Optional Deps   : openrgb: RGB control for your keyboard [installed]
                  xdotool: type strings for X11
                  ydotool: type strings using uinput [installed]
                  wtype: type strings for Wayland
Required By     : None
Optional For    : None
Conflicts With  : keyboard-center  keyboard-center-git
Replaces        : None
Installed Size  : 399.52 KiB
Packager        : Unknown Packager
Build Date      : Thu 23 Jan 2025 02:18:44 PM CET
Install Date    : Thu 23 Jan 2025 02:18:49 PM CET
Install Reason  : Explicitly installed
Install Script  : Yes
Validated By    : None
```
Maybe I'm doing something wrong again?
Any help would be appreciated!

zocker-160, 2025-01-23:
hmmm ok thanks for the report, I need to take a look, this should not be happening.

zocker-160, 2025-01-25:
I sadly cannot reproduce this issue. I also tried on a fresh install and for me it installs and runs just fine.

recallmenot, 2025-01-28:
managed to resolve it by:
```
yay -Rcns keyboard-center2
yay -S keyboard-center2
```
and choosing to rebuild the package

closed 2025-01-28 by recallmenot