zocker-160/keyboard-center #74: special quote keys (standalone accent)

recallmenot, 2025-03-06:
Hi,
I want to use keyboard-center to add markdown `quotes` and
´´´
block-quotes
´´´
When entering it in the key field directly, I get `'utf-8' codec can't encode character '\ude51' in position 1: surrogates not allowed` on hitting save.

I've managed to find a workaround: by inserting them as custom modifiers with the right-arrow as the actual key input it works but that's a bit hacky and non-obvious.

As always, thank you very much for this wonderful piece of software!

zocker-160, 2025-03-06:
confirmed, such a stupid bug, will push update soon

btw if you want to do the same using the lua interface, it would look like this (if you know a bit of coding it might be easier for you:
```lua
-- depends on keyboard layout
local LEFT_SHIFT = 42
local BACKTICK = 13

function backtick()
    for i = 1, 2 do
        KC_keyEmit(BACKTICK, 1)
        KC_keyEmit(BACKTICK, 0)
    end
end

function Start()
    KC_keyEmit(LEFT_SHIFT, 1)

    -- call only once if you only need one backtick
    for i = 1, 3 do
        backtick()
    end

    KC_keyEmit(LEFT_SHIFT, 0) 
    KC_keySyn()
end
```
The IDs might be different for you, on German keyboard I need to press shift as well.

_(I did not even know that print() can throw a Unicode exception lol one always learns something new)_

closed 2025-03-06 by zocker-160

recallmenot, 2025-03-16:
thank you very much, it works well now!