zocker-160/keyboard-center #76: OpenRGB integration seems broken

ThePixelbrain, 2025-04-20:
First of all: Thanks for creating this amazing software for Linux! It is really easy and straight forward to use, in fact a lot more straight forward than the proprietary Windows app. (RIP Logitech Gaming Software)

It seems like the OpenRGB integration broke for me when switchting to version 2.x of keyboard-center.

I have a profile named "G910" in OpenRGB, which loads successfully onto my G910 keyboard. When I open Keyboard Center my keyboard color is reset to the default lighting, even though I have specified the OpenRGB profile `G910` in Keyboard Center.

The following console log is created on startup: [Pastebin](https://pastebin.com/vEUeQUyL)

Curiously, the same exception gets thrown on Keyboard Center version 1.x however the RGB colors successfully load after a few seconds.

Some info about my system:
- Ubuntu 24.04.2 LTS (Noble Numbat)
- i3 Windows Manager
- OpenRGB  0.9.1244~1.0rc1
- Keyboard Center v2.0.6 (latest)
- Logitech G910 Orion Spectrum

If you need any more details, I'm happy to assist.

zocker-160, 2025-04-20:
Thank you for the kind words and the log, this is very useful. 

I will try to reproduce this issue, personally I was testing against a much older version of OpenRGB, this is probably a bug in my openRGB module.

May I ask you how did you install OpenRGB?

ThePixelbrain, 2025-05-03:
Sorry for the late reply.

I downloaded a OpenRGB binary from the official [website](https://openrgb.org/releases.html).
However after you asked me what version I downloaded exactly I realized that I downloaded an experimental build from their pipeline.

Switching to the latest stable version `0.9.0` resolved the issue for me. Guessing that these builds are also made in preparation for their 1.0 release, this could mean that the OpenRGB integration will break with the release of OpenRGB 1.0.

zocker-160, 2025-05-03:
Ah thank you, now I can see the issue as well!
Yep seems like they changed API of the remote interface, I will look into it.

Will keep issue open for now.

commit 32574f9a (2025-05-03): upgrade python-openrgb to v0.3.3

fixes #76
closed 2025-05-03 by zocker-160